board707/DMD_STM32 #2: can't daisy chain more than 4 modules

vtronicsautomation, 2019-10-18:
hello guys i tried to design a soccer score board with this library where im trying to design in a size of 6ftx3ft with 36 P10 modules driven by stm32f103c8  

include <angka_3.h>
#include "fonts/SystemFont5x7.h"
#include "fonts/Arial_Black_16_ISO_8859_1.h"
#include "fonts/UkrRusArial14.h";
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold9pt7b.h"

DMD_Standard_Font UkrRusArial_F(angka_3);

// We'll use SPI 2   
SPIClass dmd_spi(2);

//Fire up the DMD library as dmd
#define DISPLAYS_ACROSS 4
#define DISPLAYS_DOWN 3

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be PWM pin as PB1,PA8

// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

#define DMD_PIN_A PB6
#define DMD_PIN_B PB7
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PA8

void setup()
{
 Serial.begin(9600);
 Serial.setTimeout(5);
 // initialize Timer3
 Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
 Timer3.setPeriod(3000);          // in microseconds
 Timer3.setCompare(TIMER_CH1, 1); // overflow might be small
 Timer3.attachInterrupt(TIMER_CH1, ScanDMD);
   
 //clear/init the DMD pixels held in RAM
 dmd.clearScreen( true );   //true is normal (all pixels off), false is negative (all pixels on)
 dmd.setBrightness(6000);
 pinMode(13,OUTPUT);
 pinMode(12,OUTPUT);
 digitalWrite(13,LOW);
 digitalWrite(13,HIGH);
}

void loop()
{
    if(boot_screen_flag==1)
    {
    dmd.selectFont(&UkrRusArial_F);
    dmd.drawString( 2,3, "  POST RUN ENABLED  ", 20, GRAPHICS_NORMAL );
    delay(1000);
    dmd.drawString( 2,17, " IDC CHECK       OK", 20, GRAPHICS_NORMAL );
    delay(1000);
    dmd.drawString( 2,29, " NETWORK ADAPTER OK", 20, GRAPHICS_NORMAL );
    delay(1000);
    dmd.drawString( 2,39, " BUS SPEED TEST  OK", 20, GRAPHICS_NORMAL );  
    }
}

the problem im  facing is when i try to run more than 4 modules chained in series display starts to flicker, Is that because of timer or am i lagging some where but the library says it can run up-to 16 displays without any performance lagging but less than 4 modules it works like a charm.

board707, 2019-10-23:
Please test latest release, it should be more stable

vtronicsautomation, 2019-10-23:
Hey, thanks for your response I'll test with the new version and let you know, ...I'm working on blue pill will that be able to handle 320x320 pixels (10ftx10ft) via splitting it to run on spi1 - 5x5 & spi2 - 5x5

board707, 2019-10-23:
Hi
There is something wrong in the arithmetics : )
First - 10x10 is not two 5x5, it is four
Second - 320x320 pixels is not 10x10 panels - it is 10x20 panels

The library definitely will not work with 100 (10x10) and 200 (10x20) panels - it is too much for the code. You can try 5x5 matrix - it was not tested, but hopefully will work

vtronicsautomation, 2019-10-23:
Actually i didn't mean 10x10 panels i ment 10ft x10ft - ( 32x10=320/16x20=320)  so I thought or running 5ftx5ft(32x5=  160/ 16x10= 160) Ok spi one alone as far as i have tested so far i was able to handle 4ftx2ft ( 4x32=128 : 4x16 = 24) works perfectly without any lags , beyond this resolution there is a little lag adds up for every 512 pixels added ie 32x16 per module

board707, 2019-10-23:
Hello
Did I understand correctly that you have already tested the new code and it works with 4ftx2ft  (4x4 panels, 16total) without significant lags?

vtronicsautomation, 2019-10-24:
Hi,
 yes with the new code it works fine so far to display static text but scrolling haven't tested. I'll be adding up some serial functionalities and split screen scrolling functionalities to my, to see if there is any lags or not. Hopefully it should work fine.

closed 2019-11-08 by board707

heartlesspra, 2020-11-20:
Hello , 
I am using dmd library for stm32f103c8t6 bluepill. 
But while compiling the code i am getting error "no matching function for call to SPIClass::SPIClass(int)".

Error showing at this line of your code

// We'll use SPI 2
SPIClass dmd_spi(2);

Please help me.
Which Arduino ide and stm32core  version are you using ????

board707, 2020-11-21:
Re: [board707/DMD_STM32] can't daisy chain more than 4 modules (#2)

Hello
This library for Roger Clark stm32 core
https://github.com/rogerclarkmelbourne/Arduino_STM32
and Arduino IDE some versions 1.6 - 1.8

WBR,
Dmitry Dmitriev

Friday, November 20, 2020, 11:51:22 AM, you wrote:

Hello ,
I am using dmd library for stm32f103c8t6 bluepill.
But while compiling the code i am getting error "no matching function for call to SPIClass::SPIClass(int)".
Error showing at this line of your code
// We'll use SPI 2
SPIClass dmd_spi(2);
Please help me.
Which Arduino ide and stm32core version are you using ????
—

Best regards,
 Dmitry                            mailto:ddt@jwee.ru