board707/DMD_STM32 #7: complete display screen  flickering while scrolling single line

kvar2019, 2020-11-24:
hello,
first of all thank you so much for writing such library for stm32.
I am having issue with the code which is written to display steady message at 2nd row and scrolling message at 1st row
 at the same time.
i tried this in two ways:
1 )  call draw string to  display the string and then call the marquee. it shifts all the display including static message.
`dmd.drawString(0,(DISPLAYS_DOWN - 2)* 16,"SCROLLING",10,GRAPHICS_NORMAL,0); // at 2nd row
   dmd.drawMarquee("SCROLLING",10,(32*DISPLAYS_ACROSS)-1,DISPLAYS_DOWN - 1)* 16); // at 1st row
   
   long prev_step =millis();
   boolean ret=false;
   while(!ret){
      if ((millis() - prev_step) > 50 ) {
       ret=dmd.stepMarquee(-1,0);
       prev_step=millis();
     }
   }`

2 )  call the marquee and then draw string to display the string . it shows static message on 2nd row while scrolling message at 1st row but the screen is flickering.   i know the reason it is flickering due to marquee is shifting whole screen and i am forcing to display the static message. but i dont know how to solve. please help me.
` dmd.drawMarquee("SCROLLING",10,(32*DISPLAYS_ACROSS)-1,DISPLAYS_DOWN - 1)* 16); //at 1st row
   
   long prev_step =millis();
   boolean ret=false;
   while(!ret){
      if ((millis() - prev_step) > 50 ) {
       ret=dmd.stepMarquee(-1,0);
       dmd.drawString(0,(DISPLAYS_DOWN - 2)* 16,"SCROLLING",10,GRAPHICS_NORMAL,0); //at 2nd row
       prev_step=millis();
     }
   }`

video showing two rows but p10 display connected across for testing purpose only.
first row message ABC
second row message "1" static and other scrolling "SCROLLING"
[image]

kvar2019, 2020-11-24:
[image]

board707, 2022-01-22:
Hi!
In the master branch of the library, it is impossible to display a marquee and a stationary line at the same time, because the marquee shifts the whole screen. 
But in dev-V2 branch you can switch off screen shifting by setting 
`use_shift = false;`
at DMD_STM32a.h (near 245 line)

closed 2022-01-26 by board707