board707/DMD_STM32 #8: Whay Arial_Black_16_ISO_8859_1 wont work?

Danilo277, 2021-03-20:
//////*CODE*///////
#include <DMD_STM32.h> 
//#include "st_fonts/SystemFont5x7.h"
#include "st_fonts/Arial_Black_16_ISO_8859_1.h"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"

// We'll use SPI 2   
SPIClass dmd_spi(1);

//Fire up the DMD library as dmd
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be PWM pin as PB1,PA8

// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

#define DMD_PIN_A PB11
#define DMD_PIN_B PB12
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PB10

DMD dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, DISPLAYS_ACROSS, DISPLAYS_DOWN, dmd_spi );

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(Arial_Black_16_ISO_8859_1);

// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b,(uint8_t*)&GlametrixLight12pt8b_rus,0x80,13);
DMD_GFX_Font GlametrixBold((uint8_t*)&GlametrixBold12pt7b,(uint8_t*)&GlametrixBold12pt8b_rus, 0x80, 13); 

/*-------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/
void setup(void)
{
  
   dmd.init();
   dmd.clearScreen( true );   //true is normal (all pixels off), false is negative (all pixels on)
   // set matrix brightness (0-255)
   dmd.setBrightness(80);
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/
void loop(void)
{
   
   dmd.selectFont(&UkrRusArial_F);
   
   const char *MSG = "Fran""\xe7""ais, ""\xd6""sterreich, Magyarorsz""\xe1""g";
   dmd.drawMarquee(MSG,strlen(MSG),(32*DISPLAYS_ACROSS)-1,0);
   long start=millis();
   long timer=start;
   while(1){
     if ((timer+30) < millis()) {
       dmd.stepMarquee(-1,0);
       timer=millis();
     }
   }
}

When I upload code to stm32 nothing is shown. Why?
P.S. I tried to make my own font but the same thing happens.
Thanks in advance.

Danilo277, 2021-08-08:
***UPDATE***
if you adjust in DMD_Font.cpp 
`bool DMD_Standard_Font::is_char_in(unsigned char c) {
	if (c < get_first() || c > get_last()) return false;
	else return true;
}`

to :

`bool DMD_Standard_Font::is_char_in(unsigned char c) {
	if (c < get_first() || c > get_last()) return true;
	else return true;
}`

you can use up to 255 character. 
My next investigation is why the number of characters is limited to 255 (I need LATIN1 and LATIN2... about 400 characters ).
If anyone has a suggestion please help. I think solving this "problem" will help many people.
Best regards!

board707, 2022-01-23:
@Danilo277 
There was an error in Arial_Black_16_ISO_8859_1.h font file, the maximum character index exceeded 255. I removed one character from the font and the problem went away.

Regarding to your second question -  the number of characters in font is limited to 255 because char indexes are bytes and must not exceed 255.

closed 2022-01-23 by board707