board707/DMD_STM32 #9: issues with 1/4 scan for rgb panel

bilalibrir, 2022-02-01:
I'm having a problem using an RGB (16*32) 1/4 Scan panel with this Library with no succes , the RGB panel has the snake pattern.
are you working to implement this feature or not? if not please can you do it and thanks in advance.
I tried to read the code but didn't understand most of it.
thankyou.

board707, 2022-02-02:
Hi,
thanks for interesting.
The current version of the library supports only straight-line pattern matrices. I have experience with matrices with other types of patterns, but there are too many types and it is difficult to understand them without having a matrix in hand.

bilalibrir, 2022-02-02:
I'll help from my side and i will try my best, i have the hardware and. i can already see the the pattern i believe it's the same in the attachment.
It only uses the A,B address wires just 2 wires.
Thank you 😊😊.
[image]

board707, 2022-02-03:
Hi @bilalibrir 
As I found out, I have a similar 1/4scan matrix, only the order of its segments is 0-1-3-2 instead of 0-1-2-3 like yours.
So I'll try, but it will take a few days.

bilalibrir, 2022-02-03:
Good luck 🤞🤞.
And thank you for your help and time.
What can i do to help from my side. If you need to test code or anything, i'm happy to help.

bilalibrir, 2022-02-03:
hi again,
i think this may help.
thanks to Ruud he modified RGBmatrixPanel to work with 32*16 1/4S matrixs.
see the attached link .
[https://forum.arduino.cc/t/p10-32x16-led-panel-1-4-scan-now-works-with-adafruit_gfx/483621](url)

bilalibrir, 2022-02-04:
helllo,
update, after trying a different library "PxMatrix" with an arduino Uno and my panels this is what i find:

1- my panel uses STRAIGHT Mux pattern and need's all A,B,C,D pins to be connected (it's not a BINARY mapping of outputs A,B,C,D). you have to set pin A low to send data for the first Row and pin B after that, then pin C, then pin D.

2- for the Scan pattern my panels uses ZAGGIZ pattern , is this library the have a differant set of patterns :
( display.setScanPattern(x) where x={LINE, ZIGZAG,ZZAGG, ZAGGIZ, WZAGZIG, VZAG, ZAGZIG, WZAGZIG2}.)

here is a link to the library [https://github.com/bilalibrir/PxMatrix](url).

I think that we can learn alot form this library.
thank you for your time.

board707, 2022-02-04:
I know this library. 
As I understand, the PxMatrix library worked with your matrix, so my help is no longer needed?

bilalibrir, 2022-02-05:
hello @board707 ,
I still need your help, because I want to use an STM32 on my project.
i'm trying on my side but i don't have the skills and i don't understand the math behind, i'm still learning but i fell like my head is going to explode. so **your help is really appreciated**.

this is what i did yesterday, i modified the DMD_RGB.h (dev-V2) , line 653-654:
`
template<int COL_DEPTH>
class DMD_RGB<RGB32x16plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(4, mux_list, _pin_nOE, _pin_SCLK, pinlist,    //bilal
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 4, 32, 16)           // bilal
	{}

};
`

and the generate_muxmask() , in DMD_RGB.cpp (dev-V2) :
`
void DMD_RGB_BASE::generate_muxmask() {
#define set_mux_ch_by_mask(x)  ((uint32_t) x)
#define clr_mux_ch_by_mask(x)  (((uint32_t)x) << 16)
	
	/*
	for (uint8_t i = 0; i < nRows; i++)
	{
		mux_mask2[i] = 0;
		for (uint8_t j = 0; (1 << j) < nRows; j++)
		{
			if (i & (1 << j))
			{
				mux_mask2[i] |= set_mux_ch_by_mask(mux_ch_masks[j]);
			}
			else
			{
				mux_mask2[i] |= clr_mux_ch_by_mask(mux_ch_masks[j]);
			}
		}
	}
	mux_mask2[nRows] = mux_mask2[0];
	 */
	for (uint8_t i = 0; i < nRows; i++)     
	{
		mux_mask2[i] = 0;
		for (uint8_t j = 0;  j < nRows; j++) 
		{
			if (i ==  j)
			{
				mux_mask2[i] |= clr_mux_ch_by_mask(mux_ch_masks[j]);    //low
			}
			else
			{
				mux_mask2[i] |= set_mux_ch_by_mask(mux_ch_masks[j]);    //high
			}
		}
	}
	mux_mask2[nRows] = mux_mask2[0];
	
}
`
to output a STRAIGHT muxmask, and i think that it works but i have nothing showing on the panel, no output on the RGB's outputs.

[image]

thank you for your time and help.

board707, 2022-02-05:
Ok
your fix of the generate_muxmask() seems to be correct
However, you do not need to edit existing templates, it is better to create a new one.

Besides that, I found a few minor bugs that need to be fixed before running your matrix.
So, please wait a bit, I will post the corrected code and we will continue

board707, 2022-02-05:
Hi,
I just uploaded the corrected code to dev-V2, please update the library from github.

Then replace the DMD_RGB.h in your copy of the library with file from the archive attached to the message. In this file, I have added a new template for your matrix.

However, all of the above only fixes the problem with mux pattern.
To change the date pattern, we first need to understand how the pixels on your matrix are connected. To investigate this, I will ask you to run test code **bilalibrir_test.ino** from the archive and make a video how it work in your matrix.
Thans for your help.
[bilalibrir_test.zip](https://github.com/board707/DMD_STM32/files/8007737/bilalibrir_test.zip)

bilalibrir, 2022-02-05:
Helllo @board707, 
I will try this code Tonight, and update you with everything.
Thank you for your help.

bilalibrir, 2022-02-05:
hello @board707 ,
you are a **genius**, you made it look so easy.

https://user-images.githubusercontent.com/86444589/152657496-ec7d5fa5-2cd9-4314-8173-554103113e36.mp4

thank you for your time and help.

board707, 2022-02-05:
Hi, my congratulations.
It looks like you have the same matrix as mine.
Wait a little more, I will send you the corrected code for it.

board707, 2022-02-05:
Tried to add support for your data pattern, check it.
Replace file DMD_RGB.h in the library with one from attachment. 
[bilalibrir_2.zip](https://github.com/board707/DMD_STM32/files/8008982/bilalibrir_2.zip)

bilalibrir, 2022-02-06:
hi, this is the new video with [bilalibrir_2.zip](https://github.com/board707/DMD_STM32/files/8008982/bilalibrir_2.zip)
code:
https://user-images.githubusercontent.com/86444589/152661399-a73c2560-7063-49d3-bb43-34a522de4425.mp4

thank you for your time and help.

bilalibrir, 2022-02-06:
hello, it didn't work for me, i will upload a new video.

board707, 2022-02-06:
try this
[bilalibrir_3.zip](https://github.com/board707/DMD_STM32/files/8009624/bilalibrir_3.zip)

bilalibrir, 2022-02-06:
Morning @board707 ,
it works perfectly the pattern is good. from top to bottom, from left to right. perfect.
i tried the test example, it works to perfectly. great
i will try tonight (DISPLAYS_ACROSS 2*1) and (DISPLAYS_ACROSS 4*1). and report back.
and maybe a 2*2 zigzag.

**Thank you for your time and help, I really appreciate it.**

board707, 2022-02-06:
>  i will try tonight (DISPLAYS_ACROSS 2_1) and (DISPLAYS_ACROSS 4_1). and report back. and maybe a 2*2 zigzag.

Sorry, I'm afraid this fix will work only with one matrix, connecting several panels with non-standard pattern is a separate complex task. I will try to finish the code if you are ready to test many times.

>  I really appreciate it.
Thanks ^) you can add a star to the repo if you like it

bilalibrir, 2022-02-06:
Hi again,
I am always ready to help, for as many times it takes. 1000 time's No problem. With an open heart.
The only thing is than i have to work during the day, so i'm free every evening starting from 8 PM . And i don't work on the Friday's.
Otherwise I'm all ready to help.
Again thank you for your help and thank you for your time.

board707, 2022-02-06:
bilalibrir, Thank you for your help.
> i have to work during the day, so i'm free every evening starting from 8 PM . 

8 PM in what time zone? ^) 
I am in Moscow, Russia at GMT+3

bilalibrir, 2022-02-06:
Sorry, i'm from Algeria +1GMT.

bilalibrir, 2022-02-06:
hello, hope you are doing fine @board707, do you have any updates?

board707, 2022-02-06:
hello, thanks, I am fine
Have you tried connecting multiple matrices? What is the result?

bilalibrir, 2022-02-06:
hello, yes I did, it did not work, i will upload a video.

bilalibrir, 2022-02-06:
https://user-images.githubusercontent.com/86444589/152704442-eba992c8-eadc-456b-93a5-8fb84562d88e.mp4

https://user-images.githubusercontent.com/86444589/152704496-61d8c98e-baf8-4be0-a226-e993975a597a.mp4

board707, 2022-02-06:
Thanks for video.
For now I don't have a solution I have to think....

bilalibrir, 2022-02-06:
sorry for taking so long to upload the videos.
i can't upload more then 10Mb / file, so i had to use a video convertor.
take your time and thanks for your help.

board707, 2022-02-07:
Hi, 
Here is a new try
I'm not able to check this file with compiler right now, sorry for possible typos
[bilalibrir_4.zip](https://github.com/board707/DMD_STM32/files/8015003/bilalibrir_4.zip)

bilalibrir, 2022-02-07:
Hello , hope you are fine, I think that I found a bug in the RGB text example , i will make a video to show you, i'm not sure if it's just with my panel or for every one, you have to adjust the time to more then 200ms so you can see it.
And i will test the new code the soonest when i get home.
Thank you. Stay safe.

bilalibrir, 2022-02-07:
this is 2*1:

https://user-images.githubusercontent.com/86444589/152870040-d9aacfc5-2243-4b1a-bdf8-4d58f5922d69.mp4

and i don't have a long data cable to do a 2*2 in the correct way. sorry .

thank you for your time. and help.

board707, 2022-02-07:
Thank you for tests
Please, comment the whole procedure transform_XY() at template of your matrix (near 760-770 line of the DMD_RGB.h) and show the video

bilalibrir, 2022-02-07:
1*2:

https://user-images.githubusercontent.com/86444589/152873335-501f2da3-8c4f-4dd3-a75a-317a90a43b01.mp4

and 2*1:

https://user-images.githubusercontent.com/86444589/152873931-920629b5-d363-4c2f-ac20-cd72fdf0fb17.mp4

board707, 2022-02-07:
Thanks, I will think :)

bilalibrir, 2022-02-07:
this is a test for 2*2, sorry for the short cable:
https://user-images.githubusercontent.com/86444589/152876181-94223e98-7a35-4262-9152-669e8d80261e.mp4

take your time, thank you.

board707, 2022-02-08:
Good evening @bilalibrir 
Try this, please
 (i hope it will work with horizontal sets as 2*1, 3*1 ...)
[bilalibrir_6.zip](https://github.com/board707/DMD_STM32/files/8026404/bilalibrir_6.zip)

bilalibrir, 2022-02-08:
hello, sorry for the delay i just get home and tested the new code it work's perfectly,
for 1*1, 2*1, 3*1 and 4*1 all work's perfecty.

bilalibrir, 2022-02-08:
hello again , about the issues i noticed in the RGB example this is the video, and i thinks it's not the library code it's just the code example.

https://user-images.githubusercontent.com/86444589/153070682-e7507a1d-ea21-4e1b-b60d-8eacdb370c50.mp4

https://user-images.githubusercontent.com/86444589/153070791-80323de9-6525-45d4-8c17-66c5069d2398.mp4

board707, 2022-02-08:
> tested the new code it work's perfectly, for 1_1, 2_1, 3_1 and 4_1 all work's perfecty.

O, great. Please, try to test with vertical panel sets : 1x2

About RGB test issue - it is not a issue, this is test of dual buffering. 
Change
#define ENABLE_DUAL_BUFFER false
to true and see/

Thanks,
Dmitry

bilalibrir, 2022-02-08:
it didn't work for the vertical panel setup, sorry i will make a video for that.
for the RGB test code , the dual_buffer is already 'false' and you only see it when (case 0)  is runing:

board707, 2022-02-08:
sorry for too much tests...
I attached new one:
[bilalibrir_7.zip](https://github.com/board707/DMD_STM32/files/8027859/bilalibrir_7.zip)

>  the dual_buffer is already 'false' and you only see it when (case 0) is runing:

try to set it 'true'

bilalibrir, 2022-02-08:
https://github.com/board707/DMD_STM32/files/8027859/bilalibrir_7.zip
works fine for the 2*1.
not for 2*2.
this is the video for 2*2

https://user-images.githubusercontent.com/86444589/153085694-d4d5ca6b-67c9-4a23-9ea9-56008de83c84.mp4

:
you don't need to say sorry , i have no problem if it take's a **1000 times**, lol
**thank you for your time**. iknow that it's really hard to make code without hardware next to you.

bilalibrir, 2022-02-08:
i have tested the " #define ENABLE_DUAL_BUFFER true ", even worst the 6 first rows alwas off , and this is just for case0.
everything else is okay for case1 and case2 ....

board707, 2022-02-08:
Thanks for your help,
see you tomorrow

bilalibrir, 2022-02-08:
thanks for your help, see you tomorrow

bilalibrir, 2022-02-09:
I just tested the 2*2 and 1*4 (vertical) configuration and it didn't work, sorry.
 but it works perfect for horizontal 4*1 configuration
this video is for 2*2 configuration:
https://user-images.githubusercontent.com/86444589/153275092-09d46945-8863-4ea8-946f-49d364b08086.mp4

for the RGB test code i find that if i modify the line 181:

`                    if (step[i] != 1) dmd.drawFilledBox(0, 0, 5, dmd.height() - 1, GRAPHICS_INVERSE);

to 
`                    if (step[i] != 1) dmd.drawFilledBox(0, 0, -1, dmd.height() - 1, GRAPHICS_INVERSE); /// solved the problem

`
solve the problem for the 6 first flickering rows.
thank you.

bilalibrir, 2022-02-09:
1. for the last video it 2 x 2 configuration , the connection is 1 and 2 the two upper one's , 3 and 4 the for bottom one's.
sorry for short cable between 2 and 3 panel. ( i jest made a new long cable) yipiiii. lol.
2. I did not change the test code. only this lines (#define DISPLAYS_ACROSS 2          #define DISPLAYS_DOWN 2 ).
3. is that really a bug in the RGB test code?? or just for me? i don't understand!!

https://user-images.githubusercontent.com/86444589/153275092-09d46945-8863-4ea8-946f-49d364b08086.mp4

bilalibrir, 2022-02-09:
[image]

board707, 2022-02-09:
And one more stupid question: do you sure that all your matrices has the same data pattern?

>     3. is that really a bug in the RGB test code?? or just for me? i don't understand!!

No , it is not a bug and in any case is not your fault. 
The problem is that I can't find the correct coordinate transformation for these matrices.

board707, 2022-02-09:
sorry, i have another idea
try this
[bilalibrir_9.zip](https://github.com/board707/DMD_STM32/files/8036414/bilalibrir_9.zip)

bilalibrir, 2022-02-09:
> And one more stupid question: do you sure that all your matrices has the same data pattern?
it is not a stupid question, everything is possible, but they are all the the same  model number and the same pcb and the same manufacturing date.
i already tested them with a U62_plus module they work fine.
with you code they work fine too, in the horizontal configuration. perfactly
soo..??
I really appreciate all the work and time you put in. so thank you very much.
i will try the new code in secondes .

bilalibrir, 2022-02-09:
ohhhh yes baby hhhhhh
very very very close.
i will make a video give me a minute
very close

bilalibrir, 2022-02-10:
hello @board707  , i hope you are doing okay.
do you have any new ideas you want to try?
thank you.

board707, 2022-02-10:
Hello
Thanks but there will be no tests today. May be tomorrow.
Probably I misunderstand the scheme of panels with a complex pattern...

bilalibrir, 2022-02-10:
Thank you for all, and tomorrow is the weekend here so i will be free all the day.
Take care of your self, and have a lovely evening.

bilalibrir, 2022-02-11:
good morning @board707 i hope you are doing well,
i did some modification on the last code (v9), i think that is working perfectly, i did test it for 1by2, 1by3 and 1by4 configuration , it works fine and this is the code:

`		if ((y >= this->DMD_PIXELS_DOWN) and 1) {
			//x_0 = (this->WIDTH - 32) * (y / this->DMD_PIXELS_DOWN); // this shifted is back by 32 pixel
			//x_0 = (this->WIDTH - 64) * (y / this->DMD_PIXELS_DOWN); // you will get a 0 all the time 0/x=0
			x_0 =  (y / this->DMD_PIXELS_DOWN);
			//x_0 =  x_0 - 1;
			x_0 =  x_0 - ((((y/16)+1)*2)/2)+1; // this add's 1 or 2 or 4 or 8 .... shift dipanding of the panel y level.
			y_0 = y - y % this->DMD_PIXELS_DOWN;
			x += x_0;
			y -= y_0;
		}

		//if (x_0 != 0 ) {   // i had to modifie this too .
		if ((x_0 != 0 )or 1) {
			x -= x_0;
			y += y_0;
		}
`
i don't know how but it works, do you see or understand the logic of this and how you can improve the code, you are the math genius after all thanks for your time and help.

bilalibrir, 2022-02-11:
https://user-images.githubusercontent.com/86444589/153578669-93abefbd-d06d-4920-afac-2a4adaa72503.mp4

bilalibrir, 2022-02-11:
https://user-images.githubusercontent.com/86444589/153582188-98ab3a04-98ee-42c5-9398-2bbe90df1b59.mp4

thank you soo much.

board707, 2022-02-11:
Hi, please test this.
I don't think it will work, but I need to see the picture on 2x2
[bilalibrir_a1.zip](https://github.com/board707/DMD_STM32/files/8048894/bilalibrir_a1.zip)

bilalibrir, 2022-02-11:
this is the finel code that works for all configuration tested ( 1by1 to 1by4), and (2by1 to 4by1) and 2by2. all tested.

`protected:
	void transform_XY(int16_t& x, int16_t& y) override {
		// coordinate conversion for 4s

	// first run base transform
		DMD::transform_XY(x, y);
		int16_t x_0 = 0;
		int16_t y_0 = 0;
		int16_t eff_width = this->DisplaysTotal * this->DMD_PIXELS_ACROSS;
			x_0 =  (y / this->DMD_PIXELS_DOWN) - ((((y/16)+1)*2)/2)+1 ;
			y_0 = y - y % this->DMD_PIXELS_DOWN;
			x += x_0;
			y -= y_0;
		
		uint8_t y_shift = 0;
		if (y >= this->pol_displ) {
			y = y % this->pol_displ;
			y_shift = this->pol_displ;
		}
		int16_t x_t = (((x / 8) * 2) % (eff_width / 8)) * 8;

		if (y < this->nRows) {
			y = 2 * y + x / (eff_width / 2) + y_shift;
			x = x_t + (7 - x % 8);
		}
		else {
			y = 2 * (y - this->nRows) + x / (eff_width / 2) + y_shift;
			x = x_t + 8 + x % 8;
		}
			x -= x_0;
			y += y_0;
	}
`

[2 more replies]

closed 2022-02-11 by bilalibrir