board707/DMD_STM32 #12: RGB p10 MAtrix Panel R1 G1 Pins

miralay001, 2022-06-24:
hi 

I couldn't run the panel with R1 G1 B1 R2 G2 B2 input pins, how should I go about it? can you help me?

[image]
[image]

board707, 2022-06-27:
Hi
please be more specific about your problems with pins. Perhaps [RGB Connecting] (https://github.com/board707/DMD_STM32/wiki/Connecting-for-RGB(rus)) Wiki page will be useful (it is in Russian, but can be easily translated by Google)

As far as you can see in the photo, you have p10 32x16 4scan outdoor panels. Such matrices often have non-standard wiring of LEDs, which must be managed individually. If you are willing to spend some time testing your panels, then I can help you find the right scan type for your matrices.

miralay001, 2022-06-27:
[RGB Connecting](https://github.com/board707/DMD_STM32/wiki/Connecting-for-RGB(rus)) I looked at the sample link page, the panel connections there start with R0 G0 My panel starts with R1 G1 I connected all the other pins as in the example link, but I could not get a picture from the panel. I am using your previous library as a single color without any problems. I could not run this library.

my panel starts with R1 G1 Do you think it's a problem?

board707, 2022-06-27:
> my panel starts with R1 G1 Do you think it's a problem?

no,
 It doesn't matter. It's just a different notation, the first data pin may be indexed as R0 or R1
so R0G0B0 R1G1B1 = R1G1B1 R2G2B2

Please connect as above and try to run example **dmd_rgb_pattern_test.ino**

miralay001, 2022-06-27:
i will go through the links and test the dmd_rgb_pattern_test.ino example i will let you know

miralay001, 2022-06-27:
I tested the connections with socket and solder, I also used the sample test, no results, I activated ALTERNATIVE_DMD_TIMERS from the dmd_config file, I replaced the OE pin with PB9 and the result was like this

https://user-images.githubusercontent.com/108148338/175953827-04ee8672-c9e7-4980-8c4d-588e9f649faf.mp4

board707, 2022-06-27:
Please test with **RGB32x16plainS4_DIRECT** pattern? so
`DMD_RGB <RGB32x16plainS4_DIRECT, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);`

miralay001, 2022-06-27:
I tested it with RGB32x16plainS4_DIRECT model, the result is like this, I used the same panel in pxmatrix, no problem, I want to use your library with stm32, can my connections be a mismatch in the right pins?

https://user-images.githubusercontent.com/108148338/176050669-47076344-36f3-4aef-b053-19d4ca614b4e.mp4

board707, 2022-06-27:
> I used the same panel in pxmatrix, no problem

What settings did you use in PXmatrix for your panels (scan, scan pattern, mux pattern) ? If you could send an example of code that is known to work with your matrices on PXMatrix,  that might be helpful.

board707, 2022-06-28:
Thanks for code.
This is somewhat strange, such a config should have given more reasonable behavior for the <RGB32x16plainS4> mode in my library.
What stm32 board are you using? How do you upload firmware? (STM32_bootloader, DFU or StLink?)

miralay001, 2022-06-28:
> Kod için teşekkürler. Bu biraz garip, böyle bir yapılandırmanın kütüphanemdeki RGB32x16plainS4 modu için daha makul bir davranış vermesi gerekirdi. Hangi stm32 kartını kullanıyorsunuz? Firmware'i nasıl yüklersiniz? (STM32_bootloader, DFU veya StLink?)

arduino version i am using 1.8.13
I'm Programming over USB TTL
stm32 card link http://dan.drown.org/stm32duino/package_STM32duino_index.json

[image]

I just installed from scratch on another computer in your library 
the result did not change

https://github.com/adafruit/Adafruit-GFX-Library/releases/tag/1.7.0
https://github.com/rogerclarkmelbourne/Arduino_STM32

miralay001, 2022-06-28:
> Hello, I think that you have to use S2 mux binary not a direct one.
 
hello
I'm trying right away

board707, 2022-06-28:
> stm32 card link http://dan.drown.org/stm32duino/package_STM32duino_index.json
> 
> [image]

sorry, the board STM32f103_C6 is not supported by the library yet, just in plans. 

For c6 board a completely different config is needed because it does not have TIMER4. In addition, the flash size of c6 is relatively small, even simple examples cannot be loaded into the board without significant optimization of the stm32 core

miralay001, 2022-06-28:
[image]

but my card is not STM32f103_C6

my card is STM32f103_C8T6

board707, 2022-06-28:
>I will try and reconnect with my current STM32f103_C8T6 card.

Please select "Generic_STM32F103C series" board in Arduino IDE:
https://microkontroller.ru/wp-content/uploads/2022/03/select-stm32f103c-as-microcontroller-board-in-arduino-ide.png

miralay001, 2022-06-28:
I tried again with stm32f103c8tb
it just keeps getting brighter

maybe the stm32duino library I am using is wrong?

http://dan.drown.org/stm32duino/package_STM32duino_index.json

[image]
.

https://user-images.githubusercontent.com/108148338/176287802-3f84698c-a9e2-4896-9d00-ba542f0dafc2.mp4

board707, 2022-06-28:
> maybe the stm32duino library I am using is wrong?
> 
> http://dan.drown.org/stm32duino/package_STM32duino_index.json

Yes, the package you using is wrong. The only STM32 Arduino package supported is Roger Clarks's repo https://github.com/rogerclarkmelbourne/Arduino_STM32

closed 2022-07-09 by board707