board707/DMD_STM32 #13: Port library for esp8266 and esp32

akshaythoke9116, 2022-09-05:
Hello,
I want to use esp8266 and esp32 for monochrome display. please port or tell me how to port the library to esp

board707, 2022-09-05:
The port of the library to esp32 is in the plans, but not in the near future. Port to esp8266 is not planned because the platform limitations.

For ESP you can try other libraries
[Adafruit_Protomatter](https://github.com/adafruit/Adafruit_Protomatter)
[SmartMatrix](https://github.com/pixelmatix/SmartMatrix)
[PxMatrix](https://github.com/2dom/PxMatrix)
[ESP32-HUB75-MatrixPanel-I2S-DMA](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA)

akshaythoke9116, 2022-09-06:
I want to use unicode with the library for P10 Display.
There is no sample code or guidance for using unicode in the all above libraries for P10 Display using ESP8266.
If possible guide me for it.

board707, 2022-09-06:
Unfortunately, I can't offer any advice here. After all, I created this library exactly to solve problem with non-Latin characters in panel.
I don't know how to do it in other libraries.

akshaythoke9116, 2022-09-06:
Ok thanks

board707, 2022-09-17:
RP2040 board support added in version 0.9.0

akshaythoke9116, 2022-10-08:
I have tried your library for P10 Display using STM32 but it is not working can you help in this case

board707, 2022-10-17:
Please start new issue with detail explanation of the problem.

closed 2022-12-08 by board707