board707/DMD_STM32 #16: PICO P4 Module

maxmurugan, 2022-11-29:
I TRY YOUR CODE TOP-LINE SCROLLING AND BOTTOM-LINE STRING MESSAGE
HOW TO REMOVE BACKGROUND COLOUR AND FLICKERING?
PLEASE SUGGEST A CODE HERE.

void loop(void)
{

    // create foreground colors
    uint16_t col[] = {
        dmd.Color888(255,0, 0), // red
        dmd.Color888(0, 255, 0), // green
        dmd.Color888(0, 0, 255)  // blue

    };
    uint16_t bg = 0;  // background - black
    int col_cnt = 3;   // color count

    // text
    char s[] = "          MURUGAN CHENNAI";
    // transcode message to UTF for use with GFX fonts
    char k[30];
    
    utf8_rus(k, (const unsigned char*)s);
    char* m = s;
    // select standard font
    dmd.selectFont(&UkrRusArial_F);
    // set text foreground and background colors
    dmd.setTextColor(col[0], bg);
    // shift steps in pixels for running text (positive - shift right, negative - left)
    // running text shift interval
    uint16_t interval = 25;
    long prev_step = millis();
    dmd.setBrightness(200);

    // Cycle for tests:
    // -- running texts moving at x and y axis with single and double speed
    // -- vertical scrolling message
    while (1) {
        if ((millis() - prev_step) > interval) 
        {
                dmd.drawString(8,16,"12:34:56",8,col[1]);  //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
                
                if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
                    dmd.clearScreen(true);
                    dmd.setTextColor(col[0], bg);
                    dmd.drawMarqueeX(m, 0,0);
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
            prev_step = millis();

        }
    }
}

maxmurugan, 2022-11-29:
https://user-images.githubusercontent.com/4962604/204518187-78f45579-383a-4eb9-8c47-9917a37583f5.MOV

board707, 2022-11-29:
Hi
Please show full sketch rather than loop() procedure only.
Your video show the empty screen.

maxmurugan, 2022-11-30:
https://user-images.githubusercontent.com/4962604/204675981-bd867f03-8ea2-4994-9b98-3ca3cb1ad8b2.mp4

void loop(void)
{

    // create foreground colors
    uint16_t col[] = {
        dmd.Color888(255,0, 0), // red
        dmd.Color888(0, 255, 0), // green
        dmd.Color888(0, 0, 255)  // blue

    };
    uint16_t bg = 0;  // background - black
    int col_cnt = 3;   // color count

    // text
    char s[] = "          MURUGAN CHENNAI";
    // transcode message to UTF for use with GFX fonts
    char k[30];
   
    utf8_rus(k, (const unsigned char*)s);
    char* m = s;
    // select standard font
    dmd.selectFont(&UkrRusArial_F);
    // set text foreground and background colors
    dmd.setTextColor(col[0], bg);
    // shift steps in pixels for running text (positive - shift right, negative - left)
    // running text shift interval
    uint16_t interval = 20;
    long prev_step = millis();
    dmd.setBrightness(200);

    // Cycle for tests:
    // -- running texts moving at x and y axis with single and double speed
    // -- vertical scrolling message
    while (1) {
        if ((millis() - prev_step) > interval)
        {
                //dmd.drawString(8,16,"12:34:56",8,col[1]);  //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231

                if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
                    dmd.clearScreen(true);
                    dmd.setTextColor(col[1], bg);
                    dmd.drawMarqueeX("KAMALCO", 0,20);
                    dmd.drawMarqueeX(m, 0,20);
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
            prev_step = millis();

        }
    }
}

maxmurugan, 2022-11-30:
please help me

board707, 2022-11-30:
Sorry, your video still unavailable for me.
And you still post the part of code. I need a whole sketch to test it in my side.

maxmurugan, 2022-11-30:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
[... 113 lines cut ...]
               }
               if(inByte == '[')Sop_packet =1;
              
            }
        if ((millis() - prev_step) > interval)
        {
          //#######################  add string get flicker 
                dmd.drawString(8,20,"12:34:56",8,col[1]);  //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
           //##########################

                if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
                    dmd.clearScreen(true);
                    col_ptr++;
                    if (col_ptr >= col_cnt) col_ptr = 0;
                    dmd.setTextColor(col[col_ptr], bg);
                    //dmd.setTextColor(col[1], bg);
                    dmd.drawMarquee(m, strlen(m), dmd.width() - 1,(dmd.height()-16 > 0)? 0 : 0, 1);
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
            prev_step = millis();

        }
    }
}

maxmurugan, 2022-12-01:
top line scrolling and bottom line  to draw string

maxmurugan, 2022-12-01:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/FreeMono12pt7b.h"
#include "gfx_fonts/FreeMono24pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
[... 118 lines cut ...]
              
            }
        if ((millis() - prev_step) > interval)
        {
          //#######################  add string get flicker 
                dmd.drawString(8,0,"12:34:56",8,col[1]);  //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
           //##########################

                if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
                    dmd.clearScreen(true);
                    col_ptr++;
                    if (col_ptr >= col_cnt) col_ptr = 0;
                    dmd.setTextColor(col[col_ptr], bg);
                    //dmd.setTextColor(col[1], bg);
                    dmd.drawMarquee(m, strlen(m), dmd.width() - 1,(dmd.height() > 0)? 15 : 0, 1);
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
            prev_step = millis();

        }
    }
}

https://user-images.githubusercontent.com/4962604/205016194-ff086bf6-b582-45b2-95b0-5b7e1ee3bd6f.mp4

maxmurugan, 2022-12-01:
i have select big font 32 pixel line gosting issue created and how to solve this issue

please help me

board707, 2022-12-01:
Thank you for code and video.
 I will test it.

board707, 2022-12-01:
Please test ether the flickering disappear if move marquee by two pixels in time:

`if ((millis() - prev_step) > interval)
    {
      //#######################  add string get flicker 
            dmd.drawString(8,0,"12:34:56",8,col[1]);  //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
       //##########################

            if (dmd.stepMarquee(-2, 0) & 1) { // if text is reached screen bounds
                dmd.clearScreen(true);
                col_ptr++;
                if (col_ptr >= col_cnt) col_ptr = 0;
                dmd.setTextColor(col[col_ptr], bg);
                //dmd.setTextColor(col[1], bg);
                dmd.drawMarquee(m, strlen(m), dmd.width() - 1,(dmd.height() > 0)? 15 : 0, 1);
            }
            // output mem buffer to matrix
            dmd.swapBuffers(true);
        prev_step = millis();

    }
}`

maxmurugan, 2022-12-02:
dear sir,
I am trying to your code flickering removed but still background  show on the display

how to remove the background color?
and also explain different scan method

board707, 2022-12-02:
Please show the video with last code

maxmurugan, 2022-12-03:
/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/FreeMono12pt7b.h"
#include "gfx_fonts/FreeSerif9pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER true

#if (defined(STM32F1) || defined(STM32F4))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 13
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
[... 122 lines cut ...]
           }
           if(inByte == '[')Sop_packet =1;
          
        }
        //####################### add string get flicker
         dmd.drawString(8,0,"12:34",5,col[1],0,0,0); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
//##########################
if ((millis() - prev_step) > interval)
{
      
        if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
            dmd.clearScreen(true);
            col_ptr++;
            //if (col_ptr >= col_cnt) col_ptr = 0;
            //dmd.setTextColor(col[col_ptr], bg);
            dmd.setTextColor(col[0], bg);
            dmd.drawMarquee(m, strlen(m), dmd.width()-10,(dmd.height() > 0)? 16 : 0, 1);
        }
        // output mem buffer to matrix
        dmd.swapBuffers(true);
    prev_step = millis();

}
}
}

maxmurugan, 2022-12-03:
https://user-images.githubusercontent.com/4962604/205411455-a6c36ad3-a484-4a1e-9000-4db6df200e66.MOV

/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/FreeMono12pt7b.h"
#include "gfx_fonts/FreeSerif9pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER true

#if (defined(STM32F1) || defined(STM32F4))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 13
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
[... 120 lines cut ...]
           }
           if(inByte == '[')Sop_packet =1;
          
        }
        //####################### add string get flicker
         dmd.drawString(8,0,"12:34",5,col[1],0,0,0); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
//##########################
if ((millis() - prev_step) > interval)
{
      
        if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
            dmd.clearScreen(true);
            col_ptr++;
            //if (col_ptr >= col_cnt) col_ptr = 0;
            //dmd.setTextColor(col[col_ptr], bg);
            dmd.setTextColor(col[0], bg);
            dmd.drawMarquee(m, strlen(m), dmd.width()-10,(dmd.height() > 0)? 16 : 0, 1);
        }
        // output mem buffer to matrix
        dmd.swapBuffers(true);
    prev_step = millis();

}
}
}

maxmurugan, 2022-12-03:
Using a different color for the two lines, the background color will show

How to remove background color in the first row?

board707, 2022-12-03:
Looks like this is a bug
Thanks for the info

maxmurugan, 2022-12-03:
your library add different scan method like 
p6  - 32x32 -1/8 scan
p6 - 32x32 -1/8 scan sm5266 column driver ic

maxmurugan, 2022-12-03:
i am  waiting for your future update

maxmurugan, 2022-12-03:
please send link for last version is alpha v.0.9.0.

board707, 2022-12-05:
> p6 - 32x32 -1/8 scan
Why do you ask about it?
As far I see in your video, you use 64x32 16scan panels, isn't it?

If you interesting in the other types of panels - let me know.
For panels with 32 pix height you usually should use 1/16 scan mode.
Panels with 32x32 -1/8 scan are the different case and not supported "from the box", it should be explicitly added to the library first.

maxmurugan, 2023-01-01:
https://user-images.githubusercontent.com/4962604/210168817-3a9fb1bc-e7d5-40b1-9d2c-3eaa69a5b0a3.MOV

 /*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include <Wire.h>
#include <EEPROM.h>
#include "DMD_RGB.h"
#define RGB32x32plainS8 34 // 32x32 1/8
//###########################################################################################

template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
  {}
  protected:
  uint16_t get_base_addr(int16_t x, int16_t y) override {
  this->transform_XY(x, y);
  uint8_t pol_y = y % this->pol_displ;
  x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
  uint16_t base_addr = (pol_y % this->nRows) * this->x_len + (x / this->DMD_PIXELS_ACROSS) * this->multiplex * this->DMD_PIXELS_ACROSS;
  if (pol_y / this->nRows) base_addr += x % this->DMD_PIXELS_ACROSS ;
  else base_addr += x % this->DMD_PIXELS_ACROSS + this->DMD_PIXELS_ACROSS;
  return base_addr;
}

};
//#######################################################
#include "st_fonts/UkrRusArial14.h"
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/FreeMono12pt7b.h"
#include "gfx_fonts/FreeSerif9pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER true

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1

DMD_RGB <RGB32x32plainS8, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic charsFreeMonoBold12pt7b
// 32 pixel DMD_GFX_Font GlametrixL((uint8_t*)&FreeMono24pt7b,24 );
DMD_GFX_Font GlametrixL((uint8_t*)&FreeSerif9pt7b,12 );

/*--------------------------------------------------------------------------------------
UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {
  
  uint16_t i, j;
  for ( i =0, j =0; src[i]; i++) {
   if ((src[i] == 0xD0 )&& src[i+1])  { dest[j++] = src[++i] - 0x10;}
    else if ((src[i] == 0xD1 )&& src[i+1]) {dest[j++] = src[++i] + 0x30;  }
    else dest[j++] = src[i];
  }
  dest[j] ='\0';
  return j;
}

/*--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/
void loop(void)
{

// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255) // blue

};
uint16_t bg = 0; // background - black

// text
char s[] = "TEST";
// transcode message to UTF for use with GFX fonts
char k[100];
utf8_rus(k, (const unsigned char*)s);
char* m = s;
dmd.selectFont(&GlametrixL);
uint16_t interval = 35;
long prev_step = millis();
dmd.setBrightness(128);
uint8_t col_ptr = 0;

while (1)
{

    //####################### add string get flicker
         dmd.setTextColor(col[2], bg);  //-- blue color
     dmd.drawString(0,0,"123",5,col[2],0,0,0); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
//##########################
if ((millis() - prev_step) > interval)
{

    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
        dmd.clearScreen(true);
        col_ptr++;
        //if (col_ptr >= col_cnt) col_ptr = 0;
        //dmd.setTextColor(col[col_ptr], bg);
        dmd.setTextColor(col[0], bg);    // --red color
         dmd.drawMarquee(m, strlen(m), dmd.width()-10,(dmd.height() > 0)? 16 : 0, 0);
    }
    // output mem buffer to matrix
    dmd.swapBuffers(true);
prev_step = millis();
}
}
}

maxmurugan, 2023-01-01:
top line set blue color
bottom line scrolling set red color

both are showing blue color
end of scrolling loop get top line blink (clear screen)

maxmurugan, 2023-01-01:
More then one panel  scrolling support one screen only

board707, 2023-01-01:
Sorry, I an on vacation now.
Happy New Year

maxmurugan, 2023-01-01:
Enjoy your vacation
thanks

maxmurugan, 2023-01-22:
hi sir ,
please solve the issue
full color scrolling only one support ,please see attached video

board707, 2023-01-22:
Hi
You have inserted a several videos in the topic, please indicate which is regarded to the issue

board707, 2023-01-22:
Change your while(1) loop in the loop() procedure according to this:
```
while (1)
  {
    //##########################
    if ((millis() - prev_step) > interval)

    { dmd.setTextColor(col[0], bg);    // --red color
      if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
        dmd.clearScreen(true);
        col_ptr++;
        //if (col_ptr >= col_cnt) col_ptr = 0;
        //dmd.setTextColor(col[col_ptr], bg);

        dmd.drawMarquee(m, strlen(m), dmd.width() - 10, (dmd.height() > 0) ? 16 : 0, 0);
      }
      // output mem buffer to matrix

      //####################### add string get flicker
      dmd.setTextColor(col[2], bg);  //-- blue color
      dmd.drawStringX(0, 0, "123", col[2]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      dmd.swapBuffers(true);

      prev_step = millis();
    }
  }
```

The rest code still without changes.

maxmurugan, 2023-01-26:
https://user-images.githubusercontent.com/4962604/214894881-b23d0f38-8ac9-4d36-a1f7-581dcbef922a.MOV

After add your code working fine
thanks a lot.

maxmurugan, 2023-01-26:
https://user-images.githubusercontent.com/4962604/214895747-f81874aa-e250-48b6-8607-1482e7eb813e.mp4

After add your code working fine
thanks a lot

board707, 2023-01-26:
Thanks for your cooperation

closed 2023-01-26 by board707