board707/DMD_STM32 #23: PI PICO P6_32x32 1/8 scan module

maxmurugan, 2022-12-19:
[image]
[image]
[image]

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
 // Fonts includes
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
char a[10];
char b[10];
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB32x16plainS8 , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
  UTF8 char recoding

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/
    // create foreground colors
uint16_t col[] = {
        dmd.Color888(255,0, 0), // red
        dmd.Color888(0, 255, 0), // green
        dmd.Color888(0, 0, 255),  // blue
        dmd.Color888(0, 255, 255),  // blue
        dmd.Color888(0, 0, 0),  // blue
        };

void setup(void)
{

    // initialize DMD objects
  dmd.init();
  Serial.begin(9600);
  Serial.println("PICO P4 RTC");
  delay(1000);
   
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{
       
    dmd.selectFont(&Arial_black);
    dmd.setBrightness(255);
    dmd.setTextColor(00, 0);
    while (1) 
    {
      

              dmd.drawString(0,0,"PLAN",4,col[1]);
              dmd.drawString(0,8,"ACTAL",6,col[2]);
              dmd.drawString(0,16,"GAP",3,col[0]);
              dmd.drawString(0,24,"EFFY",4,col[1]);
     }
    }

maxmurugan, 2022-12-19:
sir, I try the P6 Module using the p8 scan top 8 pixels working another 8-pixel showing the same message

please give P6  32x32  1/8 scan oudoor module qiangli

board707, 2022-12-19:
Hi
Your hardware is 32x32 scan8 panel, but you initialized the library with 32x16 s8 panel settings, it is because you got an incorrect result.

The library does not have a support of your panel yet, so you need to change your code as follows:

1. Add this at beginning of your code  just after line `#include "DMD_RGB.h"`
```
#define RGB32x32plainS8     34			// 32x32 1/8
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
	{}

};
```

2. Edit the DMD initialisation line as:
```
DMD_RGB <RGB32x32plainS8 , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

Please let me know about the results

maxmurugan, 2022-12-19:
[image]

/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#define RGB32x32plainS8 34      // 32x32 1/8
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(4, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
  {}

};
// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
char a[10];
char b[10];
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(STM32F1) || defined(STM32F4))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB32x32plainS8  , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// - 32x16 matrix with 8scans
// - 80x40 matrix with 20scans
// - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
UTF8 char recoding

/*--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/
// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255), // blue
dmd.Color888(0, 255, 255), // blue
dmd.Color888(0, 0, 0), // blue
};

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);
Serial.println("PICO P4 RTC");
delay(1000);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{

dmd.selectFont(&Arial_black);
dmd.setBrightness(255);
dmd.setTextColor(00, 0);
while (1) 
{
  

          dmd.drawString(0,0,"PLAN",4,col[1]);
          dmd.drawString(0,8,"ACTAL",6,col[2]);
          dmd.drawString(0,16,"GAP",3,col[0]);
          dmd.drawString(0,24,"EFFY",4,col[1]);
 }
}

maxmurugan, 2022-12-19:
plesae check

board707, 2022-12-19:
Why did you changed the code that I provide you to fix the issue???
Please insert the code EXACTLY as I present it in [my message](https://github.com/board707/DMD_STM32/issues/23#issuecomment-1357480971)

maxmurugan, 2022-12-19:
[image]

/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#define RGB32x32plainS8     34      // 32x32 1/8
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
  {}

};
// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
char a[10];
char b[10];
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(STM32F1) || defined(STM32F4))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB32x32plainS8 , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// - 32x16 matrix with 8scans
// - 80x40 matrix with 20scans
// - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
UTF8 char recoding

--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/
// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255), // blue
dmd.Color888(0, 255, 255), // blue
dmd.Color888(0, 0, 0), // blue
};

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);
Serial.println("PICO P4 RTC");
delay(1000);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{

dmd.selectFont(&Arial_black);
dmd.setBrightness(255);
dmd.setTextColor(00, 0);
while (1)
{

      dmd.drawString(0,0,"PLAN",4,col[1]);
      dmd.drawString(0,8,"ACTAL",6,col[2]);
      dmd.drawString(0,16,"GAP",3,col[0]);
      dmd.drawString(0,24,"EFFY",4,col[1]);
}
}

board707, 2022-12-19:
It seems that your matrix has a complex (not a plain) scan pattern.
To understand what the pattern scheme is, I will ask you to make several tests on your matrix.
Please replace your loop() function with this code:
```
void loop(void) {
uint16_t bg = 0;                         // background - black
uint16_t fg = dmd.Color888(0, 255, 0);;  // foreground 
    // draw pixels
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);

}
```
The other portions of the code shouldn't be changed.

Could you please run this test and make a video how it work in your matrix.
Thanks for your help.

maxmurugan, 2022-12-20:
https://user-images.githubusercontent.com/4962604/208583202-a1a1aa04-03e5-4fe3-99fc-4939acf93d48.mp4

maxmurugan, 2022-12-20:
Please check video

maxmurugan, 2022-12-20:
/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#define RGB32x32plainS8 34      // 32x32 1/8
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
  {}

};
// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
char a[10];
char b[10];
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(STM32F1) || defined(STM32F4))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB32x32plainS8  , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// - 32x16 matrix with 8scans
// - 80x40 matrix with 20scans
// - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
UTF8 char recoding

/*--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/
// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255), // blue
dmd.Color888(0, 255, 255), // blue
dmd.Color888(0, 0, 0), // blue
};

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);
Serial.println("PICO P4 RTC");
delay(1000);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
uint16_t bg = 0;                         // background - black
uint16_t fg = dmd.Color888(0, 255, 0);;  // foreground 
    // draw pixels
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);

}

board707, 2022-12-20:
Thanks for the info. I will figure it out and write to you

board707, 2022-12-20:
Could you please make another test?
Please replace the template section at the beginning of the code with this one:
```
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
	{}
protected:
	uint16_t get_base_addr(int16_t x, int16_t y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		uint16_t base_addr = (pol_y / this->multiplex) * this->x_len +
		(pol_y % this->multiplex) * this->WIDTH * this->DisplaysHigh + 
                (y / this->DMD_PIXELS_DOWN) * this->WIDTH + x;
		return base_addr;
	}

};
```

The other portions of the code are still without changes.

Please run this test and make a video how it work in your matrix.
Thanks for your help.

maxmurugan, 2022-12-20:
https://user-images.githubusercontent.com/4962604/208674768-ed3ca001-6a1f-4937-b58d-5fdc973dc0a5.mp4

maxmurugan, 2022-12-20:
please check this video

board707, 2022-12-20:
Thanks you very much!

board707, 2022-12-20:
Please test next code. 
As previous replace the template section at the beginning of the code with this one:
```
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
	{}
protected:
	uint16_t get_base_addr(int16_t x, int16_t y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		x += (y / DMD_PIXELS_DOWN) * WIDTH;
		base_addr = (pol_y % nRows) * x_len + (x / DMD_PIXELS_ACROSS) * multiplex * DMD_PIXELS_ACROSS;
                if (pol_y / nRows)  base_addr += x % DMD_PIXELS_ACROSS ;
                else   base_addr += x % DMD_PIXELS_ACROSS +  DMD_PIXELS_ACROSS;
		return base_addr;
	}

};
```

maxmurugan, 2022-12-21:
https://user-images.githubusercontent.com/4962604/208802089-ae1e2c95-19cd-4f4a-b2d5-5f24d1f77d05.mp4

maxmurugan, 2022-12-21:
[image]

/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#define RGB32x32plainS8 34 // 32x32 1/8
template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
  {}
protected:
  uint16_t get_base_addr(int16_t x, int16_t y) override {
    this->transform_XY(x, y);
    uint8_t pol_y = y % this->pol_displ;
    x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
   uint16_t  base_addr = (pol_y % this->nRows) * this->x_len + (x / this->DMD_PIXELS_ACROSS) * this->multiplex * this->DMD_PIXELS_ACROSS;
                if (pol_y / this->nRows)  base_addr += x % this->DMD_PIXELS_ACROSS ;
                else   base_addr += x % this->DMD_PIXELS_ACROSS +  this->DMD_PIXELS_ACROSS;
    return base_addr;
  }

};
// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
char a[10];
char b[10];
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

#if (defined(STM32F1) || defined(STM32F4))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

#elif (defined(ARDUINO_ARCH_RP2040))

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB32x32plainS8 , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// - 32x16 matrix with 8scans
// - 80x40 matrix with 20scans
// - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
UTF8 char recoding

/*--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/
// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255), // blue
dmd.Color888(0, 255, 255), // blue
dmd.Color888(0, 0, 0), // blue
};

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);
Serial.println("PICO P4 RTC");
delay(1000);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
/*uint16_t bg = 0; // background - black
uint16_t fg = dmd.Color888(0, 255, 0);; // foreground
// draw pixels
for (int i = 0; i < dmd.height(); i++) {
for (int j = 0; j < dmd.width(); j++) {
dmd.drawPixel(j,i, fg);
delay(30);
}
}
// clear the screen
dmd.fillScreen(bg);*/

dmd.selectFont(&Arial_black);
dmd.setBrightness(255);
dmd.setTextColor(0, 0);
while (1)
{

  dmd.drawString(0,0,"PLAN",4,col[1]);
  dmd.drawString(0,8,"ACTAL",6,col[2]);
  dmd.drawString(0,16,"GAP",3,col[0]);
  dmd.drawString(0,24,"EFFY",4,col[1]);
}

}

maxmurugan, 2022-12-21:
Some extra led turn on slightly around the text 
please check

board707, 2022-12-22:
Can't it be a power problem? What the power source do you use?

maxmurugan, 2022-12-22:
5v/40 amps rong smps

maxmurugan, 2022-12-22:
I try to change new smps same dot will showing

board707, 2022-12-23:
What board are you using for testing - STM32 or RP2040?
I found a small bug in the STM32 code, that may causing the flickering, but with RP2040 board your sketch works fine for me.

One additional moment - of course, I don't have exact your panels, so I tested the code on the different matrices than yours. It may be important, because not all of the matrices works stable with 3.3v logical signals. If this is a case - you will need to use 3.3 - 5v logical level converters on every signal lines of HUB75 interface.
Do your 32x32 s8 panels the only panels that you have? Could you test the library on some different matrices?

maxmurugan, 2022-12-24:
I am using RP2040

I will try a 3.3 to 5v logic converter using 74hc245

maxmurugan, 2022-12-29:
After add 74hc245 problem solved 
[image]

maxmurugan, 2022-12-29:
Thangs for your support 

Advanced happy  new year

board707, 2022-12-29:
Thank you for your cooperation.

closed 2022-12-29 by board707