board707/DMD_STM32 #27: DMD+W5500

OkSingra, 2023-02-09:
Russian:
Могу ли я использовать данную библиотеку с модулем W5500? Есть ли прототип печатной платы для Pi Pico? Со своей стороны имею доступ ко многим панелям разных производителей, хочу протестировать на всех. Особенно интересны панели P2.5 128х64 с контактом D1между G1 и G2 (фото выложу позже) и 64х32 P5 (обе уличные). И еще нужно ли преобразовывать логику в 5V для всех панелей? подойдут ли 74HCT245?

board707, 2023-02-09:
Hi, @OkSingra 

> Can I use this library with the W5500 module?

I have no ideas :) , I did not test it with W5500.

> For my part, I have access to many panels from different manufacturers, I want to test at all. Especially interesting are the panels P2.5 128x64 with contact D1 between G1 and G2 (I will post photos later) and 64x32 P5 (both street).

I doubt that your first panel will work with library, or at least it will require some modification of the code. About the last panel, 64x32 P5 , I can say nothing without the specification.

> is it also necessary to convert the logic to 5V for all panels? Will 74HCT245 work?

Some panels has logic buffers like 74HCT245 on board already. These panels do not need the logic conversion. The others can work unstable with 3.3v logic.

If you can discuss on Russian, please contact me by email dd@jwee.ru

wsandor, 2023-02-09:
Hi, @OkSingra

I use it with Ethernet_generic library with a W5500 module without problems.

OkSingra, 2023-02-09:
Is it fixed somehow? The same error on all versions of arduino. Platformio also throws this error. I did not find a clear fix.

fatal error: pico/stdio.h: No such file or directory
 #include "pico/stdio.h"

board707, 2023-02-09:
Please show complete error message. What code do you try to compile?

OkSingra, 2023-02-09:
[image]

board707, 2023-02-09:
The "arduino-mbed" framework, that you use - is incompatible with the library. 
Please read the section [Compatible IDE and libraries](https://github.com/board707/DMD_STM32#compatible-ide-and-libraries) of the libary README.
For RP2040 boards only Earle Philhower arduino core is supported.

OkSingra, 2023-02-09:
[image]
Thanks a lot. Works great!

OkSingra, 2023-02-10:
[image]
On Windows 11 it was not possible to start the correct work with the platformio. On Linux Mint, everything worked correctly.

OkSingra, 2023-02-10:
[image]

OkSingra, 2023-02-11:
Can I change the pins at my discretion?

board707, 2023-02-11:
About the STM32 connections see the wiki: [RGB panel connections](https://github.com/board707/DMD_STM32/wiki/Connections---RGB)
For RP2040 you can select any pins under the following conditions: pins A B C D E must have consecutive numbers, the same applies to R0 G0 B0 R1 G1 B1

OkSingra, 2023-02-16:
https://youtu.be/FO7wHBfsk_8

board707, 2023-02-16:
> Please contact me via telegram @Hiderstv

sorry, I do not have a telegram

board707, 2023-02-16:
please write me ddt@jwee.ru

closed 2023-03-14 by board707