board707/DMD_STM32 #29: More then One module Scrolling issue

maxmurugan, 2023-02-17:
hi sir,
I am using 3 modules p6 32x32 
one module working good 
add more than one module scrolling only working one module

https://user-images.githubusercontent.com/4962604/219630733-30f57236-2991-417e-b4da-501456d74b26.mp4

maxmurugan, 2023-02-17:
/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include <Wire.h>
#include <EEPROM.h>
#include "DMD_RGB.h"
#define RGB32x32plainS8 34 // 32x32 1/8
//###########################################################################################

template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
byte panelsWide, byte panelsHigh, bool d_buf = false) :
DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
{}
protected:
uint16_t get_base_addr(int16_t x, int16_t y) override {
this->transform_XY(x, y);
uint8_t pol_y = y % this->pol_displ;
x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
uint16_t base_addr = (pol_y % this->nRows) * this->x_len + (x / this->DMD_PIXELS_ACROSS) * this->multiplex * this->DMD_PIXELS_ACROSS;
if (pol_y / this->nRows) base_addr += x % this->DMD_PIXELS_ACROSS ;
else base_addr += x % this->DMD_PIXELS_ACROSS + this->DMD_PIXELS_ACROSS;
return base_addr;
}

};
//#######################################################
#include "st_fonts/UkrRusArial14.h"
#include "st_fonts/Arial_Black_16.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/FreeMono12pt7b.h"
#include "gfx_fonts/FreeSerif9pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 3
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER true

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1

DMD_RGB <RGB32x32plainS8, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(Arial_Black_16);
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic charsFreeMonoBold12pt7b
// 32 pixel DMD_GFX_Font GlametrixL((uint8_t*)&FreeMono24pt7b,24 );
DMD_GFX_Font GlametrixL((uint8_t*)&FreeSerif9pt7b,12 );

/*--------------------------------------------------------------------------------------
UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {
  
  uint16_t i, j;
  for ( i =0, j =0; src[i]; i++) {
   if ((src[i] == 0xD0 )&& src[i+1])  { dest[j++] = src[++i] - 0x10;}
    else if ((src[i] == 0xD1 )&& src[i+1]) {dest[j++] = src[++i] + 0x30;  }
    else dest[j++] = src[i];
  }
  dest[j] ='\0';
  return j;
}

/*--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/
void loop(void)
{

// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255) // blue

};
uint16_t bg = 0; // background - black

// text
char s[] = "WELCOME TO KAMAL & CO...";
// transcode message to UTF for use with GFX fonts
char k[100];
utf8_rus(k, (const unsigned char*)s);
char* m = s;
dmd.selectFont(&Arial_black);
uint16_t interval = 35;
long prev_step = millis();
dmd.setBrightness(128);
uint8_t col_ptr = 0;

while (1)
  {
    //##########################
    if ((millis() - prev_step) > interval)

    { dmd.setTextColor(col[0], bg);    // --red color
      if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
        dmd.clearScreen(true);

        dmd.drawMarquee(m, strlen(m), dmd.width() - 10, (dmd.height() > 0) ? 16 : 0, 0);
      }
      // output mem buffer to matrix

      //####################### add string get flicker
      dmd.setTextColor(col[2], bg);  //-- blue color
      dmd.drawStringX(0, 0, "TEST 12345", col[2]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      dmd.swapBuffers(true);

      prev_step = millis();
    }
  }
}

maxmurugan, 2023-02-17:
please solve the issue

board707, 2023-02-17:
Hi again
First please update the library to last version v1.0.0.

Than go to the example dmd_rgb_pattern_test and replace the line 74:

```
DMD_RGB <RGB32x16plainS4, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```
with two lines below:
```
#define RGB32x32_S8_maxmurugan		3,32,32,1,35	// 32x32 1/8 matrix from @maxmurugan
DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```
Run this example with a single panel and show the video as it works.
Sorry for the inconvenience.
Dmitry

board707, 2023-02-17:
Attenetion, I just edited the code in the previous post, please copy it again

maxmurugan, 2023-02-17:
After update new library

Test two  32x32 p6 panel

one panel scrolling text fully  , second module  show blank

https://user-images.githubusercontent.com/4962604/219700951-ac1d10e3-ca7f-4c55-b76f-f4472b2449e3.mov

maxmurugan, 2023-02-17:
/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include <Wire.h>
#include <EEPROM.h>
#include "DMD_RGB.h"
#define RGB32x32_S8_maxmurugan    3,32,32,8,1 // 32x32 1/8 matrix from @maxmurugan
//###########################################################################################

template<int COL_DEPTH>
class DMD_RGB<RGB32x32_S8_maxmurugan, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
byte panelsWide, byte panelsHigh, bool d_buf = false) :
DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
{}
protected:
uint16_t get_base_addr(int16_t x, int16_t y) override {
this->transform_XY(x, y);
uint8_t pol_y = y % this->pol_displ;
x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
uint16_t base_addr = (pol_y % this->nRows) * this->x_len + (x / this->DMD_PIXELS_ACROSS) * this->multiplex * this->DMD_PIXELS_ACROSS;
if (pol_y / this->nRows) base_addr += x % this->DMD_PIXELS_ACROSS ;
else base_addr += x % this->DMD_PIXELS_ACROSS + this->DMD_PIXELS_ACROSS;
return base_addr;
}

};
//#######################################################
#include "st_fonts/UkrRusArial14.h"
#include "st_fonts/Arial_Black_16.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER true

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1

//DMD_RGB <RGB32x32plainS8, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(Arial_Black_16);
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic charsFreeMonoBold12pt7b
// 32 pixel DMD_GFX_Font GlametrixL((uint8_t*)&FreeMono24pt7b,24 );

/*--------------------------------------------------------------------------------------
UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {
  
  uint16_t i, j;
  for ( i =0, j =0; src[i]; i++) {
   if ((src[i] == 0xD0 )&& src[i+1])  { dest[j++] = src[++i] - 0x10;}
    else if ((src[i] == 0xD1 )&& src[i+1]) {dest[j++] = src[++i] + 0x30;  }
    else dest[j++] = src[i];
  }
  dest[j] ='\0';
  return j;
}
/*--------------------------------------------------------------------------------------
setup
Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

// initialize DMD objects
dmd.init();
Serial.begin(9600);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/
void loop(void)
{

// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255) // blue

};
uint16_t bg = 0; // background - black

// text
char s[] = "WELCOME TO KAMAL & CO...";
// transcode message to UTF for use with GFX fonts
char k[100];
utf8_rus(k, (const unsigned char*)s);
char* m = s;
dmd.selectFont(&Arial_black);
uint16_t interval = 35;
long prev_step = millis();
dmd.setBrightness(128);
uint8_t col_ptr = 0;

while (1)
{
//##########################
if ((millis() - prev_step) > interval)

{ dmd.setTextColor(col[0], bg);    // --red color
  if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);

    dmd.drawMarquee(m, strlen(m), dmd.width() - 10, (dmd.height() > 0) ? 16 : 0, 0);
  }
  // output mem buffer to matrix

  //####################### add string get flicker
  dmd.setTextColor(col[2], bg);  //-- blue color
  dmd.drawStringX(0, 0, "TEST 12345", col[2]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
  dmd.swapBuffers(true);

  prev_step = millis();
}
}
}

maxmurugan, 2023-02-17:
Please check

board707, 2023-02-17:
Sorry, but this is not what I asking for...
I ask you that you repeat panel tests as you done it one month ago - but now with a new library version.

Please get the example **dmd_rgb_pattern_test** and replace ONLY TWO LINES, as stated in the [post above](https://github.com/board707/DMD_STM32/issues/29#issuecomment-1434562473)
Do not add your template and your scrolling there.

maxmurugan, 2023-02-17:
https://user-images.githubusercontent.com/4962604/219710257-2bd421e7-a3af-4a31-83a4-2318d7c59e03.mp4

maxmurugan, 2023-02-17:
what is next step sir?

maxmurugan, 2023-02-17:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32 and RP2040 boards

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
#endif

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 Direct mux
// RGB32x16plainS2         - 32x16 1/2 Binary mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Direct mux

//DMD_RGB <RGB32x16plainS4, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
#define RGB32x32_S8_maxmurugan    3,32,32,8,1 // 32x32 1/8 matrix from @maxmurugan
DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(40);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}

board707, 2023-02-17:
Thank you

Please test this template. 
You should add it to your code(not to example!) instead of previous template, situated between two long lines with many-many "##########" chars

```
#define RGB32x32_S8_maxmurugan 3,32,32,8,35 // 32x32 1/8 matrix from @maxmurugan
template<int COL_DEPTH>
class DMD_RGB<RGB32x32_S8_maxmurugan, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
	{
		this->fast_Hbyte = false;
		this->use_shift = false;
	}
	// Fast text shift is disabled for complex patterns, so we don't need the method
	void disableFastTextShift(bool shift) override {}

protected:
	uint16_t get_base_addr(int16_t x, int16_t y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;

		static const uint8_t A_tbl[16] = { 0,1,2,3,4,5,6,7,
                                                                       0,1,2,3,4,5,6,7 };
		static const uint8_t B_tbl[16] = { 1,1,1,1,1,1,1,1,
                                                                        0,0,0,0,0,0,0,0 };
		static const uint8_t C_tbl[8] = { 0,1,2,3,4,5,6,7 };
		static const uint8_t Pan_Okt_cnt = this->DMD_PIXELS_ACROSS / 8;
		uint8_t Oktet_m = B_tbl[pol_y] * Pan_Okt_cnt + (x / 8) % Pan_Okt_cnt;

		uint16_t base_addr = this->x_len * A_tbl[pol_y] + (x / this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex + C_tbl[Oktet_m] * 8;
		base_addr += x % 8;
		return base_addr;
	}

};
```

maxmurugan, 2023-02-17:
wow! working two panel

Thank you very much

maxmurugan, 2023-02-17:
one more dought
how many scrolling message support?

maxmurugan, 2023-02-17:
how many line  scrolling will support?

board707, 2023-02-17:
> wow! working two panel

What about three panels as in the video at start of the issue?

> how many scrolling message support?
only one :(

maxmurugan, 2023-02-17:
Tomorrow I will test 10 panel

maxmurugan, 2023-02-18:
3 panel working with 250 characters scrolling working fine
Thanks

maxmurugan, 2023-02-18:
Uploading 69839728125__373C0ECC-BA7A-4702-8725-3DEEEC80EA74.MOV…

maxmurugan, 2023-02-19:
only support 255 char

more then 255 char not support scrolling

char* m = "Ashwin and Ravindra Jadeja to take an unassailable 2-0 lead in the series. The victory in New Delhi also earned India a big boost as far as their World Test Championship final qualification hopes are concerned. With the win India strengthened their spot at the second position in the World Test Championship points table.";

only scrolling  - Ashwin and Ravindra Jadeja to take an unass

maxmurugan, 2023-02-21:
hi sir please help ?
more then 255 char

board707, 2023-02-23:
Hi
Sorry for delay with answers. 
I am not in home and can't edit github now.

About the more than 255 chars scrolling - try to edit parameters of this functions:
```
void DMD::stringBounds(const char* bChars, uint8_t length,
	int16_t* w, int16_t* min_y, int16_t* max_y, byte orientation);

uint16_t DMD::stringWidth(const char* bChars, uint8_t length, byte orientation);
```
changing type of uint8_t length to uint16_t.
The functions should be edited in DMD_STM32a.cpp file and its prototypes in DMD_STM32a.h

maxmurugan, 2023-02-24:
ok i will try and inform you

thanks

maxmurugan, 2023-02-24:
change  to unit16_t  same issue will happened

256 byte only working

board707, 2023-03-11:
Hi
I edited the two functions as mentioned two messages above and scrolling works for me up to string 300 chars long.
If you need a string more longer - you have to edit define at line 80 of the `DMD_STM32a.h` file:
```
//Max length of scrolling text
#define MAX_STRING_LEN 300
```

board707, 2023-03-14:
Fixed in v1.0.1

closed 2023-03-14 by board707