board707/DMD_STM32 #31: SPI not working in single colour p10 panel

Ramkumar073, 2023-03-10:
I got an output while using parallel library but when it comes to spi library I didn't get anything in my p10 panel

board707, 2023-03-10:
What the controller do you use& STM32 or RP2040?

Ramkumar073, 2023-03-10:
I am using SPI 1 so A5 as a CLK and A7 as a DATA 
Here below is my coding part (I am using SPI library so I comment like this //#define DMD_PARA)

//........................................................
/*--------------------------------------------------------------------------------------
 Demo code for DMD panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
// choose between Parallel ans SPI wiring
// comment line below for SPI connection
//#define DMD_PARA

/*--------------------------------------------------------------------------------------
  Includes
--------------------------------------------------------------------------------------*/
#if defined(DMD_PARA)
#include "DMD_Monochrome_Parallel.h"
#else
#include "DMD_MonoChrome_SPI.h"
#endif

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing

#define ENABLE_DUAL_BUFFER true

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be Timer3 PWM pin as PB0,PB1
// if connect as SPI, do not use corresponding MiSO pin - PA6 for SPI1 and PB14 for SPI2

#define DMD_PIN_A PB11
#define DMD_PIN_B PB12
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PB10

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

//=== Config for Parallel connect ====
#if defined(DMD_PARA)
//pins for rows at x axis
// example for two rows
// all those pins must be selected from same port!
uint8_t pins[] = { PA5, PA7, PA6 };  // CLK , row1, row 2

//Fire up the DMD object as dmd
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
#else

//=== Config for SPI connect ====
[... 138 lines cut ...]

                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

            // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

            prev_step = millis();

        }
    }
}

board707, 2023-03-10:
Thank you.
At first look I don't see any problems with code. What is the version of the library you use? What kind of STM32 bluepill board - is it F103C8 or C6 chip?

Ramkumar073, 2023-03-10:
Actually above code was taken from your example dmd_monochrome I just modified below line only
#define DISPLAYS_ACROSS 2 (from) #define DISPLAYS_ACROSS 1
because I have to use two panel

Ramkumar073, 2023-03-10:
It is F103C8 Adafruit gfx 1.7 version and also tried to use 1.8 , 2 version

board707, 2023-03-10:
I mean what is the version of the DMD_STM32 library. You can see actual version number in the library.properties file or at start of DMD_STM32a.h file

Ramkumar073, 2023-03-10:
VERSION 0.9.3

board707, 2023-03-10:
Sorry, I just discovered that I forgot to change the version number in DMD_STM32a.h file.
Could you please to check the version in the library.properties file in the library folder in Arduino libraries directory.
Sorry for inconvenience.

Ramkumar073, 2023-03-10:
Version 1.0.0

board707, 2023-03-10:
Hi
I have tested your code with your pins configuration - it works fine for me with v1.0.0 and v0.9.5.

The only change I made in your code - fixed dest parameter type in `utf8_rus()` function header.
In your code it was:
```
int utf8_rus(char dest, const unsigned char* src)
```

Correct header should be:
```
int utf8_rus(char* dest, const unsigned char* src)
```
but I think that it was just a typo.

Ramkumar073, 2023-03-11:
Yes, that's typo mistake but I checked with corrected code and with pin connection still I am not getting anything in my display and also, I checked in pulse view there was no pulse generated by DATA pin (PA5) 
could you please suggest any other reason for this problem.

board707, 2023-03-11:
It seems to be a hardware problem. 
Please check that mcu on the board is STM32F103 and not a CS32 or CH32 clone.
Check the soldering, especially PA7 and GND pins...

Ramkumar073, 2023-03-11:
ok thanks and then I am using buffer IC 74hc245 between bluepill and p10 panel, is there any problem?

board707, 2023-03-11:
Are you using it with both Parallel and SPI mode? If it works in Parallel - I don't see why it have not work with SPI....

Ramkumar073, 2023-03-11:
yes, it works in parallel not in SPI. ok I will try with another bluepill.

board707, 2023-03-11:
> there was no pulse generation by DATA pin (PA5)

It can be just a typo again, but please note that in SPI mode DATA matrix pin should be connected to PA7, whereas CLK pin to the PA5 pin

Ramkumar073, 2023-03-11:
sry that was my typo mistake but I am sure DATA pin is connected to A7 and CLK with A5

closed 2023-06-22 by board707