board707/DMD_STM32 #35: P5 32x64 scan problem

miralay001, 2023-03-31:
I have a p5 32x64 1/8 color scanning screen, I used your patterns, but I was not successful, can you help me?

`#define NAME_OF_PANEL  MUX pins, Panel_width, Panel_height, scan, pattern ID
#define RGB64x32_S8_OKSingra     3,64,32,8,1
// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32_S8_OKSingra  , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
  UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {

    uint16_t i, j;
    for (i = 0, j = 0; src[i]; i++) {
        if ((src[i] == 0xD0) && src[i + 1]) { dest[j++] = src[++i] - 0x10; }
        else if ((src[i] == 0xD1) && src[i + 1]) { dest[j++] = src[++i] + 0x30; }
        else dest[j++] = src[i];
    }
    dest[j] = '\0';
    return j;
}

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

    // initialize DMD objects
	dmd.init(); 
    
}

/*--------------------

------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
uint16_t bg = 0;                         // background - black
uint16_t fg = dmd.Color888(0, 255, 0);;  // foreground 
    // draw pixels
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);

}`

the most i could do

https://user-images.githubusercontent.com/108148338/229152384-c77dcd28-360a-43e4-a639-6ce3649922f4.mp4

board707, 2023-03-31:
Hi
Could you show clear photo of the panel from the reverse side?

miralay001, 2023-03-31:
[image]
[image]

HRL - P5 -8S 1921 64X32-V2.0 OUTDOOR PANEL

board707, 2023-04-01:
Thank you
Could you also indicate, a marks of all the chips on the matrix? What's the board are you using - stm32 or rp2040?

Try to replace the class initialization line with this:
```
DMD_RGB_SHIFTREG_ABC <RGB64x32_S8_OKSingra , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

miralay001, 2023-04-01:
```
#define NAME_OF_PANEL MUX pins, Panel_width, Panel_height, scan, pattern ID
#define RGB64x32_S8_OKSingra 3,64,32,8,0
```
`DMD_RGB_SHIFTREG_ABC <RGB64x32_S8_OKSingra , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);`

i tried the code

the card i use : stm32f103c8t6

I've used your library before with the p10 matrix without any problems.

i have p5 right now

chip on panel SM162085

[image]

https://user-images.githubusercontent.com/108148338/229279622-fa297dc4-4518-459b-b233-c031270a5b28.mp4

board707, 2023-04-01:
hmm... I don't understand this matrix

try to use
`#define RGB64x32_S8_OKSingra 3,64,32,8,1`
or
`#define RGB64x32_S8_OKSingra 3,64,32,8,2`

board707, 2023-04-01:
Sorry
After reading the datasheet I think I won't managed your panel to work with my library
At least until then I will not have such a panel for tests.
Or until someone else posts a sample code to work with this chip.

miralay001, 2023-04-01:
thank you for helping

there is no different chip on the panel
MM245B and SM18208SJ 

Here are the pins I use and all the code

```
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PB11
//#define NAME_OF_PANEL  MUX pins, Panel_width, Panel_height, scan, pattern ID
//#define RGB32x16_S2_OKSingra  1,32,16,2,55
#define NAME_OF_PANEL MUX pins, Panel_width, Panel_height, scan, pattern ID
#define RGB64x32_S8_OKSingra     3,64,32,8,0

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB<RGB64x32_S8_OKSingra , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
  UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {

    uint16_t i, j;
    for (i = 0, j = 0; src[i]; i++) {
        if ((src[i] == 0xD0) && src[i + 1]) { dest[j++] = src[++i] - 0x10; }
        else if ((src[i] == 0xD1) && src[i + 1]) { dest[j++] = src[++i] + 0x30; }
        else dest[j++] = src[i];
    }
    dest[j] = '\0';
    return j;
}

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

    // initialize DMD objects
	dmd.init(); 
    
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
uint16_t bg = 0;                         // background - black
uint16_t fg = dmd.Color888(0, 255, 0);;  // foreground 
    // draw pixels
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);

}

```

miralay001, 2023-04-01:
> hmm... Bu matrisi anlamıyorum
> 
> kullanmayı dene `#define RGB64x32_S8_OKSingra 3,64,32,8,1` ya `#define RGB64x32_S8_OKSingra 3,64,32,8,2`

I tried both patterns

   RGB64x32_S8_OKSingra 3,64,32,8,1   =  pattern in first video
   RGB64x32_S8_OKSingra 3,64,32,8,0   =  i am getting error

miralay001, 2023-04-01:
> hmm... I don't understand this matrix
> 
> try to use `#define RGB64x32_S8_OKSingra 3,64,32,8,1` or `#define RGB64x32_S8_OKSingra 3,64,32,8,2`

I tried both patterns

RGB64x32_S8_OKSingra 3,64,32,8,1 = pattern in first video
RGB64x32_S8_OKSingra 3,64,32,8,0 = i am getting error

board707, 2023-04-01:
> RGB64x32_S8_OKSingra 3,64,32,8,1 = pattern in first video

It looks like one of pins A, B or C is not works. Please check your connections. Or try to swap A and C, for example

> RGB64x32_S8_OKSingra 3,64,32,8,2 = i am getting error

Please show the error message

miralay001, 2023-04-01:
After changing PIN numbers

```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB7
#define DMD_PIN_C PB8
#define DMD_PIN_D -
#define DMD_PIN_E -
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C};

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PB11
//#define NAME_OF_PANEL  MUX pins, Panel_width, Panel_height, scan, pattern ID
//#define RGB32x16_S2_OKSingra  1,32,16,2,55
#define NAME_OF_PANEL MUX pins, Panel_width, Panel_height, scan, pattern ID
#define RGB64x32_S8_OKSingra     3,64,32,8,1

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB<RGB64x32_S8_OKSingra , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
  UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {

    uint16_t i, j;
    for (i = 0, j = 0; src[i]; i++) {
        if ((src[i] == 0xD0) && src[i + 1]) { dest[j++] = src[++i] - 0x10; }
        else if ((src[i] == 0xD1) && src[i + 1]) { dest[j++] = src[++i] + 0x30; }
        else dest[j++] = src[i];
    }
    dest[j] = '\0';
    return j;
}

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

    // initialize DMD objects
	dmd.init(); 
    
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
uint16_t bg = 0;                         // background - black
uint16_t fg = dmd.Color888(0, 255, 0);;  // foreground 
    // draw pixels
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);

}
```

https://user-images.githubusercontent.com/108148338/229301364-80d5c6c6-009d-4627-8b95-79ab4d688806.mp4

board707, 2023-04-01:
Success!
 Now A B C pins are works.

Could you please test
```
#define RGB64x32_S8_OKSingra     3,64,32,8,0
```
and
```
#define RGB64x32_S8_OKSingra     3,64,32,8,2
```
again with new A B C pins?

miralay001, 2023-04-01:
`#define RGB64x32_S8_OKSingra     3,64,32,8,0 `  test with pattern

https://user-images.githubusercontent.com/108148338/229315865-31bab274-229b-4e62-b827-d87073679d80.mp4

`#define RGB64x32_S8_OKSingra     3,64,32,8,2 `  test with pattern

https://user-images.githubusercontent.com/108148338/229315887-57b25011-c41a-4719-8669-9dac65c707cd.mp4

miralay001, 2023-04-01:
dmd_rgb.ino Test `#define RGB64x32_S8_OKSingra     3,64,32,8,2`

https://user-images.githubusercontent.com/108148338/229316148-6bf79adb-a342-4da4-aa84-a9eea610f699.mp4

dmd_rgb.ino Test `#define RGB64x32_S8_OKSingra     3,64,32,8,1`

https://user-images.githubusercontent.com/108148338/229316269-996065be-cc21-4e3f-b6be-1f33771ca57e.mp4

dmd_rgb.ino Test `#define RGB64x32_S8_OKSingra     3,64,32,8,0`

https://user-images.githubusercontent.com/108148338/229316322-d1ef7d7a-66bc-4496-95fb-8753d06f00ec.mp4

board707, 2023-04-02:
Thank you very much.

I will try to understand the pattern and write to you.

board707, 2023-04-03:
Hi
Let's test your pattern.
Please add this piece to your code just after `#include "RGB.h"` line:
```
#define RGB64x32_s8_miralay		3,64,32,8,56	// 64x32 s8

template<int COL_DEPTH>
class DMD_RGB<RGB64x32_s8_miralay, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(1, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 2, 32, 16)
	{
		this->fast_Hbyte = false;
		this->use_shift = false;
	}
	// Fast text shift is disabled for complex patterns, so we don't need the method
	void disableFastTextShift(bool shift) override {}

protected:
	uint16_t get_base_addr(int16_t x, int16_t y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
                
                static const uint8_t Pan_quart_cnt = this->DMD_PIXELS_ACROSS / 4;
                uint16_t base_addr = this->x_len * (pol_y % this->n_rows) + (x / this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex ;
                 if (pol_y < this->n_Rows)   base_addr += 4;
		 base_addr += ((x / 4) % Pan_quart_cnt) * 8 + x % 4;
		
		return base_addr;
	}

};
```

and for DMD initializtation use the matrix type `RGB64x32_s8_miralay` too

miralay001, 2023-04-03:
`/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#define RGB64x32_s8_miralay    3,64,32,8,56  // 64x32 s8

template<int COL_DEPTH>
class DMD_RGB<RGB64x32_s8_miralay, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(1, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 2, 32, 16)
  {
    this->fast_Hbyte = false;
    this->use_shift = false;
  }
  // Fast text shift is disabled for complex patterns, so we don't need the method
  void disableFastTextShift(bool shift) override {}

protected:
  uint16_t get_base_addr(int16_t x, int16_t y) override {
    this->transform_XY(x, y);
    uint8_t pol_y = y % this->pol_displ;
    x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
                
                static const uint8_t Pan_quart_cnt = this->DMD_PIXELS_ACROSS / 4;
                uint16_t base_addr = this->x_len * (pol_y % this->n_Rows) + (x / this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex ;
                 if (pol_y < this->n_Rows)   base_addr += 4;
     base_addr += ((x / 4) % Pan_quart_cnt) * 8 + x % 4;
    
    return base_addr;
  }

};

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB7
#define DMD_PIN_C PB8
#define DMD_PIN_D -
#define DMD_PIN_E -
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C};

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PB11
//#define NAME_OF_PANEL  MUX pins, Panel_width, Panel_height, scan, pattern ID
//#define RGB32x16_S2_OKSingra  1,32,16,2,55
#define NAME_OF_PANEL MUX pins, Panel_width, Panel_height, scan, pattern ID

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB<RGB64x32_s8_miralay , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
  UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {

    uint16_t i, j;
    for (i = 0, j = 0; src[i]; i++) {
        if ((src[i] == 0xD0) && src[i + 1]) { dest[j++] = src[++i] - 0x10; }
        else if ((src[i] == 0xD1) && src[i + 1]) { dest[j++] = src[++i] + 0x30; }
        else dest[j++] = src[i];
    }
    dest[j] = '\0';
    return j;
}

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{

    // initialize DMD objects
	dmd.init(); 
    
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
uint16_t bg = 0;                         // background - black
uint16_t fg = dmd.Color888(0, 255, 0);;  // foreground 
    // draw pixels
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);

}`
[image]

I get this error

miralay001, 2023-04-03:
latest status

https://user-images.githubusercontent.com/108148338/229567802-2fcb5a41-75e4-46f9-b314-b5d7cea39c6a.mp4

board707, 2023-04-03:
I'm sorry, as I don't have a matrix, I not able to test the code before provide it to you

Please test the next variant:
```
#define RGB64x32_s8_miralay		3,64,32,8,56	// 64x32 s8

template<int COL_DEPTH>
class DMD_RGB<RGB64x32_s8_miralay, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 64, 32)
	{
		this->fast_Hbyte = false;
		this->use_shift = false;
	}
	// Fast text shift is disabled for complex patterns, so we don't need the method
	void disableFastTextShift(bool shift) override {}

protected:
	uint16_t get_base_addr(int16_t x, int16_t y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
                
                static const uint8_t Pan_quart_cnt = this->DMD_PIXELS_ACROSS / 4;
                uint16_t base_addr = this->x_len * (pol_y % this->nRows) + (x / this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex ;
                 if (pol_y < this->nRows)   base_addr += 4;
		 base_addr += ((x / 4) % Pan_quart_cnt) * 8 + x % 4;
		
		return base_addr;
	}

};
```

miralay001, 2023-04-03:
thank you so much you are a genius :)

https://user-images.githubusercontent.com/108148338/229602195-14904cb3-936e-4257-9b0d-61c950a6731b.mp4

miralay001, 2023-04-03:
> `/*------------------------------------------------------------ -------------------------------------------------- RGB paneller için Demo
> 
> DMD_STM32a STM32F103xxx kartı için örnek kod -------------------------------------------------------- ----------------------------------------- */ #include "DMD_RGB.h" #define RGB64x32_s8_miralay 3,64,32,8,56 // 64x32 s8
> 
> şablon sınıfı DMD_RGB<RGB64x32_s8_miralay, COL_DEPTH> : genel DMD_RGB_BASE2<COL_DEPTH> { genel: DMD_RGB(uint8_t* mux_list, bayt _pin_nOE, bayt _pin_SCLK, uint8_t* pinlist, bayt panellerWide, bayt panellerYüksek, bool d_buf = yanlış) : SEMD_2(CGB_DEPTH) 1, mux_list, _pin_nOE, _pin_SCLK, pinlist, panellerWide, panellerYüksek, d_buf, COL_DEPTH, 2, 32, 16) { this->fast_Hbyte = false; this->use_shift = false; } // Karmaşık kalıplar için hızlı metin kaydırma devre dışı bırakıldığından, void disableFastTextShift(bool shift) override {} yöntemine ihtiyacımız yok
> 
> korumalı: uint16_t get_base_addr(int16_t x, int16_t y) geçersiz kılma { this->transform_XY(x, y); uint8_t pol_y = y % bu->pol_displ; x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
> 
> ```
>             static const uint8_t Pan_quart_cnt = this->DMD_PIXELS_ACROSS / 4;
>             uint16_t base_addr = this->x_len * (pol_y % this->n_Rows) + (x / this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex ;
>              if (pol_y < this->n_Rows)   base_addr += 4;
>  base_addr += ((x / 4) % Pan_quart_cnt) * 8 + x % 4;
> 
> return base_addr;
> ```
> 
> }
> 
> };
> 
> // Yazı tipleri içerir #include "st_fonts/UkrRusArial14.h" #pragma GCC tanı yoksay "-Wnarrowing" #pragma GCC tanı yoksay "-Woverflow" #include "gfx_fonts/GlametrixLight12pt7b.h" #include "gfx_fonts/GlametrixBold12pt7b.h" # pragma GCC teşhis uyarısı "-Wnarrowing" #pragma GCC teşhis uyarısı "-Woverflow"
> 
> //x ve y eksenindeki panel sayısı #define DISPLAYS_ACROSS 1 #define DISPLAYS_DOWN 1
> 
> // Çıktı tamponlamayı etkinleştir // true ise, // swapBuffers(true) komutundan sonra yalnızca çıktıları matrise değiştirir // Dual buffer etkin değilse, tüm çıktı doğrudan matriste çizilir // ve swapBuffers(true) cimmand hiçbir şey yapmaz #define ENABLE_DUAL_BUFFER yanlış
> 
> // ==== DMD_RGB pinleri ==== // mux pinleri - A, B, C... tüm mux pinleri aynı porttan seçilmelidir! #define DMD_PIN_A PB6 #define DMD_PIN_B PB5 #define DMD_PIN_C PB4 #define DMD_PIN_D  PB4 #define DMD_PIN_E - // tüm mux pinlerini listeye koy uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C};
> 
> // pin OE, PB0 PB1 PA6 PA7'den biri olmalıdır #define DMD_PIN_nOE PB0 #define DMD_PIN_SCLK PB7 //#define NAME_OF_PANEL MUX pinleri, Panel_width, Panel_height, scan, pattern ID //#define RGB32x16_S2_OKSingra 1,32,16,2,55 #define NAME_OF_PANEL MUX pimleri, Panel_genişliği, Panel_yüksekliği, tarama, desen kimliği
> 
> // R0, G0, B0, R1, G1, B1 kanalları ve saat pinleri. // Varsayılan olarak kitaplık, RGB renk sırasını kullanır. // Bunu değiştirmeniz gerekirse - R0, G0, B0, R1, G1, B1 pinlerini yeniden sıralayın. // Tüm bu pinler de aynı porttan seçilmelidir! uint8_t özel_rgbpinler[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
> 
> // DMD nesnesini dmd<MATRIX_TYPE, COLOR_DEPTH> olarak çalıştırıyoruz // 16 taramalı ve 4bit renkli 64x32 matris kullanıyoruz: DMD_RGB<RGB64x32_s8_miralay , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_LKBU, custom_rgbpins, DISPLAYS_FLAYS_DUS); // diğer seçenekler şunlardır: // - 8 taramalı 32x16 matris // - 20 taramalı 80x40 matris // - 32 taramalı 64x64 matris // Renk derinliği - <COLOR_4BITS_Packed>, <COLOR_4BITS> veya <COLOR_1BITS>
> 
> // --- Yazı tiplerini tanımla ---- // DMD.h eski stil yazı tipi DMD_Standard_Font UkrRusArial_F(UkrRusArial_14); // Latin ve Kiril karakterleri için ayrı bölümleri olan GFX yazı tipi DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);
> 
> /*------------------------------------------------ -------------------------------------------------- UTF8 karakter kaydı
> 
> -------------------------------------------------- ------------------------------------ _/ int utf8_rus(char_ dest, const unsigned char* src) {
> 
> ```
> uint16_t i, j;
> for (i = 0, j = 0; src[i]; i++) {
>     if ((src[i] == 0xD0) && src[i + 1]) { dest[j++] = src[++i] - 0x10; }
>     else if ((src[i] == 0xD1) && src[i + 1]) { dest[j++] = src[++i] + 0x30; }
>     else dest[j++] = src[i];
> }
> dest[j] = '\0';
> return j;
> ```
> 
> }
> 
> / _------------------------------------------------- ------------------------------------- ana döngü başlamadan önce Arduino mimarisi tarafından çağrılan kurulum -- -------------------------------------------------- ----------------------------------_ /
> 
> geçersiz kurulum (geçersiz) {
> 
> ```
> // initialize DMD objects
> dmd.init(); 
> ```
> 
> }
> 
> / _------------------------------------------------- ------------------------- döngü Arduino mimarisi ana döngü -------- -------------------------------------------------- ----------------------------_ /
> 
> geçersiz döngü(void) { uint16_t bg = 0; // arka plan - siyah uint16_t fg = dmd.Color888(0, 255, 0);; // ön plan // pikselleri çiz for (int i = 0; i < dmd.height(); i++) { for (int j = 0; j < dmd.width(); j++) { dmd.drawPixel(j,i , fg); gecikme(30); } } // ekranı temizle dmd.fillScreen(bg);
> 
> }` [image]
> 
> bu hatayı alıyorum

board707, 2023-04-03:
Ben Turkce bilmiyorum...

Please test the pattern with panel chaining and fonts.

miralay001, 2023-04-03:
works well thanks

https://user-images.githubusercontent.com/108148338/229604860-df223c50-3a95-448d-840c-40f51efd6b38.mp4

closed 2023-04-04 by miralay001