board707/DMD_STM32 #36: how can I get height of characters in the font?

mucahitturk, 2023-04-05:
I want to learn character's height of the font. How can I do this?

mucahitturk, 2023-04-05:
I add uint8_t fontHeight(); in DMD_STM32a.h and add 
/*--------------------------------------------------------------------------------------
		height of selected font
--------------------------------------------------------------------------------------*/
uint8_t DMD::fontHeight()
{
	return (uint8_t)Font->get_height();
}
in DMD_STM32a.cpp.So it works for me.

mucahitturk, 2023-04-05:
What would be the difference if we change the 700 number value in the function we use as dmd.init(700).

There is the dmd.clearScreen() function, in addition to this, what if we had a function to delete the line we want?

Finally, now we only allow one line to shift, can we shift multiple lines, such as line deletion that I talked in the above?

board707, 2023-04-05:
> What would be the difference if we change the 700 number value in the function we use as dmd.init(700).

Hi
This parameter was left from the old library and its value depends on the type of matrices.
For monochone P10 panels this number controls a scan period duration in microseconds, but for RGB it determine the panel update rate in frames per second.
I am not recommended to use this, it can be changed in next releases.  

> what if we had a function to delete the line we want?

The easiest way to delete a part of screen - draw on it a rectangle with background color.

mucahitturk, 2023-04-06:
Okay, Thank you for your suggestions.
How can I create my font. Do you have a any program to create?

mucahitturk, 2023-04-06:
Is it possible for me to write with different fonts in different parts of the panel?

board707, 2023-04-06:
yes, it possible, see example:
```
//first string at 0,0
dmd.selectFont(&Arial_black);
dmd.drawStringX(0, 0, "TEST 12345", color); 

//second string at 10,10
dmd.selectFont(&Arial_14);
dmd.drawStringX(10, 10, "Hello", color2); 
```

mucahitturk, 2023-04-06:
ok thank you for your answer.
I have a bootloader for stm32f103c8t6. But I could not change the start address of the system. Do you have any idea for that? I want to use bootloader to upload system at anytime.

board707, 2023-04-07:
It seems not a problem with library

mucahitturk, 2023-04-12:
I have two different P10 panels. One has through-hole mounted LEDs with a 74HC595A on top, while the other has SMD mounted LEDs with an SM16188B driver. I am able to control the through-hole mounted LED panel with the 74HC595A, but I am unable to control the SMD mounted LED panel with the SM16188B. What can I do to be able to drive the P10 panel with SMD LEDs?

mucahitturk, 2023-04-12:
If feasible, may I integrate SM16188B driver into this library by myself? If I intend to make modifications, what changes or efforts shall I perform and where?

board707, 2023-04-12:
The library is open source, you can modify it however you want.
I don’t know what exactly needs to be fixed, since I didn’t go into the details of how this driver works.

mucahitturk, 2023-04-14:
Thank you for your help.

mucahitturk, 2023-04-24:
I have developed software that works with SPI_Monochrome, as well as a separate application for SM16188 integrated circuit on Arduino IDE. I am interested in integrating these two devices to work on the same system. Can I achieve this using an interface-based or inheritance-based approach? If you would like, I can share the code I have written for the SM16188 chip with you.

board707, 2023-04-24:
> Can I achieve this using an interface-based or inheritance-based approach?

It is difficult to say without a seeing the code.

> If you would like, I can share the code I have written for the SM16188 chip with you.

It would be very interesting.

closed 2023-06-22 by board707