board707/DMD_STM32 #38: P6 32x32 qiangli New MATRIX TYPES

maxmurugan, 2023-05-11:
Hi sir 

I am now using new p6 32x32 matrix with 1/8 scan
please check module picture and code

/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32 and RP2040 boards

Pattern test for 32x16 1/2 1/4 RGB panels
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
//Number of panels in x and y axis
#define RGB32x32_S8_maxmurugan    3,32,32,8,1 // 32x32 1/8 matrix from @maxmurugan

#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
uint16_t bg = 0; // background - black
uint16_t fg = 0; // foreground

void setup(void)
{
dmd.init();
fg = dmd.Color888(0, 0, 255);
dmd.setBrightness(50);
}

void loop(void) {
// fill the matrix with points row by row
for (int i = 0; i < dmd.height(); i++) {
for (int j = 0; j < dmd.width(); j++) {
dmd.drawPixel(j,i, fg);
delay(40);
}
}
// clear the screen
dmd.fillScreen(bg);
}

https://github.com/board707/DMD_STM32/assets/4962604/7f73f4e1-05f3-4515-b2d6-9ae47f0c0337

[image]

board707, 2023-05-11:
Hi again :)

By info from the panel sticker, your matrix contains a two IC : DP5125C + DP32020
I didn't found any data about DP5125 driver, but DP32030 is stated as 595 type (shift register) decoding 8-channel display driver chip
So you need to try my  DMD_RGB_SHIFTREG_ABC class.

board707, 2023-05-11:
In your code above
1. Change the matrix define to:
```
#define RGB32x32_S8_maxmurugan2   3,32,32,8,0 // 32x32 1/8 
```
2. Call the matrix class as
```
 DMD_RGB_SHIFTREG_ABC <RGB32x32_S8_maxmurugan2, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

Leave other code not changed, run the test and show the results.

board707, 2023-05-11:
something goes wrong...

maxmurugan, 2023-05-11:
how to check?

board707, 2023-05-11:
The main driver DP5125 is the same as in your [issue #23](https://github.com/board707/DMD_STM32/issues/23), which we finally were able to run correctly.
The other IC Dp32020 is a 595 (SHIFTREG) type multiplexor, according to the info from the manufacturer site [www.depuw.com](http://www.depuw.com/productlist_50_86.html).

So everything should work with this config. Please double check all connections, especially the connections of pins A B C on the panel and the corresponding pins on the RP Pico board.

If possible, ask the seller for datasheets for IC chips

maxmurugan, 2023-05-11:
Issue#23  same wiring connection used

board707, 2023-05-11:
Thank you.
Give me time to check one guess....

board707, 2023-05-11:
Unfortunately, I have found that shift multiplexing currently doesn't work with Raspberry Pico.
 It will take me a few days to fix this.

maxmurugan, 2023-05-12:
ok sir

board707, 2023-05-12:
Please replace two files in the library with files from archive and try [the last test](https://github.com/board707/DMD_STM32/issues/38#issuecomment-1543447930) again

maxmurugan, 2023-05-13:
https://github.com/board707/DMD_STM32/assets/4962604/c7df0b81-fe10-4b37-b930-4b8cf8afb7c5

maxmurugan, 2023-05-13:
please check

board707, 2023-05-13:
Thank you, 
Its great, the picture means that the shift_reg multiplexing now works.

Now we need to find the coordinate transformation, this may take time.

maxmurugan, 2023-05-13:
ok sir than you.

board707, 2023-05-14:
Hi
We'll have to do some tests.
Please replace one file in the library to file from archive.
Then add this lines to the beginning of the sketch:

```
#define RGB32x32_S8_maxmurugan3   3,32,32,8,1 // 32x32 1/8 
#define RGB32x32_S8_maxmurugan4   3,32,32,8,2 // 32x32 1/8 
#define RGB32x32_S8_maxmurugan5   3,32,32,8,33 // 32x32 1/8 
```

and run test foe every of this matrix defines, show the results.
Thank you
[rp2040_shiftreg_2.zip](https://github.com/board707/DMD_STM32/files/11471682/rp2040_shiftreg_2.zip)

maxmurugan, 2023-05-15:
#define RGB32x32_S8_maxmurugan3   3,32,32,8,1 // 32x32 1/8 

https://github.com/board707/DMD_STM32/assets/4962604/b3b02d35-4337-4dac-9859-a6dbae75a66f

maxmurugan, 2023-05-15:
#define RGB32x32_S8_maxmurugan4   3,32,32,8,2 // 32x32 1/8 

https://github.com/board707/DMD_STM32/assets/4962604/ddc277fe-b081-4b83-bd5c-88acd818d61f

maxmurugan, 2023-05-15:
#define RGB32x32_S8_maxmurugan5   3,32,32,8,33 // 32x32 1/8 

https://github.com/board707/DMD_STM32/assets/4962604/f145f2bb-5843-4354-97c8-0beeb52f3638

maxmurugan, 2023-05-15:
sorry for the delay response

maxmurugan, 2023-05-15:
I try your last option  #define RGB32x32_S8_maxmurugan5   3,32,32,8,33 // 32x32 1/8 

working but more then 1 panel scrolling issue

please check video

https://github.com/board707/DMD_STM32/assets/4962604/dc6d1d3b-69e5-48d5-a468-5ab76e2c63bb

 file

board707, 2023-05-15:
> working but more then 1 panel scrolling issue

Sorry, last time I posted a wrong file. Test with this one and use this option:
```
 #define RGB32x32_S8_maxmurugan5 3,32,32,8,33 // 32x32 1/8
```
[rp2040_shiftreg_3.zip](https://github.com/board707/DMD_STM32/files/11474817/rp2040_shiftreg_3.zip)

board707, 2023-05-15:
test this
[rp2040_shiftreg_3_1.zip](https://github.com/board707/DMD_STM32/files/11475228/rp2040_shiftreg_3_1.zip)

maxmurugan, 2023-05-15:
template<int COL_DEPTH>
class DMD_RGB<RGB32x32_S8_maxmurugan, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
	{
	    this->fast_Hbyte = false;
	    this->use_shift = false;
  	}
  // Fast text shift is disabled for complex patterns, so we don't need the method
  void disableFastTextShift(bool shift) override {}
protected:
uint16_t get_base_addr(int16_t &x, int16_t &y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;

		static const uint8_t A_tbl[16] = { 0,1,2,3,4,5,6,7,
                                                   0,1,2,3,4,5,6,7 };
		static const uint8_t B_tbl[16] = { 1,1,1,1,1,1,1,1,
                                                    0,0,0,0,0,0,0,0 };
		static const uint8_t C_tbl[8] = { 0,1,2,3,4,5,6,7 };
		static const uint8_t Pan_Okt_cnt = this->DMD_PIXELS_ACROSS / 8;
		uint8_t Oktet_m = B_tbl[pol_y] * Pan_Okt_cnt + (x / 8) % Pan_Okt_cnt;

		uint16_t base_addr = this->x_len * A_tbl[pol_y] + (x / this->DMD_PIXELS_ACROSS) * this->DMD_PIXELS_ACROSS * this->multiplex + C_tbl[Oktet_m] * 8;
		base_addr += x % 8;
		return base_addr;
	}

};

this working morethen one module

board707, 2023-05-15:
You are right, thank you!

maxmurugan, 2023-05-15:
ok sir
thangs a lot

board707, 2023-05-15:
Fixed in v1.0.6

closed 2023-05-15 by board707