board707/DMD_STM32 #41: Some time display going stuck

maxmurugan, 2023-06-29:
hi sir,

P6 RGB or P10 Red color module  display issue showing 
at the same time serial port data send receiving working

please  check picture.

after poweroff and on going to normal mode ,how to solve this issue

[image]

board707, 2023-06-29:
Sorry, I don't understand.
Please explain more clearly, what is a problem

maxmurugan, 2023-06-29:
this is p6 module using PICO board
[image]

maxmurugan, 2023-06-29:
working fine after few hour display showing garbage please check video.

https://github.com/board707/DMD_STM32/assets/4962604/797d4938-7323-412a-8c12-5b2e466a39c0

maxmurugan, 2023-06-29:
At the **garbage** time Restart the controller going to  normal working

maxmurugan, 2023-06-29:
same issue I will face p10 red color module.

board707, 2023-06-29:
Good afternoon
Too little information to give advice
This behavior can be caused by many reasons - errors in the code, power problems, some kind of electromagnetic interference.
For example, one of the options is memory problems in the library code or in the main program. Could you please provide a example code, where the problem is observed.

maxmurugan, 2023-06-29:
please check

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include <Wire.h>
#include <EEPROM.h>
#include <IRremote.h>
#include "DMD_RGB.h"
#define RGB32x32plainS8 34 // 32x32 1/8
//###########################################################################################

template<int COL_DEPTH>
class DMD_RGB<RGB32x32plainS8, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(3, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, 8, 32, 32)
  {}
  protected:
  uint16_t get_base_addr(int16_t x, int16_t y) override {
  this->transform_XY(x, y);
  uint8_t pol_y = y % this->pol_displ;
  x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
  uint16_t base_addr = (pol_y % this->nRows) * this->x_len + (x / this->DMD_PIXELS_ACROSS) * this->multiplex * this->DMD_PIXELS_ACROSS;
  if (pol_y / this->nRows) base_addr += x % this->DMD_PIXELS_ACROSS ;
  else base_addr += x % this->DMD_PIXELS_ACROSS + this->DMD_PIXELS_ACROSS;
  return base_addr;
}

};
//################################################################################
byte second1, minute1, hour1, dayOfWeek1, dayOfMonth1, month1,year1;
char a[15];
char b[15];
// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#include "st_fonts/max32.h"
#include "st_fonts/Font6x16.h"
#include "st_fonts/Font7x16.h"
#include "st_fonts/Basket_16.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/FreeSans9pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1

#define ENABLE_DUAL_BUFFER true
#define ARDUINO_ARCH_RP2040

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
[... 201 lines cut ...]
  Wire.beginTransmission(111);
  Wire.write(0);
  Wire.write(decToBcd(second1) | 0x80);
  Wire.write(decToBcd(minute1));
  Wire.write(decToBcd(hour1));
  Wire.write(decToBcd(dayOfWeek1) | 0x08);
  Wire.write(decToBcd(dayOfMonth1));
  Wire.write(decToBcd(month1));
  Wire.write(decToBcd(year1));
  Wire.endTransmission();
}
void getDate_MCP7940()
{
  Wire.beginTransmission(111);
  Wire.write(0);
  Wire.endTransmission();
  Wire.requestFrom(111,7);
  second1     = bcdToDec(Wire.read() & 0x7f);
  minute1     = bcdToDec(Wire.read());
  hour1       = bcdToDec(Wire.read() & 0x3f);
  dayOfWeek1  = bcdToDec(Wire.read() & 0x07);
  dayOfMonth1 = bcdToDec(Wire.read());
  month1      = bcdToDec(Wire.read());
  year1       = bcdToDec(Wire.read());
}

board707, 2023-06-29:
I am not sure is this a cause of the problem or not, but there are errors in the code.
At the end of the loop()

```
dmd.drawString(17,0,a,sizeof(a),col[1]);
```
The using of `sizeof()` is incorrect in this case, it returns the size of array rather than a length of string. Use the `strlen()` instead:
```
dmd.drawString(17,0,a,strlen(a),col[1]);
```
There are a several lines with the same error.

Next, these two methods are wrong:
```
byte decToBcd(byte val)
{
return ( (val/1016) + (val%10) );
}
byte bcdToDec(byte val)
{
return ( (val/1610) + (val%16) );
}
```
Since the calculation is carried out in integers, the first expression, `(val/1610)`, is always zero and whole result will be `val % 10` in first function and `val %16` in second.

maxmurugan, 2023-06-30:
thanks sir
i will change this code and check

maxmurugan, 2023-06-30:
this is copy past issue

byte decToBcd(byte val)
{
  return ( (val/10*16) + (val%10) );
}
byte bcdToDec(byte val)
{
  return ( (val/16*10) + (val%16) );
}

maxmurugan, 2023-06-30:
(val/10x16)

maxmurugan, 2023-07-01:
i get new error in p10 red module 

https://github.com/board707/DMD_STM32/assets/4962604/64da251a-9b4d-4ea9-bb7d-ce3d5c0d753e

please check and give the solution sir

maxmurugan, 2023-07-01:
#include "DMD_Monochrome_Parallel.h"
#include "Arduino.h"
#include "Wire.h"
#include "uEEPROMLib.h"
#include "DHT.h"
//###########################################################################################
// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#include "st_fonts/Font7x16.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#define DHTPIN 3
#define PLOY 0xA001
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER true
#define DMD_PIN_A 10
#define DMD_PIN_B 11
#define DMD_PIN_nOE 14
#define DMD_PIN_SCLK 15
uint8_t pins[] = { 6, 7, 8 };  // CLK , row1, row 2
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(SystemFont5x7);
DMD_Standard_Font font7x16(Font7x16);
// uEEPROMLib eeprom;
uEEPROMLib eeprom(0x50);
const int dimmerValues[13] = {10, 25 , 50, 64 , 80, 100 , 120 ,140, 250, 500, 750, 1000,1024};
//####  DHT22 #############
#define DHTTYPE DHT21
DHT dht(DHTPIN, DHTTYPE);
unsigned long DHT_interval=2500; // the time we need to wait
unsigned long DHT_previousMillis=0; // millis() returns an unsigned long.
unsigned long DHT_currentMillis;
float h;
float t;
float T;
float H;
int humidity = 0;
int temperature = 0;
String Dht_1;
String Dht_2;
char Dht_a[20] ="0";
char Dht_b[20] ="0";
//#########################
byte Sop_packet =0;
int str =0;
char a1[20] ="0";
int brightness = 4;
byte receivedChars[100];
boolean newData = false;
byte ndx = 0;
byte rc;
char Message[255];
uint8_t Message_pos = 0;
char Read_Data;
byte ON = 0;
unsigned long Adc_1A;
String Adc_1;
String Adc_2;
String Adc_3;
String Adc_4;
String Adc_5;
String Adc_6;
String Add_Register1;
String Add_Register2;
String Add_Register3;
unsigned int Register_1;
unsigned int Register_2;
[... 320 lines cut ...]
                }
               delay(5);
            }
            else if(Message[0]=='0'&&Message[1]=='1'&&Message[2]=='h')
            {
              Adjust_Humi = ((Message[3])*1000)+((Message[4]-'0')*100)+((Message[5]-'0')*10)+Message[6]-'0';
               if (!eeprom.eeprom_write(42, Adjust_Humi)) {
                 Serial.println("Failed");
                } 
                else {
                  Serial.println("Adjust_Humi OK");
                }
               delay(5);
            }
        }
         delay(5);
     }

      
      else if(Serial.read() == '[') 
       {
         recvInProgress = true;
       }
  }
}

maxmurugan, 2023-07-01:
[DHT.txt](https://github.com/board707/DMD_STM32/files/11924864/DHT.txt)

maxmurugan, 2023-07-01:
please check  txt file code sir

maxmurugan, 2023-07-01:
Showing junk display at this time PICO Serial data received

board707, 2023-07-01:
Sorry but your code seems to have a some errors:

What are you try to achieve in this line?
```
 if((Adjust_Temp/1000)== '+')   {
```
your Adjust_Temp is integer, comparing it with char not making sense for me

This line is not the one with errors...

If you find a bug in a library, please prepare a short code example where the bug appears. I don't have the ability to run your entire project with modbus, serial and sensors.

maxmurugan, 2023-07-03:
i am trying to find this bug ,every 2 to 3 hour create this issue

maxmurugan, 2023-07-03:
please need help

every 1 hour, how to restart your library  or Software restart for raspberry pi pico

board707, 2023-07-03:
> how to restart your library or Software restart for raspberry pi pico

Hi
To soft restart a Pico board you can use a watchdog feature:
https://github.com/raspberrypi/pico-examples/blob/master/watchdog/hello_watchdog/hello_watchdog.c

maxmurugan, 2023-07-07:
I   completely remove DHT library code ,Now no error on the led display
thanks for your support

maxmurugan, 2023-07-07:
I have another one question

i have custom led board 74 hc595 and uln2803 driver ic

1/2 scan method it is possible drive your library

16x48 pixel

board707, 2023-07-07:
> have custom led board 74 hc595 and uln2803 driver ic

Uln2803 driver? Are you sure? It is not common for LED panels using this driver.
Is the panel has a HUB75 connector ?
Please show the pictures

maxmurugan, 2023-07-07:
it is  old board design  using railway application
[image]

maxmurugan, 2023-07-07:
1 and 9 column connected 16x48  total 6 74hc595 ic 
8 and 16 connected

tomorrow will send picture

maxmurugan, 2023-07-07:
Right side is first input for 74hc595

board707, 2023-07-07:
> total 6 74hc595 ic

It seems to me that for 16x48 panel it should be 12 74hc595 ic in total.

What for do you plan to use this panels? Without major code modification, they will not work with this library. I could add support for these panels to the code if I had them available. Without panels for tests, this is hardly possible.

Why would you like to use exact this panels? Do you have dozens or hundreds? Apparently, this is a very old design and it is unlikely that anyone will actively use them. If you have only a few panels, it is hardly worth spending so much effort on them, it is more profitable to buy modern ones.

maxmurugan, 2023-07-11:
ok thanks

closed 2023-07-11 by maxmurugan