board707/DMD_STM32 #42: How to connect a single Monochrome P10 for Pico

donly, 2023-07-07:
I just follow the wiki [Connections Monochrome](https://github.com/board707/DMD_STM32/wiki/Connections---Monochrome) and use [Arduino Pico](https://github.com/earlephilhower/arduino-pico) to follow the pin connections in the example [dmd_monochrome](https://github.com/board707/DMD_STM32/blob/dev-V2/examples/STM32F4_RP2040/dmd_monochrome/dmd_monochrome.ino#L68C1-L68C1).
When I use the connection shown in the diagram below and upload the code, the LED screen does not work and remains black.
[image]
[image]
SO, where is the problem please?

board707, 2023-07-07:
Hi! 
The pins on your picture is not labeled, so it making it difficult to understand the exact pin mappings. Could you please show a pin definition part of your code?

You can see a tested variant of RP2040 pins in the `examples/STM32F4_RP2040/dmd_monochrome/dmd_monochrome.ino` example:
```
#define DMD_PIN_A 1
#define DMD_PIN_B 2
#define DMD_PIN_nOE 14
#define DMD_PIN_SCLK 15

// These pins must be consecutive in ascending order
uint8_t pins[] = { 6, 7, 8 };  // CLK , row1, row 2
```

donly, 2023-07-07:
Yes, I use the same code with you.
And the pins I used is same with the first picture.

Which two do pin1 and pin2 refer to in this picture https://github.com/board707/DMD_STM32/raw/old-V1/.github/hub12.svg

board707, 2023-07-07:
> Which two do pin1 and pin2 refer to

The pin 1 & 2 are refer to A & B signals.
I will assemble a test circuit and check this example with Raspberry Pico again.

donly, 2023-07-07:
I am so sorry, what I want to ask the `7, 8` pins (row1, row2) refer to which pins.

board707, 2023-07-07:
Hi
The pins 7, 8... are related to panels configuration with many horizontal rows, as described [in Wiki ](https://github.com/board707/DMD_STM32/wiki/Connections---Monochrome#parrallel-connection---dmd_monochrome_parallelh). Each pin is connected to panel R_DATA pin of corespondent's row.
If you drive a single panel, you need only one CLK pin and one R_DATA pin.

I just check the library example, it works fine for me with pins, listed below:
```
#define DMD_PIN_A 1
#define DMD_PIN_B 2
#define DMD_PIN_nOE 14
#define DMD_PIN_SCLK 15

// These pins must be consecutive in ascending order
uint8_t pins[] = { 6, 7 };  // CLK , R_DATA
```
Also please make sure the line
`#define DMD_PARA`
 at the beginning of the sketch (line 8) is not commented out.

donly, 2023-07-08:
```
// These pins must be consecutive in ascending order
uint8_t pins[] = { 6, 7 };  // CLK , R_DATA
```
This is the key, now it works.
Thank you @board707, have a nice weekend.

closed 2023-07-08 by donly