board707/DMD_STM32 #43: Bluepill with GKGD P6 display issue

phalgunimondal, 2023-07-11:
I connected P6 outdoor LED module with Bluepill STM32 but its showing two line on display 

/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// Fonts includes
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
char a[10];
char b[10];
// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB_SHIFTREG_ABC <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// - 32x16 matrix with 8scans
// - 80x40 matrix with 20scans
// - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_black(SystemFont5x7);
// GFX font with sepatate parts for Latin and Cyrillic chars
//DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
UTF8 char recoding

-------------------------------------------------------------------------------------*/

// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255), // blue
dmd.Color888(0, 255, 255), // blue
dmd.Color888(0, 0, 0), // blue
};

void setup(void)
{
Serial.end();
// initialize DMD objects
dmd.init();
Serial.begin(9600);
Serial.println("PICO P4 RTC");
delay(1000);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
uint16_t bg = 0; // background - black
uint16_t fg = dmd.Color888(0, 255, 0);; // foreground
// draw pixels
for (int i = 0; i < dmd.height(); i++) {
for (int j = 0; j < dmd.width(); j++) {
dmd.drawPixel(j,i, fg);
delay(30);
}
}
// clear the screen
dmd.fillScreen(bg);

}

phalgunimondal, 2023-07-11:
I have P6 3232-2727-8S-v3.0 (GKGD)

phalgunimondal, 2023-07-11:
> Hi
> 
> Are you sure that your panel has multiplexor of SHIFTREG type ? Try to replace DMD_RGB_SHIFTREG_ABC class to the DMD_RGB, use the initialization below:
> 
> ```
> DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
> ```

I have multiplexor with "MW245" ICs on LED display. So its multiplexor.

[image]

board707, 2023-07-11:
Please record a video of the code working in your panel ( at least 2-3 lines of pixels) and insert the link to the message.

board707, 2023-07-11:
Thank you, but your video is not very informative. It would be better to record the moment when the picture on the panel changes.

Also, please find the idnumbers of all chips on the back of the panel. There must be at least three types of them.

phalgunimondal, 2023-07-12:
Hi,
 Now i am able to print text on P6 display :) but RED color text (pixel) not showing .
problem was in pin selection i replace PA15 to PA6 

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32 and RP2040 boards

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"
#include "st_fonts/SystemFont5x7.h"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

/* =================== *
*     STM32F4 pins     *
*  =================== */
#if (defined(__STM32F1__) || defined(__STM32F4__))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB9
#define DMD_PIN_D PB4
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C  };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

/* =================== *
*     RP2040 pins     *
*  =================== */
#elif (defined(ARDUINO_ARCH_RP2040))
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = { 11, 0,1,2,3,4,5 }; // CLK, R0, G0, B0, R1, G1, B1
#endif

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 Direct mux
// RGB32x16plainS2         - 32x16 1/2 Binary mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Direct mux

DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_Standard_Font Arial_black(SystemFont5x7);
uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground
// create foreground colors
uint16_t col[] = {
dmd.Color888(255,0, 0), // red
dmd.Color888(0, 255, 0), // green
dmd.Color888(0, 0, 255), // blue
dmd.Color888(0, 255, 255), // blue
dmd.Color888(0, 0, 0), // blue
dmd.Color888(255, 255, 255), // blue
};
void setup(void)
{
    dmd.init();
    fg = dmd.Color444(255, 0, 0);
    dmd.setBrightness(255);
}

void loop(void) {
    dmd.selectFont(&Arial_black);
dmd.setBrightness(255);
dmd.setTextColor(0,0);
while (1) 
{
  

          dmd.drawString(32,0,"TEST",4,col[2]);
          dmd.drawString(32,8,"DISPLAY",7,col[2]);
          dmd.drawString(0,16,"Phalguni",8,col[0]);
          dmd.drawString(0,24,"Mondal",6,col[1]);
 }

}

Result Video:
https://github.com/board707/DMD_STM32/assets/22762672/5c559777-12a6-49c9-8c09-39acf63bc16c

board707, 2023-07-12:
Please, change the color of the all texts to red. Do not change anything else in the code. Check the picture.

phalgunimondal, 2023-07-12:
Nothing text display on P6 . I checked power supply 5V/20A.
No luck 

https://github.com/board707/DMD_STM32/assets/22762672/6c7cefb5-0cd7-4c9a-81f7-2e1f51463f09

davemaster, 2023-07-12:
> Nothing text display on P6 . I checked power supply 5V/20A. No luck
> WhatsApp.Video.2023-07-12.at.5.45.45.PM.mp4

Greetings,

Sometimes, the solution is start from the beginning. If You say your power supply it's ok... start with ONLY ONE RGB  module, and PLEASE SHOW HERE your pinout (picture), again, check it, some pins can't be used as you wish.

Best regards

board707, 2023-07-12:
By the way, why did you replaced `dmd.Color888()` method with

 > dmd.Color444(255,0, 0), // red

in your code?

Probably it is not a cause of your issue, but it is a syntactically incorrect.
Also with line:

> dmd.setTextColor(0,0);

it should be
 
```
dmd.setTextColor(col[0],0);
```

phalgunimondal, 2023-08-17:
Hi all,
problem was in my hardware pinout .
thanks for help

closed 2023-08-17 by phalgunimondal