board707/DMD_STM32 #44: P10 RGB Panel new Scan

maxmurugan, 2023-07-28:
hi sir 
this new p10 module qiangli 1/4 th scan
/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32 and RP2040 boards

Pattern test for 32x16 1/2 1/4 RGB panels
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
//Number of panels in x and y axis
#define RGB32x16plainS4   2,32,16,4,1
//#define RGB32x32_S8_maxmurugan2   3,40,20,5,1 // 32x32 1/8
//#define RGB32x32_S8_maxmurugan3   3,32,32,8,1 // 32x32 1/8 
//#define RGB32x32_S8_maxmurugan4   3,32,32,8,2 // 32x32 1/8 
//#define RGB32x32_S8_maxmurugan5   3,32,32,8,33 // 32x32 1/8 

#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12
uint8_t custom_rgbpins[] = { 11, 16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
//DMD_RGB <RGB32x32_S8_maxmurugan2, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB_SHIFTREG_ABC <RGB32x16plainS4, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
uint16_t bg = 0; // background - black
uint16_t fg = 0; // foreground

void setup(void)
{
dmd.init();
fg = dmd.Color888(0, 0, 255);
dmd.setBrightness(50);
}

void loop(void) {
// fill the matrix with points row by row
for (int i = 0; i < dmd.height(); i++) {
for (int j = 0; j < dmd.width(); j++) {
dmd.drawPixel(j,i, fg);
delay(40);
}
}
// clear the screen
dmd.fillScreen(bg);
}

[image]

https://github.com/board707/DMD_STM32/assets/4962604/f7e32387-820e-41a8-b051-8db68dec271a

board707, 2023-07-28:
Hello!

Please test it with DMD_RGB class using panel templates 
```
#define RGB32x16plainS4_pt2		2,32,16,4,2	
```
or
```
#define RGB32x16plainS4_pt2_DIRECT	4,32,16,4,2	
```

maxmurugan, 2023-07-31:
now i am using below setting

https://github.com/board707/DMD_STM32/assets/4962604/7a81fd42-8f12-44e1-9811-5c32ba3b6461

#define RGB32x16plainS4_pt2    2,32,16,4,2 
DMD_RGB <RGB32x16plainS4_pt2, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

maxmurugan, 2023-08-01:
thanks for your support

closed 2023-08-01 by maxmurugan

board707, 2023-08-01:
Sorry for the delay in reply. I'm glad everything worked.

TechTalkies, 2024-01-01:
I have this same pattern board but the above mentioned solution does not work now. Please let me know if it has changed.  Pattern 56 mentions 1/5 scan panel now.

"RGB32x16_S4_MAX " is not defined in the library and if I define it myself, I get the following error.

Compilation error: no matching function for call to 'DMD_RGB<2, 32, 16, 4, 56, 4>::DMD_RGB(uint8_t [5], int, int, uint8_t [7], int, int, bool)'

board707, 2024-01-07:
Please test this:
```
#define RGB32x16_S4_TT 2,32,16,4,63
template<int COL_DEPTH>
class DMD_RGB< RGB32x16_S4_TT, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
	{
	public:
		DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
			byte panelsWide, byte panelsHigh, bool d_buf = false) :
			DMD_RGB_BASE2<COL_DEPTH>(2, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 4, 32, 16)
			{
			this->fast_Hbyte = false;
			this->use_shift = false;
			}
		// Fast text shift is disabled for complex patterns, so we don't need the method
		void disableFastTextShift(bool shift) override {}

	protected:
		uint16_t get_base_addr(int16_t& x, int16_t& y) override {
			this->transform_XY(x, y);
			uint8_t pol_y = y % this->pol_displ;
			x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
			uint16_t base_addr = (pol_y % this->nRows) * this->x_len + (x / 4) * this->multiplex * 4 + x %4;
                        if (pol_y < this->nRows) base_addr+= 4;
			return base_addr;
			}
	};
```

maxmurugan, 2024-01-08:
https://github.com/board707/DMD_STM32/assets/4962604/a367a889-a599-40b2-ba0c-75e224a83bf
hi sir,
tested above code

maxmurugan, 2024-01-08:
https://github.com/board707/DMD_STM32/assets/4962604/eec57ced-1528-4ad0-a62c-0bd5188367b7

maxmurugan, 2024-01-08:
Uploading IMG_8724.MOV…

maxmurugan, 2024-01-08:
https://github.com/board707/DMD_STM32/assets/4962604/bdc6da42-ebf2-451d-aabe-44a0f1a494ed