board707/DMD_STM32 #45: Circles/ellipses support

Nikita-bunikido, 2023-08-02:
Library does support drawing rectangles: ```void drawFilledBox(int x1, int y1, int x2, int y2, uint16_t color);```, but doesn't support drawing circles yet. So I came up with this solution:

``` C
typedef struct vec2 { int16_t x, y; } vec2;
typedef uint16_t Rgb565; 

...

static void put_circle(vec2 center, int16_t r, Rgb565 color) {
    for (vec2 pos = { center.x - r, center.y - r }; pos.y < center.y + r; pos.y ++)
    for (pos.x = center.x - r; pos.x < center.x + r; pos.x ++)
        (void)((distance(pos, center) < r) && (*get_screen(pos) = color));
}
```

Assuming blue background, ```put_circle((vec2){5, 5}, 5, 0xf81f);``` produces this:

[image]

It would be nice to have such function in a library as a method like ```drawFilledBox()```

board707, 2023-08-02:
Hi, thamks for your interest.

The library inherits drawing methods from AdafruitGFx library. Did you try to use the circle drawing function from there?

Nikita-bunikido, 2023-08-16:
> Did you try to use the circle drawing function from there?

Thank you, found this:
```void fillCircle (int16_t x0, int16_t y0, int16_t r, uint16_t color);```

Works well for me!

closed 2023-08-16 by Nikita-bunikido