board707/DMD_STM32 #46: Error in STM32F401RC6 compilation

phalgunimondal, 2023-08-17:
I am using **STM32F401RC6** controller and this core https://github.com/stm32duino/Arduino_Core_STM32
not sure why below compilation error comes

**arduino-cli compile -b STMicroelectronics:stm32:GenF4 dmd_monochrome**

In file included from C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.h:24,
                 from C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:9:
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:40: error: **'voidFuncPtr' has not been declared**
  235 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ^~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:264:9: error: 'PortType' does not name a type
  264 |         PortType clk_clrmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:265:9: error: 'PortType' does not name a type
  265 |         PortType clkmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:275:9: error: 'OE_PWM_Polarity' does not name a type
  275 |         OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
      |         ^~~~~~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp: In constructor 'DMD_RGB_BASE::DMD_RGB_BASE(byte, uint8_t*, byte, byte, uint8_t*, byte, byte, bool, uint8_t, uint8_t, byte, byte)':
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:32:9: error: 'OE_polarity' was not declared in this scope
   32 |         OE_polarity = OE_PWM_NEGATIVE;
      |         ^~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:32:23: error: 'OE_PWM_NEGATIVE' was not declared in this scope
   32 |         OE_polarity = OE_PWM_NEGATIVE;
      |                       ^~~~~~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp: In member function 'virtual void DMD_RGB_BASE::init(uint16_t)':
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:167:27: error: invalid conversion from 'void (*)()' to 'int' [-fpermissive]
  167 |         initialize_timers(scan_running_dmd_R);
      |                           ^~~~~~~~~~~~~~~~~~
      |                           |
      |                           void (*)()
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:52: note:   initializing argument 1 of 'virtual void DMD::initialize_timers(int)'
  235 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ~~~~~~~~~~~~^~~~~~~
In file included from C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.cpp:27:
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:40: error: 'voidFuncPtr' has not been declared
  235 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ^~~~~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:264:9: error: 'PortType' does not name a type
  264 |         PortType clk_clrmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:265:9: error: 'PortType' does not name a type
  265 |         PortType clkmask = 0;
      |         ^~~~~~~~
C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:275:9: error: 'OE_PWM_Polarity' does not name a type
  275 |         OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
      |         ^~~~~~~~~~~~~~~

Used library Version Path
SPI          1.0.0   C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SPI
SrcWrapper   1.0.1   C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SrcWrapper

Used platform            Version Path
STMicroelectronics:stm32 2.6.0   C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0
Error during build: exit status 1

board707, 2023-08-17:
see the Wiki
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

closed 2023-09-15 by board707

vahoo22, 2023-09-17:
> Я использую контроллер **STM32F401RC6** и это ядро [​​https://github.com/stm32duino/Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) , не знаю, почему возникает ошибка компиляции ниже.
> 
> **arduino-cli компиляция -b STMicroelectronics:stm32:GenF4 dmd_monochrome**
> 
> В файле, включенном из C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.h:24, из C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040 \dmd_monochrome\DMD_RGB.cpp:9: C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:40: ошибка: 'voidFuncPtr' не **объявлен** 235 | виртуальный void Initialize_timers (обработчик voidFuncPtr); | ^~~~~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:264:9: ошибка: «PortType» не называет тип 264 | Тип порта clk_clrmask = 0; | ^~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:265:9: ошибка: «PortType» не называет тип 265 | Типпорта clkmask = 0; | ^~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:275:9: ошибка: «OE_PWM_Polarity» не называет тип 275 | OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE; | ^~~~~~~~~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp: В конструкторе 'DMD_RGB_BASE::DMD_RGB_BASE(byte, uint8_t*, байт, байт, uint8_t*, байт, байт, bool, uint8_t, uint8_t, байт, байт)': C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:32:9: ошибка: «OE_polarity» не была объявлена ​​в этой области 32 | OE_полярность = OE_PWM_NEGATIVE; | ^~~~~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:32:23: ошибка: в этом не было объявлено 'OE_PWM_NEGATIVE' объем 32 | OE_полярность = OE_PWM_NEGATIVE; | ^~~~~~~~~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp: В функции-члене 'virtual void DMD_RGB_BASE::init (uint16_t)': C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_RGB.cpp:167:27: ошибка: недопустимое преобразование из 'void ()()' в ' _int' [-fpermissive] 167 | инициализировать_таймеры (scan_running_dmd_R); | ^~~~~~~~~~~~~~~~~~~ | | | void (_ )() C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:235:52: примечание: инициализация аргумента 1 из 'virtual void DMD::initialize_timers(int) ' 235 | виртуальный void Initialize_timers (обработчик voidFuncPtr); | ~~~~~~~~~~~~~~~~~~~ В файле, включенном из C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.cpp:27: C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040 \ dmd_monochrome\DMD_STM32a.h:235:40: ошибка: «voidFuncPtr» не объявлен 235 | виртуальный void Initialize_timers (обработчик voidFuncPtr); | ^~~~~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:264:9: ошибка: «PortType» не называет тип 264 | Тип порта clk_clrmask = 0; | ^~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:265:9: ошибка: «PortType» не называет тип 265 | Типпорта clkmask = 0; | ^~~~~~~~ C:\Users\Phalguni\Downloads\DMD_STM32-dev-V2\examples\STM32F4_RP2040\dmd_monochrome\DMD_STM32a.h:275:9: ошибка: «OE_PWM_Polarity» не называет тип 275 | OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE; | ^~~~~~~~~~~~~~~
> 
> Используемая библиотека Версия Путь SPI 1.0.0 C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SPI SrcWrapper 1.0.1 C:\Users\Phalguni\AppData\Local \Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SrcWrapper
> 
> Used platform Version Path STMicroelectronics:stm32 2.6.0 C:\Users\Phalguni\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0 Error during build: exit status 1

hi, my friend,
returns to the project DMD_STM32 after compilation the same errors,
Do I have the same problem as you solved this problem?

vahoo22, 2023-09-17:
hi,  friend,
after compilation project DMD_STM32  the same errors, Do I have the same problem as you solved this problem?

board707, 2023-09-17:
Hi, you used incompatible stm32 core. To solve the issue please read my answer just above your message.