board707/DMD_STM32 #47: Low color//brightness problems

ndavi, 2023-08-21:
Hello !

Thanks you for this great library.

I have a problem with a P8 Outdoor display, everything seems to works fine, but, I can't have low colors for each pixels on the display.

It's like, when I'm making a RGB color of (15,15,15), the display is black instead of displaying the leds with low brightness. I can have low brightness on the panel by lowering the global brightness of the display, but I want to have low brightness for each pixel individually.

Maybe it's because of the color storage of 4 bits per channels ? So I if want to have high contrast it's not possible ?

Is there any workaround ?

Thanks you !

board707, 2023-08-22:
Hi
Thanks for your feedback.

The color function used as is:
```
uint16_t DMD_RGB_BASE::Color888(uint8_t r, uint8_t g, uint8_t b) {
	return ((uint16_t)(r & 0xF8) << 8) | ((uint16_t)(g & 0xFC) << 3) | (b >> 3);
}
```
so for your color(15, 15, 15) the result should not be a complete black, but a minimum brightness level (1, 1, 1)

I will test it in my panels in a few days and let you know the results.

ndavi, 2023-08-22:
Hello,

For this function yes, it will be a minimum brightness level, but

I think the result is a complete black because 15 in binary is 0000 1111.

So, for 4 bits colors, we only take the first 4 bits, so this loosing of precision is what put this panel in black, that's it ?

`
// Adafruit_GFX uses 16-bit color in 5/6/5 format, while matrix needs
// 4/4/4.  Pluck out relevant bits while separating into R,G,B:
r = c >> 12;        // RRRRrggggggbbbbb
g = (c >> 7) & 0xF; // rrrrrGGGGggbbbbb
b = (c >> 1) & 0xF; // rrrrrggggggBBBBb
`

Here his the re-conversion that is loosing precision.

Also, if for example, I put r = 1 in this function (The drawPixel function), I can't have the low brightness that I want for my screen panel. So, for me to have this working, I think that what I need is a full 8 bits colors precisions everywhere, without converting anything and loosing color precision.

Here the example :

The minimum color brightness that I can have on the panel by using dmd.Color888 and dmd.brightness at full

[image]

Here the minimum color level that I can have by setting the dmd.brightness to 1

[image]

Thanks you

board707, 2023-08-22:
Thanks for your error analysis, it seems that the conversion function needs some improvements. I need a time to fix it.

ndavi, 2023-08-22:
Thanks you for your time ! 

I would love to help but I don't know how it all works so I can't see how to improve it. 
I've switched from Smartmatrix to this library because it's seems more stable with my panels, but not for the colors sadly

board707, 2023-08-25:
Hi
I apologize for the delay in answering.
I re-analyzed the color conversion functions and came to the conclusion that there are no errors in them. You absolutely correctly wrote that if you take the 8-bit color value 15 (binary representation 0b00001111) and then convert it to 4-bit, the result will be zero. But  there is no error in this, it is the result of using 4-bit encoding.
I understand that this may not be suitable for your project, however, this is how the library is written. It inherited 4-bit color resolution from the Adafruit RGBMatrixPanel  library.