board707/DMD_STM32 #48: Color problem in panel increase

miralay001, 2023-09-13:
Hello, I use your library all the time.
There is no problem with green color when using a 2x2 P5 panel

I'm having trouble with panels turning green in 4x4 applications

There is no problem with other colors

my card is stmf4 black

I would be happy if you help me

https://github.com/board707/DMD_STM32/assets/108148338/490b296c-74e6-4d43-b003-a7fc4262795b

board707, 2023-09-14:
Hi
I need more detail about your setup.
What is the "P5 panel" you use? 
Could you please show the full code running in this video?

miralay001, 2023-09-14:
I'm sorry, this was a problem caused by my mistake, I stole your valuable time.

board707, 2023-09-14:
Glad to hear the problem has been resolved. Please fill free to contact us in case of any problems. 
You can star the project if you find it useful :)

closed 2023-09-14 by board707