board707/DMD_STM32 #54: Weird scan pattern

DoomHammer, 2023-10-24:
Hi @board707 !

Here's some more details about the cursed panel with weird pattern.

It's 2 mux pins, 2 scan lines, and what you see uses `#define DISPLAY_TYPE 2,32,16,2,1`

[image]

https://github.com/board707/DMD_STM32/assets/695373/eea6d7c9-18c0-40e3-afa8-76f75ee6c169

DoomHammer, 2023-10-24:
Do you want recording from there or just a plain "works/doesn't"?

board707, 2023-10-24:
If any of them will run more than 2 pixels horizontally in a serie...

DoomHammer, 2023-10-24:
With 1 mux line, the pixels get two levels of brightness (or something like that) after some time

## 2,32,16,2,2

https://github.com/board707/DMD_STM32/assets/695373/05dec579-4d87-429d-ae69-8b36a9d25f30

## 2,32,16,2,3

https://github.com/board707/DMD_STM32/assets/695373/44517e7e-11bd-49f6-9d69-de5eec0ba835

## 1,32,16,2,2 (this was mentioned twice by you, on purpose?)

https://github.com/board707/DMD_STM32/assets/695373/60e57f76-1d44-4014-9072-0a2a8eb73376

DoomHammer, 2023-10-24:
Again I'm using a PCB here, so no way to change connections. Will try when I set up another board.

board707, 2023-10-24:
Connecting B to the GND is not the main thing; when selecting one pin mux , the B pin should already have a low level.
The main thing is: do the pixels go to the end of the panel?

board707, 2023-12-14:
Hi, did you resolved the problem?

DoomHammer, 2023-12-14:
Not yet, didn't have time to give it my proper attention since the last time :(

board707, 2023-12-14:
Ok, 
If you will need my help, write.

Merry Christmas!

closed 2024-03-21 by board707