board707/DMD_STM32 #57: Add a new panel

VitaliyDKZ, 2023-11-13:
## Panel HRLOP10-4S-3535
#### Dimensions (in pixels) : 32x16
#### Scan factor: 1/4
#### Chips on rear side:  probably 74 series (only 74 visible)

## Panel photos
[image]
[image]

## Sample videos
### Pattern: 2,32,16,4,1
### Mux pins: 2
https://github.com/board707/DMD_STM32/assets/5511332/c93c5dcb-7f71-48c7-a837-f5b1743f5067

## Specifications
  - Library version: 1.01
  - Platform: STM32
  - MCU board: blackpil STM32F103C8T6
  - Connections: used pins
     
     DMD_PIN_A PB6
     DMD_PIN_B PB5

     DMD_PIN_nOE PB0
     DMD_PIN_SCLK PB7
     
     CLK:PA6   R1:PA0    G1:PA1    B1:PA2    R2:PA3    G2:PA4    B2:PA5

board707, 2023-11-13:
Hi
Please show video pixel test for pattern 2,32,16,4,2

VitaliyDKZ, 2023-11-13:
##2,32,16,4,2

https://github.com/board707/DMD_STM32/assets/5511332/962835e3-d99e-4359-97a2-7ef2bcb6ffaa

##2,32,16,4,3

https://github.com/board707/DMD_STM32/assets/5511332/c8e935fe-06e6-4c14-b51f-f74f2442a157

board707, 2023-11-13:
Thank you. i will try figure out this...

board707, 2023-11-14:
HI
Please add this code to your sketch just below `#include <DMD_RGB.h>` line:
```
#define RGB32x16_S4_VitalyDKZ    2,32,16,4,60

template<int COL_DEPTH>
class DMD_RGB<RGB32x16_S4_VitalyDKZ, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
	{
	public:
		DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
			byte panelsWide, byte panelsHigh, bool d_buf = false) :
			DMD_RGB_BASE2<COL_DEPTH>(2, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 4, 32, 16)
			{
			this->fast_Hbyte = false;
			this->use_shift = false;
			}
		// Fast text shift is disabled for complex patterns, so we don't need the method
		void disableFastTextShift(bool shift) override {}

	protected:
		uint16_t get_base_addr(int16_t& x, int16_t& y) override {
			this->transform_XY(x, y);
			uint8_t pol_y = y % this->pol_displ;
			x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
			uint16_t base_addr = (pol_y % this->nRows) * this->x_len + x * this->multiplex ;
                        if (pol_y < this->nRows) base_addr+= 1;
			return base_addr;
			}
	};
```

The same pattern must be used on DMD instance definition:
```
DMD_RGB <RGB32x16_S4_VitalyDKZ , COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER); 
```

Please let me know about results.

VitaliyDKZ, 2023-11-14:
Great everything works as it should!

https://github.com/board707/DMD_STM32/assets/5511332/5db638d1-4a24-46da-b8e4-86d8a15add16

https://github.com/board707/DMD_STM32/assets/5511332/1d70f856-2a91-4104-8f79-94737b394758

board707, 2023-11-14:
Thank you for your feedback
I will add the pattern to main repo in few days, before that moment you can use this code in your sketches

board707, 2023-11-21:
New pattern added in cac169c

closed 2023-11-21 by board707