board707/DMD_STM32 #60: Add a new panel (P10out-264T-2310-3A-20000)

VitaliyDKZ, 2023-12-13:
## Panel description 
#### Dimensions (in pixels) : 32x16
#### Scan factor: 1/2
#### Chips on rear side:  without marking (erased)
#### Model: P10-2S-3535-16x32-V2.3

## Panel photos
[image]
[image]

## Sample videos
#### Pattern: 2,32,16,4,1

https://github.com/board707/DMD_STM32/assets/5511332/431bd25e-1bea-429b-b0c4-e1509b66c49a

## Specifications

  - Library version: 1.01
  - Platform: STM32
  - MCU board: blackpil STM32F103C8T6
  - Connections: used pins
      DMD_PIN_A PB6
      DMD_PIN_B PB5
      DMD_PIN_nOE PB0
      DMD_PIN_SCLK PB7
      uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

board707, 2023-12-14:
Hi
Thank you for the info.

If the panel is labeled as 1/2 scan, your test pattern is incorrect, it is for 1/4 scan panel.
Please show the video for patterns:

2,32,16,2,1

1,32,16,2,1

Please see what kind of chips are used on the back of the panel. There must be at least three types of chips.

VitaliyDKZ, 2023-12-14:
Sorry I misspelled it I used 1,32,16,2,1 and wrote 1,32,16,4,1 (took the template from the last post) previous video is exactly 1,32,16,2,1.
Below I attached video for 2,32,16,2,1

https://github.com/board707/DMD_STM32/assets/5511332/017b349d-1b41-409e-8c88-01aa1f47a10f

board707, 2023-12-14:
Thank you, the last pattern is less compatible with the panel, so still using patterns, start from 1...

Could you please to record the video for 1,32,16,2,2 ?

VitaliyDKZ, 2023-12-14:
##1,32,16,2,2
https://github.com/board707/DMD_STM32/assets/5511332/61f352af-a7f9-48e9-aa12-41cc7d5f6b5a

VitaliyDKZ, 2023-12-14:
of course
## 1,32,16,2,2

https://github.com/board707/DMD_STM32/assets/5511332/56adbcbb-a8a4-494c-99d2-3af06b4844b1

board707, 2023-12-14:
Thank you? I will try to figure out your pattern in few days

board707, 2023-12-15:
HI
Please add this code to your sketch just below #include <DMD_RGB.h> line:
```
#define RGB32x16_S2_VitalyDKZ    1,32,16,2,61

template<int COL_DEPTH>
class DMD_RGB<RGB32x16_S2_VitalyDKZ , COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
	{
	public:
		DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
			byte panelsWide, byte panelsHigh, bool d_buf = false) :
			DMD_RGB_BASE2<COL_DEPTH>(1, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, 2, 32, 16)
			{
			this->fast_Hbyte = false;
			this->use_shift = false;
			}
		// Fast text shift is disabled for complex patterns, so we don't need the method
		void disableFastTextShift(bool shift) override {}

	protected:
		uint16_t get_base_addr(int16_t& x, int16_t& y) override {
			this->transform_XY(x, y);
			uint8_t pol_y = y % this->pol_displ;
			x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
			uint16_t base_addr = (pol_y % this->nRows) * this->x_len + x * this->multiplex ;
                        base_addr += this->multiplex - pol_y / this->nRows - 1;
			return base_addr;
			}
	};
```

VitaliyDKZ, 2023-12-16:
Well done! Its Works perfectly 
[image]

closed 2023-12-16 by board707

board707, 2023-12-20:
Pattern added in main library, version v1.1.1, 2f32d2a