board707/DMD_STM32 #64: PIO support?

xuniuer, 2023-12-20:
Hi there,

Could I use RP2004 framework based on PlatformIO?

board707, 2023-12-20:
Hi
The term PIO as applied to the rp2040 has a completely different meaning - this is the name given to the assembler machines in the processor core. My code uses these machines, in this sense the library is compatible with PIO :)
If you mean the PlatformIO programming environment, then I don't use it and therefore I have no experience embedding my code into it.
I think that it should not be more complicated than any other Arduino library, but I haven’t tried it myself.

xuniuer, 2023-12-20:
I mean the later～PlatformIO.

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/20/2023 15:34 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [board707/DMD_STM32] PIO support? (Issue #64) |

Hi
The term PIO as applied to the rp2040 has a completely different meaning - this is the name given to the assembler machines in the processor core. My code uses these machines, in this sense the library is compatible with PIO :)
If you mean the PlatformIO programming environment, then I don't use it and therefore I have no experience embedding my code into it.
I think that it should not be more complicated than any other Arduino library, but I haven’t tried it myself.

board707, 2023-12-20:
As far I know Platformio can use Arduino libraries, which means this library can be added to the Platformio project too.

xuniuer, 2023-12-20:
I will have a try.

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/20/2023 16:17 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [board707/DMD_STM32] PIO support? (Issue #64) |

As far I know Platformio can use Arduino libraries, which means this library can be added to the Platformio project too.

wsandor, 2023-12-20:
Hi,
I used the library whitin PlatformIO - not the latest version, but I think It should work also.

closed 2024-03-27 by board707