board707/DMD_STM32 #66: P4 display get vertical line issue

maxmurugan, 2023-12-28:
hi sir

RPI Pico controller some get vertical line issue continuously ,how to solve the issue

please check below code

#include "DMD_RGB.h"
//###########################################################################################
//#######################################################
#include "st_fonts/SystemFont5x7.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
byte second1, minute1, hour1, dayOfWeek1, dayOfMonth1, month1,year1;
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 2
#define ENABLE_DUAL_BUFFER true

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12

uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1

//DMD_RGB <RGB32x32plainS8, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB_SHIFTREG_ABC <RGB80x40plainS20, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font systemfont5x7(SystemFont5x7);

char*DaysOfWeek[] = {"DUMMY","SUNDAY", "MONDAY", "TUESDAY", "WEDNESDAY", "THURSDAY", "FRIDAY", "SATURDAY"};
char*DaysOfMonth[] = {"DUMMY","JANUARY", "FEBRUARY", "MARCH", "APRIL", "MAY", "JUNE", "JULY", " AUGUST", "SEPTEMBER", "OCTOBER", "NOVEMBER", "DECEMBER"};
int Total_week = 52;
int Char_Width_A = 6;//char width
int Char_Width_B = 8;//char width
int Module_Width = 80;
byte R1 = 1;
byte R2 = 2;
byte R3 = 3;
byte R4 = 4;

void setup()
{
  dmd.init();
  dmd.setBrightness(200);
  Serial.begin(9600);
  Serial.println("PICO P10 RTC");
  delay(100);
  hour1 =10;
  minute1 = 25;
  second1 = 30;
  dayOfWeek1 =1;
}
void loop()
{
      uint16_t col[] = {
      dmd.Color888(255,0, 0), // red
      dmd.Color888(0, 255, 0), // green
      dmd.Color888(0, 0, 255), // blue
      dmd.Color888(0, 255, 255), // blue
      dmd.Color888(255, 255, 0), // blue
      dmd.Color888(255, 0, 255), // blue
      dmd.Color888(255, 255, 255) // blue
      };
      uint16_t bg = 0;  // background - black
      dmd.selectFont(&systemfont5x7);
      char a[10];
      char b[10];
      char c[15];
      char d[15];
       while (1) 
       {

                    sprintf(a,"%02d:%02d",hour1,minute1);
                    sprintf(b,"%02d.%02d.20%02d",dayOfMonth1,month1,year1);
                    sprintf(c,"%s%02d","WEEK - ",Total_week);
                    sprintf(d,"%s",DaysOfWeek[dayOfWeek1]);
                    dmd.drawString(0,0,a,strlen(a),col[0]);
                    dmd.drawString(0,24,b,strlen(b),col[1]);
                    dmd.drawString(0,42,d,strlen(d),col[2]);
                    dmd.drawString(0,62,c,strlen(c),col[3]);
                    dmd.swapBuffers(true);
                  
       }
}
[image]

maxmurugan, 2023-12-28:
[image]

maxmurugan, 2023-12-28:
p4 - 80x40  two panel  please check

maxmurugan, 2023-12-28:
at this time serial data received continues  only display showing error

board707, 2023-12-28:
Hi
Please provide a more details of the problem.
What's the version of library do you using? (version number can be found in the library.properties file in the library folder).
Is the issue there is from the start of the project or it appeared recently after changing the panel, the board, library upgrade, or something else?

maxmurugan, 2023-12-28:
library - version [1.1.2] - 2023-12-22 

before i use p6 module no issue also try old version same issue will happen.
[image]
[image]

maxmurugan, 2023-12-28:
after few second going to vertical line

maxmurugan, 2023-12-28:
Arduino - 1.8.19
Pico -3.6.2
dmd library - 1.1.2

board707, 2023-12-28:
So it is a complete new panel that you didn't use before?
Please show a clear photo of the panel rear side and indicate partnumbers of any types of chips on the panel.

maxmurugan, 2023-12-28:
[image]

board707, 2023-12-28:
According the label the drivers are DP5125 and RUL5158, it should be supported.
As I see, your Pico pins are different that used in the library examples/ Did you used the same set with previous panel, that (as I understand) works fine?

maxmurugan, 2023-12-28:
Same pin I used last year to till now
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1

maxmurugan, 2023-12-28:
Same pin  configuration I used more then 5 projects. No issues 

I have change new pico and smps also issues will happen

board707, 2023-12-28:
To clarify the matter - is the same code, same library version and the pins works with previous panels and don't with new ones?

maxmurugan, 2023-12-28:
before i use version=1.0.6   

vertical line will create in this project

board707, 2023-12-28:
Did you try to use v1.0.6 with these (the new) panels?

( I apologize if you have already answered this question, but apparently I did not understand the answer )

maxmurugan, 2023-12-28:
Yes try to use v1.0.6 with these (the new) panels

board707, 2023-12-28:
Try to change the DMD class in the sketch from `DMD_RGB_SHIFTREG_ABC`to `DMD_RGB`

i.e :
```
DMD_RGB <RGB80x40plainS20, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

maxmurugan, 2023-12-28:
is not clear message

maxmurugan, 2023-12-28:
DMD_RGB_SHIFTREG_ABC  class only working in this panel

board707, 2023-12-28:
> DMD_RGB_SHIFTREG_ABC class only working in this panel

Is it means that you see the text on the matrix or the "working" means you see the vertical lines only as on photo from the beginning of the topic?

maxmurugan, 2023-12-28:
text on the matrix  after few second  verticals line showing using - DMD_RGB_SHIFTREG_ABC class

board707, 2023-12-28:
I have no ideas, sorry.
Test all your connections, try another Raspberry Pico and another panel... 
Try another time a new panel with previous library version.
Please let me know if the things will change.

maxmurugan, 2023-12-28:
ok i will try to change new module and library
thanks

maxmurugan, 2023-12-30:
hi sir
Replace new type of module using same code test more then 12hours NO issue.
working fine

DMD_RGB_SHIFTREG_ABC <RGB80x40_S10_LNikon, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

[image]
[image]

board707, 2023-12-30:
Thank you for your feedback

closed 2024-01-02 by maxmurugan