board707/DMD_STM32 #68: RGB 32x16 led matrix using stm32f103 controller matrix have A, B, C others pins like D AND E is ground.

nawab-sw, 2024-01-16:
I am working on RGB 32x16 led matrix using stm32f103 controller all the connection same as in liabrary given but still not get the result, matrix is not faulty i have checked it with esp8266 working but don't know what is issues with stm32f103, please help me like wring connection pins between controller and 32x16 matrix, and i want to inform you matric have A,B, C others pins like D AND E is ground

board707, 2024-01-16:
Hi
Please show a clear photo of the panels's rear side. What are the markings on the chips on the back of the matrix? There should be at least three different types of the chips.

nawab-sw, 2024-01-16:
[image]
[image]
[image]
[image]
[image]

board707, 2024-01-16:
I don't see a chip's ID on your photos. 

However, it appears to be a standard panel. Try changing the A B C pins and start again
```
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB8
```
You don't need D and E

closed 2024-01-17 by board707

nawab-sw, 2024-01-17:
KINDLY SEE THE WITH MONOCHROME LED MATRIX   Monochrome P10 LED Matrix with
64x16 (32x16 + 32x16 Two panel in series) using stm32f103. · Issue #69 ·
board707/DMD_STM32 (github.com)
<https://github.com/board707/DMD_STM32/issues/69>