board707/DMD_STM32 #72: Add multiple HUB75 PORT .

ravigyanvihar, 2024-01-29:
Sir ,
I want to make a PCB where I can add multiple hub75 port. 
Is it possible to make multiple the OE signal so at one time I can run one row of panel and so on...

board707, 2024-01-29:
Hi
Thanks for your interest.
I doubt that STM32 boards will support more than one hub75 interface. I tested two hub75 channels on Raspberry Pico RP2040, but on stm32, I think, there are not enough resources.

DoomHammer, 2024-01-31:
Is there an example of how to use 2 channels?

board707, 2024-01-31:
Do you mean a two channels on  Raspberry Pico RP2040 ?
I haven't published these tests. I'll try to find my code and add it to the examples, but it will take some time.

ravigyanvihar, 2024-01-31:
Ok sir . I am waiting

DoomHammer, 2024-01-31:
Yes, 2 channels on Pico.

Also: how is that supposed to work. Do we get two different framebuffers or what?

board707, 2024-01-31:
> how is that supposed to work

Sorry, I did this almost 2 years ago, in the summer of 2022, I don't remember the details. I need to find that code.

board707, 2024-01-31:
Hi, friends !
I found my old code, but this is only a test, it is far from real use.
The general principle of operation is simple. If you look at the current version of the library, working with the hub75 channel for the RP2040 processor consists of three parts. The first part is a PIO machine that controls the multiplexer channels A B C D E. The second PIO machine outputs main RGB color channel data with CLK and LAT signals. The data for RGB pins delivered via DMA. And the third part is brightness control via PWM on the OE pin, controlled by an independent timer.

In the case of using two hub75 channels, control of the ABCDE and OE pins remains common to the two HUB channels, and two PIO machines are configured to output the color data of the first and second hub75 channels.
Thus, for the operation of two parallel hub75 channels, 22 pins are required - 6 common (ABCDE + OE) and 8 for each channel - 6x RGB + LAT + CLK . Tests have shown that the scheme is quite workable, although it requires the use of the same type of panel on two channels.
As you can see, in this mode, almost all pins of the rp2040 are occupied, which limits the use of this circuit. Therefore, I did not add this mode into the library.

board707, 2024-01-31:
I think it will be better to convert this topic to the discussion, do you agree?

ravigyanvihar, 2024-02-01:
Yes absolutely