board707/DMD_STM32 #74: P10 red 63 panel using 112 pixel height  clock issue

maxmurugan, 2024-02-03:
hi sir
I am using  63 panel p10 module used

#define DISPLAYS_ACROSS 9
#define DISPLAYS_DOWN 7

maximum support DISPLAYS_DOWN 3 working 
more then DISPLAYS_DOWN 3 not working

please check my code below sir,
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_Monochrome_Parallel.h"

#include "st_fonts/Arial_79.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 9
#define DISPLAYS_DOWN 7

#define ENABLE_DUAL_BUFFER false

#define DMD_PIN_A 10
#define DMD_PIN_B 11
#define DMD_PIN_nOE 14
#define DMD_PIN_SCLK 15

uint8_t pins[] = { 6, 7, 8, 9,16,17,18,19};  // CLK , row1, row 2,, row 3
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
DMD_Standard_Font arial_9(Arial_79);

int Hour =06;
int minute1 =34;
char a[10];

void setup(void)
{
  dmd.init();
  Serial.begin(9600);
  delay(100); 
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{
       
    dmd.selectFont(&arial_9);
    dmd.setBrightness(100);
    Serial.print("WELCOME PICO P10 RED");
    while (1) 
    {
                      sprintf(a,"%2d:%02d",Hour,minute1);
                      dmd.drawString(0,0,a,strlen(a),1);
                      dmd.swapBuffers(true);

     }

}

board707, 2024-02-03:
Hi
You have a problem with your row pins:
```
uint8_t pins[] = { 6, 7, 8, 9,16,17,18,19}; // CLK , row1, row 2,, row 3...
```
On Raspberry Pico all row pins must be in ascending order, without gaps. So if you started from "6,7,8,9", the next pins must be 10,11, 12... etc

maxmurugan, 2024-02-03:
ok sir i will try

maxmurugan, 2024-02-06:
Tested  for width - 9 module and height - 7 module as per your pin detail working perfectly..
thangs for support.

https://github.com/board707/DMD_STM32/assets/4962604/07d75489-b2cf-475d-a8a1-077812ff4b27

board707, 2024-02-06:
Great!

closed 2024-02-07 by board707