board707/DMD_STM32 #77: pin connections

tahafathi2013, 2024-02-21:
what is pin connections?

board707, 2024-02-21:
Sorry, I dont understand what you asked.

tahafathi2013, 2024-02-21:
i have bluepill 
i use connections on the code but it not working on old version and new version.
what is wire connections from p10 matrix display to bluepill pins.

p10 pins         bluepill pins  on SPI1 AND SPI2
OE                       ?
A                         ?
B                         ?
-                          ?
CLK                     ?
SCLK                  ?
DATA                 ?

board707, 2024-02-21:
Please see pins in the examples. In dmd_momochrome.ino:
```
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PA15

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15

```

> C ____?

Monochrome panels doesn't have a pin C.

closed 2024-02-22 by board707