board707/DMD_STM32 #78: error in compile whit arduino ide

tahafathi2013, 2024-02-21:
when compile project on stm32 f103 i have a error

In file included from c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_Monochrome_Parallel.h:2,
                 from C:\Users\DELL\Desktop\DMD_STM32-dev-V2_2\DMD_STM32-dev-V2\examples\STM32F1\dmd_monochrome\dmd_monochrome.ino:13:
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:334:40: error: 'voidFuncPtr' has not been declared
  334 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ^~~~~~~~~~~
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:368:9: error: 'PortType' does not name a type
  368 |         PortType clk_clrmask = 0;
      |         ^~~~~~~~
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:369:9: error: 'PortType' does not name a type
  369 |         PortType clkmask = 0;
      |         ^~~~~~~~
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:379:9: error: 'OE_PWM_Polarity' does not name a type
  379 |         OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
      |         ^~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

how fix it????

board707, 2024-02-21:
please show a compile error message in full

tahafathi2013, 2024-02-21:
WARNING: library DMD_STM32 claims to run on STM32F1, STM32F4, rp2040 architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
In file included from c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_Monochrome_Parallel.h:2,
                 from C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved2024122-20196-smqrxt.x2qvr\dmd_monochrome\dmd_monochrome.ino:13:
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:334:40: error: 'voidFuncPtr' has not been declared
  334 |         virtual void initialize_timers(voidFuncPtr handler);
      |                                        ^~~~~~~~~~~
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:368:9: error: 'PortType' does not name a type
  368 |         PortType clk_clrmask = 0;
      |         ^~~~~~~~
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:369:9: error: 'PortType' does not name a type
  369 |         PortType clkmask = 0;
      |         ^~~~~~~~
c:\Users\DELL\Documents\Arduino\libraries\DMD_STM32/DMD_STM32a.h:379:9: error: 'OE_PWM_Polarity' does not name a type
  379 |         OE_PWM_Polarity OE_polarity = OE_PWM_POSITIVE;
      |         ^~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

this is eror message full.
[image]

board707, 2024-02-21:
> WARNING: library DMD_STM32 claims to run on STM32F1, STM32F4, rp2040 architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).

According to the error message, you use an incompatible STM32 arduino package. The library can be used with Roger Clark's STM32duino package only. Please read [Compatible IDE and libraries](https://github.com/board707/DMD_STM32/tree/dev-V2?tab=readme-ov-file#compatible-ide-and-libraries) section in the Readme

tahafathi2013, 2024-02-22:
im install and use roger clarks stm32duino packase and after compile i have error

[image]

how fix it?

board707, 2024-02-22:
Install Adafruit_GFX library version 1.7, the link in the Readme

board707, 2024-02-22:
Try this
```
drawString(int bX, int bY, const char* bChars, int length,
		T color, byte orientation = 0);
```
example:
```
uint16_t fg = dmd.Color888(0, 0, 255);  // blue
char s[] = "Hello";
drawString(0,0, s, strlen(s), fg);  // print "Hello" with current font at x, y = 0,0
```

tahafathi2013, 2024-02-22:
please full example for red p10 matrix display for show static text by english fonts.

board707, 2024-02-22:
Sorry,  I'll write to you in the evening, after 10pm (GMT+3)

tahafathi2013, 2024-02-22:
ok tank you.
but im use your example codes : 

uint16_t fg = dmd.Color888(0, 0, 255);  // blue
char s[] = "Hello";
drawString(0,0, s, strlen(s), fg);  // print "Hello" with current font at x, y = 0,0

but i have error : 

[image]

tahafathi2013, 2024-02-22:
error: 'class DMD_Monochrome_Parallel' has no member named 'Color888'

tahafathi2013, 2024-02-22:
ok thank you.
but i have error fo try your code : 

uint16_t fg = dmd.Color888(0, 0, 255);  // blue
char s[] = "Hello";
drawString(0,0, s, strlen(s), fg);  // print "Hello" with current font at x, y = 0,0

ERROR : 
[image]

board707, 2024-02-22:
oops, Color888 is a function for RGB panels :)

It should be correct for P10:
```
uint16_t fg = 1;
char s[] = "Hello";
drawString(0,0, s, strlen(s), fg); 
```

tahafathi2013, 2024-02-22:
sorry not working . . .

tahafathi2013, 2024-02-22:
sorry this conde not show HELLO words 
but i dont have Error.
can you another example code for show text to work it?

board707, 2024-02-22:
```
#include "DMD_RGB.h"

// Fonts includes
#include "st_fonts/Arial14.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0, PA1, PA2, PA3, PA4, PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>(STM32 only), <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_F(Arial_14);

void setup(void)
{
  // initialize DMD objects
  dmd.init();
  dmd.setBrightness(200);
  
}

void loop(void)
{
  
  // select standard font
  dmd.selectFont(&Arial_F);

  // text
  char text[] = "DMD_STM32";
 
  // draw fixed text
  uint8_t text_start = (dmd.width() - dmd.stringWidth(text)) /2;  // center aligned text
  dmd.drawStringX(text_start, 15, text, 1);

  // output mem buffer to matrix
  dmd.swapBuffers(true);
 
}
```

board707, 2024-02-22:
```
// choose between Parallel ans SPI wiring
// comment line below for SPI connection
#define DMD_PARA
/*--------------------------------------------------------------------------------------
  Includes
--------------------------------------------------------------------------------------*/
#if defined(DMD_PARA)
#include "DMD_Monochrome_Parallel.h"
#else
#include "DMD_MonoChrome_SPI.h"
#endif

// Fonts includes
#include "st_fonts/Arial14.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 2

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be Timer3 PWM pin as PB0,PB1
// if connect as SPI, do not use corresponding MiSO pin - PA6 for SPI1 and PB14 for SPI2
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
//#define DMD_PIN_nOE PA6
#define DMD_PIN_SCLK PA15
//#define DMD_PIN_SCLK PB4

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

//=== Config for Parallel connect ====
#if defined(DMD_PARA)
//pins for rows at x axis
// example for two rows
// all those pins must be selected from same port!
uint8_t pins[] = { PA5, PA7, PA6 };  // CLK , row1, row 2

//Fire up the DMD object as dmd
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
#else

//=== Config for SPI connect ====
SPIClass dmd_spi(1);
DMD_MonoChrome_SPI dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, DISPLAYS_ACROSS, DISPLAYS_DOWN, dmd_spi, ENABLE_DUAL_BUFFER);
#endif

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_F(Arial_14);

void setup(void)
{
  // initialize DMD objects
  dmd.init();
  dmd.setBrightness(200);
  
}

void loop(void)
{
  
  // select standard font
  dmd.selectFont(&Arial_F);

  // text
  char text[] = "DMD_STM32";
 
  // draw fixed text
  uint8_t text_start = (dmd.width() - dmd.stringWidth(text)) /2;  // center aligned text
  dmd.drawStringX(text_start, 15, text, 1);

  // output mem buffer to matrix
  dmd.swapBuffers(true);
 
}
```

tahafathi2013, 2024-02-23:
I try this example but not working
I use one p10 panel and spi 2

[image]

board707, 2024-02-23:
Did you try other library examples for monochrome panels?

What is your panel? Could you please show a photo of the panel rear side?

tahafathi2013, 2024-02-23:
[image]

board707, 2024-02-23:
Sorry, your panel looks like a very old style panel with DIP leds. Probably, the interface of your panel is not HUB12, but HUB08. The library has never been tested on HUB08 because I don't have similar panels.

tahafathi2013, 2024-02-23:
But it work on my panel by default your example for monochrome.
Just not work static text to show by any fonts.

https://github.com/board707/DMD_STM32/assets/19773695/afc56767-e2fd-4106-abdd-e256293b18f7

board707, 2024-02-23:
As far as I can see, only half of the LEDs are lit. It can hardly be said that the panel is working.
However, static text should also be shown in this mode. Please indicate which example works for you in this video. Did you added at least a minor changes in the example code?

tahafathi2013, 2024-02-24:
I just changed the SPI mode 1 to 2 and programmed it on the stm32 bluephiil board, but it didn't display anything correctly. Do I need to make any special changes?

board707, 2024-02-24:
Either way, your panels are unlikely to work correctly. In the video, the text is shown every other line and it hardly makes sense to set up some custom fonts when the usual example does not work.
I have no immediate plans to adapt the library for such panels, since they are already very outdated.
If it worked for you with an Arduino Mega board, probably the best solution would be to use Arduino instead of STM32.

tahafathi2013, 2024-02-24:
من با برد اردوینو مگا و نانو امتحان کردم ولی سرعت رفرش بسیار پایین است و به صورت چشمک زن قابل مشاهده هست 
نمیدونم سرعت کلاک آتمگا ۵۶۰  پایین هست یا داخل کد کتابخانه دی ام دی ۲  تنظیمات خاصی باید انجام داد.
من از برنامه پیشفرض خود کتابخانه استفاده کردم ولی گاهی در حین نمایش متن ال ای دی ها فریز میشود و به صورت نامنظم روشن میماند
ایا شما با اردوینو مگا و یا نانو امتحان کردین؟ یا مثال بی نقصی سراغ دارید که من تست کنم؟

board707, 2024-02-24:
> I tried with Arduino Mega and Nano board, but the refresh rate is very low

Please show your code for this panel using Mega or Nano.

tahafathi2013, 2024-02-24:
`#include <SPI.h>
#include <DMD.h>
#include <TimerOne.h>
#include "SystemFont5x7.h"
#include "Arial_black_16.h"

#define ROW 1
#define COLUMN 1
#define FONT Arial_Black_16

DMD led_module(ROW, COLUMN);

void scan_module()
{
  led_module.scanDisplayBySPI();
}

void setup()
{
  Timer1.initialize(2000);
  Timer1.attachInterrupt(scan_module);
  led_module.clearScreen( true );
}
void loop()
{

   led_module.clearScreen(true);

  led_module.selectFont(FONT);
  uint16_t fg = 1;
  char s[] = "22.5";
  led_module.drawString(0, 0, s, strlen(s), 0);

    
  }

whit arduino nano`

tahafathi2013, 2024-02-24:
`// choose between Parallel ans SPI wiring
// comment line below for SPI connection
#define DMD_PARA
/*--------------------------------------------------------------------------------------
  Includes
--------------------------------------------------------------------------------------*/
#if defined(DMD_PARA)
#include "DMD_Monochrome_Parallel.h"
#else
#include "DMD_MonoChrome_SPI.h"
#endif

// Fonts includes  Arial_black_16.h
#include "st_fonts/Arial14.h"
//#include "st_fonts/Arial14.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be Timer3 PWM pin as PB0,PB1
// if connect as SPI, do not use corresponding MiSO pin - PA6 for SPI1 and PB14 for SPI2
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
//#define DMD_PIN_nOE PA6
#define DMD_PIN_SCLK PA15
//#define DMD_PIN_SCLK PB4

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

//=== Config for Parallel connect ====
#if defined(DMD_PARA)
//pins for rows at x axis
// example for two rows
// all those pins must be selected from same port!
uint8_t pins[] = { PA5, PA7, PA6 };  // CLK , row1, row 2

//Fire up the DMD object as dmd
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
#else

//=== Config for SPI connect ====
SPIClass dmd_spi(2);
DMD_MonoChrome_SPI dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, DISPLAYS_ACROSS, DISPLAYS_DOWN, dmd_spi, ENABLE_DUAL_BUFFER);
#endif

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_F(Arial_14);

void setup(void)
{
  // initialize DMD objects
  dmd.init();
  dmd.setBrightness(200);
  
}

void loop(void)
{
    dmd.clearScreen(true);

  // select standard font
  dmd.selectFont(&Arial_F);

  // text
  char text[] = "DMD_STM32";
 
  // draw fixed text
  uint8_t text_start = (dmd.width() - dmd.stringWidth(text)) /2;  // center aligned text
  dmd.drawStringX(text_start, 15, text, 1);

  // output mem buffer to matrix
  dmd.swapBuffers(true);
 
}`
i upload code via upload method SERIAL

board707, 2024-02-24:
If you use a SPI, you should comment the line 
```
#define DMD_PARA
```
at the beginning of the code

tahafathi2013, 2024-02-24:
`
// choose between Parallel ans SPI wiring
// comment line below for SPI connection
//#define DMD_PARA
/*--------------------------------------------------------------------------------------
  Includes
--------------------------------------------------------------------------------------*/
#if defined(DMD_PARA)
#include "DMD_Monochrome_Parallel.h"
#else
#include "DMD_MonoChrome_SPI.h"
#endif

// Fonts includes  Arial_black_16.h
#include "st_fonts/Arial14.h"
//#include "st_fonts/Arial14.h"
// Fonts includes
//#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ----- Select pins for P10 matrix connection ------------
// pins A, B, SCLK may be any digital I/O, pin nOE should be Timer3 PWM pin as PB0,PB1
// if connect as SPI, do not use corresponding MiSO pin - PA6 for SPI1 and PB14 for SPI2
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
//#define DMD_PIN_nOE PA6
#define DMD_PIN_SCLK PA15
//#define DMD_PIN_SCLK PB4

// pins for SPI connect
// SPI specific pins as CLK and R_DATA has predefined values:
//   for SPI(1) CLK = PA5  R_DATA = PA7
//   for SPI(2) CLK = PB13  R_DATA = PB15
// --------------------------------------------------------

//=== Config for Parallel connect ====
#if defined(DMD_PARA)
//pins for rows at x axis
// example for two rows
// all those pins must be selected from same port!
uint8_t pins[] = { PA5, PA7, PA6 };  // CLK , row1, row 2

//Fire up the DMD object as dmd
DMD_Monochrome_Parallel dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, pins, DISPLAYS_ACROSS, DISPLAYS_DOWN,ENABLE_DUAL_BUFFER);
#else

//=== Config for SPI connect ====
SPIClass dmd_spi(2);
DMD_MonoChrome_SPI dmd(DMD_PIN_A, DMD_PIN_B, DMD_PIN_nOE, DMD_PIN_SCLK, DISPLAYS_ACROSS, DISPLAYS_DOWN, dmd_spi, ENABLE_DUAL_BUFFER);
#endif

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_F(Arial_14);

void setup(void)
{
  // initialize DMD objects
  dmd.init();
  dmd.setBrightness(200);
  
}

void loop(void)
{
    dmd.clearScreen(true);

  // select standard font
  dmd.selectFont(&Arial_F);

  // text
  char text[] = "DMD_STM32";
 
  // draw fixed text
  uint8_t text_start = (dmd.width() - dmd.stringWidth(text)) /2;  // center aligned text
 // dmd.drawStringX(text_start, 15, text, 1);
dmd.drawString(0, 0, text, 9, 1);
  // output mem buffer to matrix
  dmd.swapBuffers(true);
 
}
`
im use this code but flashing on the end of panels

board707, 2024-02-24:
Try to move all loop() contents to the setup()

tahafathi2013, 2024-03-27:
hi 
i change setup for arduino
install arduino 1.8 from this link  https://www.arduino.cc/en/software/OldSoftwareReleases
and install your librarry but i have error 

[image]

how fix it?

board707, 2024-03-27:
`gfxfont.h` - this file is a part of AdafruitGFx library. It required for DMD_STM32 installation.
Did you install it as stated in Readme?

tahafathi2013, 2024-03-27:
yes im install AdafruitGFx  v1

[image]

board707, 2024-03-27:
Please install AdafruitGfx from this link
https://github.com/adafruit/Adafruit-GFX-Library/releases/tag/1.7.0

closed 2024-04-10 by board707