board707/DMD_STM32 #79: issues with P8 32*16 1/4 scan panel

saka2080, 2024-02-22:
I have been  trying to display a dot and a char on  this P8 32x16 1/4 scan rgb matrix panel,  but not succeeded.  I am using blue pill and arduino IDE and STLink to burn the program. Sometimes, the matrix shows garbage pixels lit and sometimes it is  totally blank. I have connected pins as:
R0- PA0      G0-PA1
B0- PA2      GND-GND
R1- PA3      G1-PA4
B1- PA5      GND-GND
A- PB6        B-PB5
C- PB4         GND-GND
CLK- PA15  LAT-PB7
OE- PB0     GND-GND

I used the following code:

#include "DMD_RGB.h"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER false
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B,DMD_PIN_C};

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

DMD_RGB <RGB32x16plainS4, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

DMD_Standard_Font Arial_Black_F(Arial_Black_16);

void setup(void)
{
//enableDebugPorts();
    // initialize DMD objects
	dmd.init(); 

dmd.drawChar(0, 0, 'A', 3, 0) ;
dmd.drawPixel(10, 10, 5);
dmd.swapBuffers(true);
//
    delay(5000);
}
`
[image]
[image]

board707, 2024-02-22:
>  P8 32x16 1/4 scan rgb matrix panel

Hi!
First of all - are you sure that it is a 1/4 scan? The labels on the panel is not clear about this, and panel has A, B and C pins - it is typically for 1/8 scan and not for 1/4.

I can't read chip labels on your photo - do you know the driver and multiplexor type of the panel?

saka2080, 2024-02-22:
One IC is readable ,it is TC4538. I guess it is 1/4 scan, because it shows  P8_1_4_3535_PR4538_HSD_32x16_NRES in the label.

saka2080, 2024-02-22:
Thank you for your prompt reply. Even I used 
`DMD_RGB <RGB32x16plainS8, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);`
but display did not show anything.

board707, 2024-02-22:
How do you upload the code to your STM board? Via St-Link or via USB bootloader?

saka2080, 2024-02-22:
Via St-link

board707, 2024-02-22:
When you using a St-link, pins PB3, PB4 and PA15 are blocked.
To release these pins add a line 
```
enableDebugPorts();
```
just at the start of `setup()` function and test again.

saka2080, 2024-02-22:
No luck.   I used enableDebugPorts(); but display did not show anything.

But when I do using RGBmatrixPanel using mega2560 using the code below the display works with 2 panel. But I don't know why it is not working with bluepill

[image]
[image]
```
#include "gamma.h"
#include <Adafruit_GFX.h>    // Core graphics library
#include "digital_7_mono16pt7b.h"
#include "RGBmatrixPanel.h" // Hardware-specific library (Ruud’s version)
#define clear() fillScreen(0)

#define CLK 11 // MUST be on PORTB! (Use pin 11 on Mega)
#define LAT 10
#define OE 9
#define A A0
#define B A1
#define C A2

RGBmatrixPanel matrix(A, B, C, CLK, LAT, OE, false,2); // 1 = amount of panels

void setup() {
matrix.begin();
  matrix.setTextSize(1);
  matrix.setCursor(2,25);
  matrix.setTextColor(matrix.Color333(7,0,0));
  matrix.print("91");
  matrix.setCursor(34,9);
  matrix.print("91");
}
```

board707, 2024-02-22:
Could you replace the setup() and loop() functions to this code:
```
void setup(void)
{
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
```

saka2080, 2024-02-22:
I used the following code with the following setup.
[image]
[image]

But the display did not show a single pixel. I verified the pin to pin connections too.

```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
//#define DMD_PIN_D PB3
//#define DMD_PIN_E PB8
// put all mux pins at list
//uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2         - 32x16 1/2 Binary mux

DMD_RGB <RGB32x16plainS8, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
```

board707, 2024-02-22:
Sorry, I forgot to add the line `enableDebugPorts();` to the `setup()`
```
void setup(void)
{
   enableDebugPorts();
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}
```
without this line any tests is a waste of the time.

saka2080, 2024-02-22:
I have been trying but no result. But I found that if  there is some garbage data (by turing ON/OFF power supply of panel), the dmd.setBrightness() is working.

board707, 2024-02-22:
We don't know what driver your board has, it may not be compatible with the library. But, since the panels work with the RGBmatrixPanel  library, they should also work with this code.
Unfortunately, I have no more ideas for now...

saka2080, 2024-02-22:
it is TC4538. the label is P8_1_4_3535_PR4538_HSD_32x16_NRES.

board707, 2024-02-22:
No, the 4538 chip is not a driver, it is multiplexor. Every panel contains at least two types a chips - a driver and a multiplexor. 
We do not know the driver, but it seems to be compatible because the panel works with RGBMatrixPanel library.

board707, 2024-03-20:
Hi
As I see in the pictures, your panel seems to be 1/8 scan.
Could you try the code from [comment](https://github.com/board707/DMD_STM32/issues/79#issuecomment-1959339712) with color mode changed to COLOR_4BITS_Packed:
```
DMD_RGB <RGB32x16plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

saka2080, 2024-03-20:
Hi,
I tried as you said, i got it as below in the video.

https://github.com/board707/DMD_STM32/assets/160825352/ee5052f7-e01e-48ef-8b0c-10773d658c36

for  the code
`void setup(void)
{
  enableDebugPorts();
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}`
It seems something is working.

board707, 2024-03-20:
The video is strange...
Please show the full code.

Could you try the same code with RGB32x16plainS4 pattern?

saka2080, 2024-03-20:
```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER false
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4

uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
DMD_RGB <RGB32x16plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
  enableDebugPorts();
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);

    dmd.drawPixel(0,0, dmd.Color888(255, 0, 255));
    delay(1000);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
```
OK. I ll try same code and post here.

board707, 2024-03-20:
```
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
```
These lines should be 1 in both cases.

saka2080, 2024-03-20:
Using the code below,  it  produced following output.

```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER false
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4

uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
DMD_RGB <RGB32x16plainS4, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
  enableDebugPorts();
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);

    dmd.drawPixel(0,0, dmd.Color888(255, 0, 255));
    delay(2000);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
```

https://github.com/board707/DMD_STM32/assets/160825352/0180eaf4-6fee-453a-bef5-333fc7fcfe38

board707, 2024-03-20:
Sorry, your video is too short.
Could you provide a view at least until a lower part of panel starts filling?

saka2080, 2024-03-20:
https://github.com/board707/DMD_STM32/assets/160825352/820472ad-bcf0-4be0-afb1-d14efb1236d3

saka2080, 2024-03-20:
https://github.com/board707/DMD_STM32/assets/160825352/13d871c4-7f44-4c29-a8cc-149d21cb587e

saka2080, 2024-03-20:
another video

https://github.com/board707/DMD_STM32/assets/160825352/adee2197-6153-4439-be08-77a0980363fa

Thanks..

board707, 2024-03-20:
I only see the message: "The file is corrupt", sorry

saka2080, 2024-03-20:
https://github.com/board707/DMD_STM32/assets/160825352/0fc61999-8877-4966-82de-ae321e3c7a98

board707, 2024-03-20:
Thank you.
So your panel is 1/4 scan, I will build the coordinate pattern tomorrow.

board707, 2024-03-21:
Please add this code snippet to your sketch just after `#include "DMD_RGB.h"` line:
Use template `RGB32x16_S4_saka2080` for init of the DMD class instance.

```
template<int COL_DEPTH>
#define RGB32x16_S4_saka2080 2,32,16,4,64
class DMD_RGB<RGB32x16_S4_saka2080, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
	{
	public:
		DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
			byte panelsWide, byte panelsHigh, bool d_buf = false) :
			DMD_RGB_BASE2<COL_DEPTH>(2, mux_list, _pin_nOE, _pin_SCLK, pinlist,
				panelsWide, panelsHigh, d_buf, COL_DEPTH, 4, 32, 16)
			{
			this->fast_Hbyte = false;
			this->use_shift = false;
			}
		// Fast text shift is disabled for complex patterns, so we don't need the method
		void disableFastTextShift(bool shift) override {}

	protected:
		uint16_t get_base_addr(int16_t& x, int16_t& y) override {
                        static const uint8_t  pix_base =8;
			this->transform_XY(x, y);
			uint8_t pol_y = y % this->pol_displ;
			x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
			 uint16_t base_addr = (pol_y % this->nRows) * this->x_len +
                           (x / pix_base) * this->multiplex * pix_base +
                           x % pix_base;
      if (pol_y < this->nRows)  base_addr += pix_base;
			return base_addr;
			}
	};
```

saka2080, 2024-03-21:
Hi.
Its showing error as:
exit status 1
wrong number of template arguments (6, should be 2)

and the code is:

```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels
 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

template<int COL_DEPTH>
#define RGB32x16_S4_saka2080 2,32,16,4,64
class DMD_RGB<RGB32x16_S4_saka2080, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
  {
  public:
    DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
      byte panelsWide, byte panelsHigh, bool d_buf = false) :
      DMD_RGB_BASE2<COL_DEPTH>(2, mux_list, _pin_nOE, _pin_SCLK, pinlist,
        panelsWide, panelsHigh, d_buf, COL_DEPTH, 4, 32, 16)
      {
      this->fast_Hbyte = false;
      this->use_shift = false;
      }
    // Fast text shift is disabled for complex patterns, so we don't need the method
    void disableFastTextShift(bool shift) override {}

  protected:
    uint16_t get_base_addr(int16_t& x, int16_t& y) override {
                        static const uint8_t  pix_base =8;
      this->transform_XY(x, y);
      uint8_t pol_y = y % this->pol_displ;
      x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
       uint16_t base_addr = (pol_y % this->nRows) * this->x_len +(x / pix_base) * this->multiplex * pix_base +  x % pix_base;
      if (pol_y < this->nRows)  base_addr += pix_base;
      return base_addr;
      }
  };
  
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER false
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4

uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
DMD_RGB <RGB32x16plainS4, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
  enableDebugPorts();
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);

    dmd.drawPixel(0,0, dmd.Color888(255, 0, 255));
    delay(2000);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
```

board707, 2024-03-21:
What the library version you using?
The templates with two parameters stops using in 0.9.x versions, but actual branch is v1.1.x

saka2080, 2024-03-21:
Thank you very much. it was older version. Using with new version worked properly.
Can  i connect two  or four panels together?  I tried
 #define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
but that did not work. should i change other parameters too?

board707, 2024-03-21:
This pattern should works with any number of panels. 
Please be specific, whats mean "did not work" in your case?
Could you show a illustrate video?

saka2080, 2024-03-21:
Sorry. You are  right. Actually the panel was faulty. It worked with another panel.
Thank you so much.

board707, 2024-03-21:
Great.

By the way, I saw your question about P8 20*40 1/5 panel support in the project for ESP32 boards. If you want we can run this panel on my library.

saka2080, 2024-03-21:
Sure. 
Thanks.

closed 2024-03-22 by board707

board707, 2024-04-11:
Pattern `RGB32x16_S4_saka2080` added to the library v1.1.3. Now you can use it directly in the DMD class init:
```
DMD_RGB <RGB32x16_S4_saka2080, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

saka2080, 2024-05-18:
Thanks a lot. Nice to know after returning from my off times.