board707/DMD_STM32 #80: Add a new panel Please

digi55, 2024-02-22:
## Panel description 
#### Dimensions (in pixels) : 80x40    P4 Panel
#### Scan factor: 
[image]
[image]
[image]
[image]

#### Chips on rear side:  TC7262D, FM6124D, HCT245

## Panel photos

## Sample videos

## Specifications

  - Library version: 1.1.2
  - Platform: STM32
  - MCU board: bluepil STM32F103C8T6
  - Connections:  A=PB6, B=PB5, C=PB4, D= PB3, D=PB8     nOE=PB0,   SCLK=PB7   CLK=PA6,  R0=PA0,  G0=PA1, B0=PA2, 
 R1=PA3,  G1=PA4,  B1=PA5

digi55, 2024-02-22:
Hello
sorry I don't know the scan factor of this panel
yes i tested (RGB80x40_S10_LNikon)
I am sending a video
I used , dmd rgb turk in example

https://github.com/board707/DMD_STM32/assets/160831820/336e327b-2576-4630-b037-4115f808cc64

board707, 2024-02-22:
Sorry, but your video is empty.
Can you upload it on Youtube and send the link?

digi55, 2024-02-22:
https://github.com/board707/DMD_STM32/assets/160831820/096c6da2-2942-464b-8e6d-17bfa1dabc0f

dmd_rgb_test_pattern with RGB80x40_S10_LNikon
It constantly writes on these 4 lines.

board707, 2024-02-22:
How do you upload the code to your STM board? Via St-Link or via USB bootloader?

Also, in your connections:
>  A=PB6, B=PB5, C=PB4, D= PB3, D=PB8 
Is the second "D" pin just a typo and should be read as "E=PB8" ?

board707, 2024-02-22:
Please add the line `enableDebugPorts();` at the first line of `setup()` function

digi55, 2024-02-22:
enableDebugPorts(); After this command

Everything is OK.
Thank you so much, For your helps.

board707, 2024-02-22:
Thank you for your feedback!

closed 2024-02-22 by board707