board707/DMD_STM32 #83: Add a new panel please.

boba7, 2024-02-26:
## Panel description:  P10
#### Dimensions (in pixels) : 32x16
#### Scan factor: 1/2 
#### Chips on rear side:  MBI5124GP A0SF94CTHB,  FS4953L 2552ML, 74HC245

## Panel photos
[image]

## Sample videos
After i played around with it it looks like 32x16 1/2 DIRECT, colors work great, but pixels are all over the place. Managed to do this rgb_pattern_test with 2,32,16,2,1
https://drive.google.com/file/d/11GrxtMxfKp6CHncX78fGxA4tgWcdIvDC/view?usp=drivesdk
## Specifications

  - Library version: 1.1.2
  - Platform: STM32
  - MCU board: STM32F103C
  - Connections: used pins: A = PB6, B = PB5, C = NC, D = NC, E = NC,  nOE = PB0, SCLK = PA7, CLK = PA6, R0 = PA0, G0 = PA1, B0 = PA2, R1 = PA3, G1 = PA4, B1 = PA5

board707, 2024-02-26:
Hi
Please show the video for rgb_pattern_test with 2,32,16,2,2

boba7, 2024-02-26:
Here it is with 2,32,16,2,2
https://drive.google.com/file/d/11x0oFkoV6i7-9aCGgfVNz-fBaXZDX8PG/view?usp=drivesdk

board707, 2024-02-26:
Thank you, I'll take a time to figure out this.

board707, 2024-02-26:
Hi
Insert the code below in your sketch just after line
`#include "DMD_RGB.h"`

```
#define RGB32x16_s2_boba7 2,32,16,2,63

template<int COL_DEPTH>
class DMD_RGB<RGB32x16_s2_boba7, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
  public:
    DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
            byte panelsWide, byte panelsHigh, bool d_buf = false) :
      DMD_RGB_BASE2<COL_DEPTH>(2, mux_list, _pin_nOE, _pin_SCLK, pinlist,
                               panelsWide, panelsHigh, d_buf, COL_DEPTH, 2, 32, 16)
    {
      this->fast_Hbyte = false;
      this->use_shift = false;
    }
    // Fast text shift is disabled for complex patterns, so we don't need the method
    void disableFastTextShift(bool shift) override {}

protected:
	uint16_t get_base_addr(int16_t &x, int16_t &y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
		x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
              
		static const uint8_t px_base = 8;
                static const uint8_t D_tbl[4] = { 3, 2, 1, 0 };
                static const uint8_t C_tbl[4] = { 1, 0, 1, 0 };

		uint16_t base_addr = (pol_y % this->nRows) * this->x_len + (x / px_base) * this->multiplex * px_base ;
                uint8_t Oktet_m = pol_y / this->nRows;
                base_addr += D_tbl[Oktet_m] * px_base;
                if ( C_tbl[Oktet_m] == 1) {
                 base_addr += (px_base -1) - x % px_base;
                }
               else
               {
                 base_addr += x % px_base;
               }
		return base_addr;

		}

};
```

boba7, 2024-02-26:
It's working! Thank you!
[image]

board707, 2024-02-26:
Added your matrix to main library.
You can use it just as `RGB32x16_s2_boba7` pattern.

Thank you for trying the library.

vandana-itspe, 2024-04-23:
hi @board707, I am using RGB32x16_s2_boba7 matrix and thanks for the library, I am facing a small issue, the right side of the board is flickering and the color is comparatively dim. Could you help me out with this?

board707, 2024-04-23:
Hi
Could you show the video illustrated the issue?

vandana-itspe, 2024-04-23:
> Hi Could you show the video illustrated the issue?

Hi, here is the video. 

https://github.com/board707/DMD_STM32/assets/151005627/bd4e5ab6-a8c2-4c40-bc87-d62863b2bd21

the dimming part is not clearly visible in the video, but I can see with the naked eye

board707, 2024-04-23:
It looks like a broken panel. Do you have another one for test?

vandana-itspe, 2024-04-23:
It is not broken, I have cascaded 2 (32x16) panels in parallel.

board707, 2024-04-23:
Please describe your connections in more detail. The RGB panels can't connected in parallel, you should chain it one to another.

vandana-itspe, 2024-04-23:
yes, I have connected in a daisy chain,  the output of one panel goes to input of next.

board707, 2024-04-23:
Try to swap first and second panels

vandana-itspe, 2024-04-23:
As I swapped the panels, the issue has moved to above panel.

board707, 2024-04-23:
If the issue does still on the same panel - it means that the problem is in the panel itself.

vandana-itspe, 2024-04-23:
but the issue has shifted to another panel after I swap, before it was on the bottom panel and after swap it is on the above panel

board707, 2024-04-23:
Please correct me if I wrong....
First the issue was on bottom panel, right?
Then you swapped the panels, and issue now is in the above panel, isn't it?
But when you swapped the panels, the bottom panel became the top - therefore the issue now in the same panel as before.