board707/DMD_STM32 #85: how to add number with different background

maxmurugan, 2024-02-29:
/*--------------------------------------------------------------------------------------
Demo for RGB panels

DMD_STM32a example code for STM32F103xxx board
------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#include "Arduino.h"
//###########################################################################################
#include "st_fonts/SystemFont5x7.h"
#include "st_fonts/Font_13x14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"

#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
#define ENABLE_DUAL_BUFFER true

#define  Relay_Out 3
// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1
DMD_RGB <RGB32x32_S8_maxmurugan, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_Standard_Font systemFont5x7(SystemFont5x7);
DMD_Standard_Font font7x16_usb(Font_13x14);
const int dimmerValues[11] = {10, 25 , 50 , 75 , 90, 120, 140, 175, 180 , 200 , 220};
int16_t brightness=5;

void setup(void)
{
// initialize DMD objects
  dmd.init();
  Serial.begin(9600);

}

/*--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------*/
void loop(void)
{
  uint16_t bg = 0; // background - black
    // create foreground colors
    uint16_t col[] = {
    dmd.Color888(255,0, 0), // red
    dmd.Color888(0, 255, 0), // green
    dmd.Color888(255, 255, 0), // blue
    dmd.Color888(0, 255, 255), // blue
    dmd.Color888(255, 255, 255), // blue
    
    };
dmd.selectFont(&font7x16_usb);
dmd.setBrightness(dimmerValues[brightness]);
// set text foreground and background colors
  dmd.setTextColor(col[0], col[01]);
while (1)
{
  dmd.setTextColor(col[0], col[01]); // background color
  dmd.drawString(0,1,"1",1,col[4]);
  dmd.setTextColor(col[0], col[0]);
  dmd.drawString(17,1,"2",1,col[4]);
  dmd.setTextColor(col[0], col[2]);
  dmd.drawString(0,17,"0",1,col[3]);
  dmd.setTextColor(col[0], col[01]);
  dmd.drawString(17,17,"1",1,col[4]);
  dmd.swapBuffers(true);

}
}

this code is correct? add number with background color 
Now i use 4 window with different background color

please give any other solution add background color?

[image]

board707, 2024-03-01:
Hi
Is this picture a result of running your code? Looks good.
What is the issue with the code?

maxmurugan, 2024-03-02:
No issue just clarify background color code

board707, 2024-03-02:
The code is generally correct.
When you use a `.setTextColor()` with `drawString()`, only the second parameter (background) in the setTextColor is meaningful, since the text color is overwritten in the drawString().
When you use a setTextColor with marquee, both parameters matter, since the marquee does not have a separate color setting

maxmurugan, 2024-03-02:
thanks a lot

closed 2024-03-02 by maxmurugan