board707/DMD_STM32 #87: RGB P5 MODULE PIN CONNECTION

tahafathi2013, 2024-03-05:
im use p5 rgb module.
But i have not pin E on my module.
Module have 
R1 - G1
B1 - GND
R2 - G2
B2 - GND
A - B
C - D
CLK - LAT
OE - GND

[image]

tahafathi2013, 2024-03-05:
[image]

[image]

[image]

[image]

tahafathi2013, 2024-03-05:
https://github.com/board707/DMD_STM32/assets/19773695/c6e91bf9-3cfa-425d-8874-62212ecb3464

board707, 2024-03-05:
Hi, 
you have a 16scan panel, it is not need a E pin.

If you have a problem with the library with your panel, please show your test code with all connections defined.

tahafathi2013, 2024-03-05:
im use this code :

`/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux
// example:
//   #define RGB80x40plainS20  5,80,40,20,0		// 5 mux pins(A B C D E), 80x40, scan 1/20, pattern 0
//
//             pattern_ID is unique number for pattern design
//			   pattern ID between 0 and 31 and after 255 are reserved for library use
//                       and from 32 to 127 - for user panels, added by author
//            for new panels added yourself use pattern_ID between 128 and 255
//
DMD_RGB <RGB64x32plainS16  , 4> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
`

im connected led panel to stm32 pins:

#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8

and connected this pins : 

uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

my panel dont have sclk pin to connect PB7 but im connect this pin to PA15

board707, 2024-03-05:
Don.t confuse the CLK and the SCLK pins.
The library SCLK pin is the same as LAT pin on the panel.
You should connect PA15 to CLK and PB7 to LAT.

tahafathi2013, 2024-03-06:
how fix it???

board707, 2024-03-06:
Sorry, l am not home now.

tahafathi2013, 2024-03-06:
Ok
No problem.
When you at home chek this problem and help me  to fix it.please

board707, 2024-03-06:
Do you use a st-link to upload the code to the stm32?

tahafathi2013, 2024-03-06:
No im using serial programmer

tahafathi2013, 2024-03-06:
how fix it problem??

tahafathi2013, 2024-03-06:
[image]

tahafathi2013, 2024-03-06:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false
#define LED_PIN PC13

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// CLK, R0, G0, B0, R1, G1, B1
#define DMD_PIN_R0 PA0
#define DMD_PIN_G0 PA1
#define DMD_PIN_B0 PA2
#define DMD_PIN_R1 PA3
#define DMD_PIN_G1 PA4
#define DMD_PIN_B1 PA5
#define DMD_CLK PA15

// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PA7
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { DMD_CLK, DMD_PIN_R1, DMD_PIN_G1, DMD_PIN_B1, DMD_PIN_R0, DMD_PIN_G0, DMD_PIN_B0 };  // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB<RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
/// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

uint16_t bg = 0;  // background - black
uint16_t fg = 50;   // foreground

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void) {
  // initialize DMD objects
  dmd.init();
  fg = dmd.Color888(0, 0, 255);
  dmd.setBrightness(200);
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {

  // fill the matrix with points row by row
  for (int i = 0; i < dmd.height(); i++) {
    for (int j = 0; j < dmd.width(); j++) {
      dmd.drawPixel(j, i, fg);
      delay(30);
    }
  }
  // clear the screen
   dmd.fillScreen(bg);
 
}

tahafathi2013, 2024-03-06:
I remove DMD_PIN_E

tahafathi2013, 2024-03-06:
[image]

board707, 2024-03-06:
Try to use pb8 to pin D and pb10 as pin C.
Also you have an incorrect order of RGB pins: pins R0G0B0 should be first and then R1 G1 B1

tahafathi2013, 2024-03-06:
I try it but not working yet.
When my hand near wire of clk pin 

https://github.com/board707/DMD_STM32/assets/19773695/fb0f8e36-ec3b-4ae3-92df-6cf85afbe95a

[image]

tahafathi2013, 2024-03-06:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false
#define LED_PIN PC13

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB10
#define DMD_PIN_D PB8
// CLK, R0, G0, B0, R1, G1, B1
#define DMD_PIN_R0 PA0
#define DMD_PIN_G0 PA1
#define DMD_PIN_B0 PA2
#define DMD_PIN_R1 PA3
#define DMD_PIN_G1 PA4
#define DMD_PIN_B1 PA5
#define DMD_CLK PA15

// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PA7
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { DMD_CLK, DMD_PIN_R0, DMD_PIN_G0, DMD_PIN_B0, DMD_PIN_R1, DMD_PIN_G1, DMD_PIN_B1 };  // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB<RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
/// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

uint16_t bg = 0;  // background - black
uint16_t fg = 50;   // foreground

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void) {
  // initialize DMD objects
  dmd.init();
  fg = dmd.Color888(0, 255, 0);
  dmd.setBrightness(200);
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void) {
 fg = dmd.Color888(0, 0, 255);
  // fill the matrix with points row by row
  for (int i = 0; i < dmd.height(); i++) {
    for (int j = 0; j < dmd.width(); j++) {
      dmd.drawPixel(j, i, fg);
      delay(30);
    }
  }
  // clear the screen
   dmd.fillScreen(bg);
   
   
   fg = dmd.Color888(0, 255, 0);
  // fill the matrix with points row by row
  for (int i = 0; i < dmd.height(); i++) {
    for (int j = 0; j < dmd.width(); j++) {
      dmd.drawPixel(j, i, fg);
      delay(30);
    }
  }
  // clear the screen
   dmd.fillScreen(bg);
   
   
   fg = dmd.Color888(255, 0, 0);
  // fill the matrix with points row by row
  for (int i = 0; i < dmd.height(); i++) {
    for (int j = 0; j < dmd.width(); j++) {
      dmd.drawPixel(j, i, fg);
      delay(30);
    }
  }
  // clear the screen
   dmd.fillScreen(bg);
 
}

board707, 2024-03-07:
Sorry, I don't have any additional ideas.

tahafathi2013, 2024-03-07:
DMD_RGB<RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
IN this line code need to change RGB64x32plainS16, COLOR_4BITS????

why when CLK pin touched by my finger its working ?

board707, 2024-03-07:
>why when CLK pin touched by my finger its working ?

The matrix code uses a high frequency pulses (dozen megahertz) and it is very sensitive to unreliable connections. Do not use separate wires, use dedicated 16pin connector.

tahafathi2013, 2024-03-07:
Choose wich cpu spee?
Im use 16 pin connector or wires but dont work it.
My panel dont have pin  E but im remove pin E from code.
What is configure without pin E??
Please get me example without pin E to test screen

board707, 2024-03-07:
Pin E is not used for matrices with 16 scans or lower, so this setting doesn't matter for your configuration.
Honestly, I don't see any problems with your pins setup. I don't have an idea why it not works.

tahafathi2013, 2024-03-08:
ok
your examples why not working adafruit gfx 1.11 ??
what your panel models tested by your examples stm32 f103c8t6c???

tahafathi2013, 2024-03-10:
It may be because of uploading the code through serial? It means that it has not been transmitted through the correct serial?
Could it be because of the (https://github.com/rogerclarkmelbourne/Arduino_STM32)?

tahafathi2013, 2024-03-10:
[image]

When programmed with stlink

tahafathi2013, 2024-03-10:
When press reset key moving leds.

https://github.com/board707/DMD_STM32/assets/19773695/bee7594e-a650-46df-a0c3-855354a1219b

digi55, 2024-03-10:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux
// <RGB80x40plainS20> - 80x40 matrix with 20scans

DMD_RGB <RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
//DMD_RGB <RGB80x40plainS20, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
//DMD_RGB <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
  enableDebugPorts();
  dmd.init();
  fg = dmd.Color888(0, 0, 255);
  dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}

digi55, 2024-03-10:
Hi, Try this Code. Working without problem with me.

digi55, 2024-03-10:
Thank you so much @board707

tahafathi2013, 2024-03-10:
> /*-------------------------------------------------------------------------------------- Demo for RGB panels
> 
> DMD_STM32a example code for STM32F103xxx board
> 
> Pattern test for 32x16 1/2 1/4 RGB panels ------------------------------------------------------------------------------------- */ #include "DMD_RGB.h"
> 
> //Number of panels in x and y axis #define DISPLAYS_ACROSS 1 #define DISPLAYS_DOWN 1
> 
> // Output buffering - false #define ENABLE_DUAL_BUFFER false
> 
> // ==== DMD_RGB pins ==== // mux pins - A, B, C... all mux pins must be selected from same port! #define DMD_PIN_A PB6 #define DMD_PIN_B PB5 #define DMD_PIN_C PB4 #define DMD_PIN_D PB3 #define DMD_PIN_E PB8 // put all mux pins at list uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };
> 
> // pin OE must be one of PB0 PB1 PA6 PA7 #define DMD_PIN_nOE PB0 #define DMD_PIN_SCLK PB7
> 
> // Pins for R0, G0, B0, R1, G1, B1 channels and for clock. // By default the library uses RGB color order. // If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins. // All this pins also must be selected from same port! uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1
> 
> // Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
> 
> // For pattern testing use one of matrix below: // RGB32x16plainS4 - 32x16 1/4 Binary mux // RGB32x16plainS4_DIRECT - 32x16 1/4 DIRECT mux // RGB32x16plainS2_DIRECT - 32x16 1/2 Binary mux // - 80x40 matrix with 20scans
> 
> DMD_RGB <RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER); //DMD_RGB <RGB80x40plainS20, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER); //DMD_RGB <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
> 
> uint16_t bg = 0; // background - black uint16_t fg = 0; // foreground
> 
> void setup(void) { enableDebugPorts(); dmd.init(); fg = dmd.Color888(0, 0, 255); dmd.setBrightness(50); }
> 
> void loop(void) { // fill the matrix with points row by row for (int i = 0; i < dmd.height(); i++) { for (int j = 0; j < dmd.width(); j++) { dmd.drawPixel(j,i, fg); delay(30); } } // clear the screen dmd.fillScreen(bg); }

what use board?? 
im using bluepill
it work on bluepill?

digi55, 2024-03-10:
Yes, Iam using also P5 panel like yours and bluepill. 
Be careful about pin number

tahafathi2013, 2024-03-10:
[image]

I using your code but not working .
Can you upload image from your wire connections an your panel details.
My panel not have E pin.

digi55, 2024-03-10:
Picture won't help since I'm using a FLAT cable. Pin numbers are as seen in the sample code I sent.
PinE is not required on this panel.

tahafathi2013, 2024-03-10:
i when use fg = dmd.Color888(255, 255, 255); not change the color.

tahafathi2013, 2024-03-10:
> Picture won't help since I'm using a FLAT cable. Pin numbers are as seen in the sample code I sent. PinE is not required on this panel.

my wire connections same as sample code.but not working

digi55, 2024-03-10:
This code fills the panel with blue dots.

tahafathi2013, 2024-03-10:
I change the colors

 fg = dmd.Color888(255, 255, 255)
But not working.

digi55, 2024-03-10:
send a picture of your connections

digi55, 2024-03-10:
I will check for you

tahafathi2013, 2024-03-10:
Are you install and configure stm boards from https://github.com/rogerclarkmelbourne/Arduino_STM32 ???
Change any thing on your board like resistors?

I burn bootloader by cub programmer. And upload the code by stlink.

tahafathi2013, 2024-03-10:
[image]

[image]

tahafathi2013, 2024-03-10:
I change it but not working yet.

[image]

digi55, 2024-03-10:
Sorry ,I don't have any advice

tahafathi2013, 2024-03-10:
> Sorry ,I don't have any advice

Is it possible that the micro or the micro flash memory is damaged?

tahafathi2013, 2024-03-10:
Are you install and configure stm boards from https://github.com/rogerclarkmelbourne/Arduino_STM32 ???
Change any thing on your board like resistors?

digi55, 2024-03-10:
yes,  I am using https://github.com/rogerclarkmelbourne/Arduino_STM32 
No, Nothing change

tahafathi2013, 2024-03-10:
are you install maple driver? 
i just copy arduino_stm32 folder to my doucument/arduino/hardware .
and install sam cortext_m3 and use your code

tahafathi2013, 2024-03-11:
what is stlink pin connections to stm32 bluepiil?
i need to disconnect stlink programmer after uploading code???
can you make a vide from pin connection and your programmer and your arduino board setting? please

closed 2024-04-10 by board707