board707/DMD_STM32 #88: Add a new panel p4 GKGD

OkSingra, 2024-03-12:
## Panel description 
#### Dimensions (in pixels) : 80x40
#### Scan factor: 1/10
#### Chips on rear side:  TC7258, DP5125, HCT245

## Panel photos
[image]
[image]

## Sample videos
https://youtu.be/0p1VcwUeOEg?si=G7sypG-Y_HRYbILC
## Specifications

  - Library version: 1.01
  - Platform: Pico
  - MCU board: rpi 2040
  - Connections: directly

board707, 2024-03-12:
Hi
I am on vacation now, I'll return after 20th of March

board707, 2024-03-20:
Hi
Which patern used in your video?

OkSingra, 2024-03-21:
DMD_RGB <RGB32x16plainS4, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

board707, 2024-03-21:
Hmmm...

Please, add the line 
```
#define RGB80x40_S10_OKSingra     	4,80,40,10,2	
```
and show the video with this pattern.

OkSingra, 2024-03-22:
https://youtu.be/CrBiFKOpCQ8?si=rMaDW9aubWuPBPhj

board707, 2024-03-22:
Thank you.
Add this code after `#include DMD_RGB.h` line
```
#define RGB80x40_S10_OKSingra     	4,80,40,10,16	
template <int MUX_CNT, int P_Width, int P_Height, int SCAN, int COL_DEPTH>
class DMD_RGB<MUX_CNT, P_Width, P_Height, SCAN, 16, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
	DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
		byte panelsWide, byte panelsHigh, bool d_buf = false) :
		DMD_RGB_BASE2<COL_DEPTH>(MUX_CNT, mux_list, _pin_nOE, _pin_SCLK, pinlist,
			panelsWide, panelsHigh, d_buf, COL_DEPTH, SCAN, P_Width, P_Height)
	 {
      this->fast_Hbyte = false;
      this->use_shift = false;
      }
    // Fast text shift is disabled for complex patterns, so we don't need the method
    void disableFastTextShift(bool shift) override {}

protected:
	uint16_t get_base_addr(int16_t& x, int16_t& y) override {
		this->transform_XY(x, y);
		uint8_t pol_y = y % this->pol_displ;
                static const uint8_t pix_base = 16;
		x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
		uint16_t base_addr = (pol_y % this->nRows) * this->x_len +
			(x / pix_base) * this->multiplex * pix_base +
			x % pix_base ;
		if (pol_y < this->nRows)  base_addr += pix_base;
               
               return base_addr;
	}
};
```

OkSingra, 2024-03-22:
Everything works fine Thanks

board707, 2024-03-22:
I will be glad if the library turns out to be useful.

closed 2024-03-22 by board707