board707/DMD_STM32 #91: Add a new panel please

digi55, 2024-03-22:
## Panel description  P2.5
#### Dimensions (in pixels) : 128x64
#### Scan factor: 1/32
#### Chips on rear side:  DP3364, HCT245

## Panel photos
[image]
[image]
[image]
[image]

## Sample videos

## Specifications

  - Library version: 2.0
  - Platform: STM32
  - MCU board: bluepill STM32103C8T6
  - Connections: used pins  A=PB6, B=PB5, C=PB4, D= PB3, E=PB8 nOE=PB0, SCLK=PB7 CLK=PA6, R0=PA0, G0=PA1, B0=PA2,
R1=PA3, G1=PA4, B1=PA5

board707, 2024-03-22:
Hi
Unfortunately, but prognosis is negative: I didn't find anything about DP3364 driver, but its closest analog, DP3264 is S-PWM type and incompatible with the library.

Try to test the panel with DMD_RGB_SHIFTREG_ABC class and pattern RGB128x64plainS32

board707, 2024-03-25:
It is what I expected...
Unfortunately, this panel will not be supported until someone other can run it with Arduino.

closed 2024-03-26 by board707

funnymind, 2024-08-30:
Hi @board707, I have similar panels now, I saw that you were ready to do some tests on them, if you still don't mind then I'm ready.