board707/DMD_STM32 #95: Switch On the Pico board Showing horizontal line after few second working

maxmurugan, 2024-04-03:
https://github.com/board707/DMD_STM32/assets/4962604/87debcaf-8ff6-44f5-af7b-1d7f05d92a5b

// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A 6
#define DMD_PIN_B 7
#define DMD_PIN_C 8
#define DMD_PIN_D 9
#define DMD_PIN_E 10
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };
#define DMD_PIN_nOE 15
#define DMD_PIN_SCLK 12
uint8_t custom_rgbpins[] = { 11,16,17,18,19,20,21 }; // CLK, R0, G0, B0, R1, G1, B1

DMD_RGB_SHIFTREG_ABC <RGB80x40_S10_LNikon, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

using this setting Pico board

how to solve the issue ,i try another board same issue will happened

maxmurugan, 2024-04-03:
https://github.com/board707/DMD_STM32/assets/4962604/6b14d719-81bd-489a-96b7-b96628c8b890

maxmurugan, 2024-04-03:
https://github.com/board707/DMD_STM32/assets/4962604/cd3a27c2-7fbc-4969-a50e-7ee6e5fe2f1b

maxmurugan, 2024-04-03:
please check

maxmurugan, 2024-04-03:
DP5125
DP32020A
P4 - 1/10 scan

board707, 2024-04-03:
Hi
Could you provide a more info? 
In your video there is a defect at first, and then the picture is normal. Did I understand correctly that the defect is observed only for a few seconds?
What version of the library do you have? Haven't you updated it recently? Did you see this issue before?

maxmurugan, 2024-04-03:
version=1.0.6
I am using same library P10 RGB Module working fine

Now Try to P4 Module RGB issue will happened

maxmurugan, 2024-04-03:
Startup issue after few second working   change to 3 to  4 module  same issue will happened

board707, 2024-04-03:
> Startup issue after few second working change to 3 to 4 module same issue will happened

I didn't understand this, sorry

board707, 2024-04-09:
> i have change another SMPS problem solved

What is SMPS?

maxmurugan, 2024-04-10:
XPS   5V/20A

closed 2024-04-10 by board707