board707/DMD_STM32 #96: Add a new panel

miralay001, 2024-04-04:
## Panel description 
#### Dimensions (in pixels) : 16x32
#### Scan factor: 1/4
#### Chips on rear side:  DP5125

## Panel photos

## Specifications

  - Library version: 1.01
  - Platform: STM32
  - MCU board: blackpil STM32F103C8

https://github.com/board707/DMD_STM32/assets/108148338/86d475bb-14ee-4fe4-9468-3290ecbf2f1b

I'm waiting for your help

board707, 2024-04-04:
Hi
It looks like I worked with similar panel.

Please add this code snippet to your sketch just after #include "DMD_RGB.h" line:
Use template RGB32x16_S4_miralay001 for init of the DMD class instance.
```
template<int COL_DEPTH>
#define RGB32x16_S4_miralay001 2,32,16,4,65
class DMD_RGB<RGB32x16_S4_miralay001, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
	{
	public:
		DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
			byte panelsWide, byte panelsHigh, bool d_buf = false) :
			DMD_RGB_BASE2<COL_DEPTH>(2, mux_list, _pin_nOE, _pin_SCLK, pinlist,
				panelsWide, panelsHigh, d_buf, COL_DEPTH, 4, 32, 16)
			{
			this->fast_Hbyte = false;
			this->use_shift = false;
			}
		// Fast text shift is disabled for complex patterns, so we don't need the method
		void disableFastTextShift(bool shift) override {}

	protected:
		uint16_t get_base_addr(int16_t& x, int16_t& y) override {
                        static const uint8_t  pix_base =8;
			this->transform_XY(x, y);
			uint8_t pol_y = y % this->pol_displ;
			x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
			 uint16_t base_addr = (pol_y % this->nRows) * this->x_len +
                           (x / pix_base) * this->multiplex * pix_base +
                           x % pix_base;
                          base_addr += pix_base * (pol_y / this->nRows) ;
			 return base_addr;
			}
	};
```

miralay001, 2024-04-04:
Thank You :)

https://github.com/board707/DMD_STM32/assets/108148338/f6dd7967-7b71-4f38-9482-ef69fb2f4a1e

board707, 2024-04-04:
You are welcome :)

closed 2024-04-05 by board707

board707, 2024-04-11:
Pattern `RGB32x16_S4_miralay001` added to the library v1.1.3. Now you can use it directly in the DMD class init:
```
DMD_RGB <RGB32x16_S4_miralay001, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```