board707/DMD_STM32 #99: Multiline scrolling possible

maxmurugan, 2024-04-12:
hi sir 

Currently Now multiline Fixed text and one scrolling text  working fine 

how to add multiline  Scrolling text?

board707, 2024-04-12:
It could be done the same way as "fixed text and one scrolling text'. Outputs the "fixed text" on a new position on each step - and you will get a scrolling.

maxmurugan, 2024-04-12:
give some sample code please

board707, 2024-04-12:
Please show your code for "fixed text and one scrolling text'.

maxmurugan, 2024-04-12:
void loop(void)
{
dmd.selectFont(&UkrRusArial_F);
uint16_t bg = 0; // background - black
uint16_t interval = 32;
long prev_step = millis();
while (1)
{
  /* if ((millis() - prev_step) > interval)
    {
    dmd.setTextColor(col[R1], bg); // --red color
    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);
    Refresh =0;
    dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, (dmd.height() > 0) ? 0 : 0, 0);
    }
    dmd.swapBuffers(true);
    prev_step = millis();
    }*/

    if ((millis() - prev_step) > interval)
      {
        prev_step = millis();
        dmd.setTextColor(col[0], bg); // --red color
        if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
        dmd.clearScreen(true);
        dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, (dmd.height() > 0) ? 8 : 0, 0);
        }
        dmd.drawStringX(0, 0, "FIXED TEXT", col[1]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
        dmd.swapBuffers(true);
     }
  
}
}

https://github.com/board707/DMD_STM32/assets/4962604/8e8ac2a0-ce60-46e0-9e45-03ee6668235f

board707, 2024-04-12:
Thank you.
Could you try this?
```
if ((millis() - prev_step) > interval)
  {
    static uint8_t i =0;
    prev_step = millis();
    dmd.setTextColor(col[0], bg); // --red color
    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);
    dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, (dmd.height() > 0) ? 8 : 0, 0);
    }
   
      dmd.drawStringX(i, 0, "FIXED TEXT", col[1]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      i++;
      if (i >= dmd.width() ) i =0;
    
    dmd.swapBuffers(true);
 }
```

maxmurugan, 2024-04-12:
Uploading IMG_0798.MOV…

maxmurugan, 2024-04-12:
https://github.com/board707/DMD_STM32/assets/4962604/140bd247-c084-46d9-950a-22cb62b983c2

maxmurugan, 2024-04-12:
from left need full scrolling ,like bottom line

maxmurugan, 2024-04-12:
change to left scrolling 
please check

if ((millis() - prev_step) > interval)
  {
    static int8_t i = dmd.width();
    prev_step = millis();
    dmd.setTextColor(col[0], bg); // --red color
    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);
    dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, (dmd.height() > 0) ? 8 : 0, 0);
    }
   
      dmd.drawStringX(i, 0, "FIXED TEXT", col[1]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      i--;
      if (i <= -dmd.width() ) 
      {
        i = dmd.width();
      }
    
    dmd.swapBuffers(true);
 }

https://github.com/board707/DMD_STM32/assets/4962604/fcc683fa-c64f-4905-a27f-8405f9a13903

board707, 2024-04-12:
```
if ((millis() - prev_step) > interval)
  {
    static int16_t i =0;
    prev_step = millis();
    dmd.setTextColor(col[0], bg); // --red color
    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);
    dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, (dmd.height() > 0) ? 8 : 0, 0);
    }
   
      dmd.drawStringX(i, 0, "FIXED TEXT", col[1]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      i++;
      if (i >= dmd.width() ) i = -1 * (stringWidth("FIXED TEXT"));
    
    dmd.swapBuffers(true);
 }
```

maxmurugan, 2024-04-12:
please check video 

https://github.com/board707/DMD_STM32/assets/4962604/0bfde3e1-e997-43c6-831f-fda656e417f3

maxmurugan, 2024-04-12:
wow nice working 
thanks a lot

maxmurugan, 2024-04-12:
Top line right side end of text not clear
how to clear?

board707, 2024-04-12:
may be it helps:
```
if ((millis() - prev_step) > interval)
  {
    static int16_t i =0;
    prev_step = millis();
    dmd.setTextColor(col[0], bg); // --red color
    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);
    dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, (dmd.height() > 0) ? 8 : 0, 0);
    }
   
      dmd.drawStringX(i, 0, "FIXED TEXT", col[1]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      i++;
      if (i >= (dmd.width() +2) ) i = -1 * (stringWidth("FIXED TEXT"));
    
    dmd.swapBuffers(true);
 }
```

maxmurugan, 2024-04-12:
same issue will happened

board707, 2024-04-12:
ok, I will to see at the evening

board707, 2024-04-12:
It is seems to work:
```
const char Fixed[] = "FIXED TEXT";
int16_t fixed_w = 0;
int16_t min_y, max_y;
dmd.stringBounds(Fixed, 0, &fixed_w, &min_y, &max_y);
while (1)
{

if ((millis() - prev_step) > interval)
  {
    static int16_t i =0;
    prev_step = millis();
    dmd.setTextColor(col[0], bg); // --red color
    if (dmd.stepMarquee(-1, 0) & 1) { // if text is reached screen bounds
    dmd.clearScreen(true);
    dmd.drawMarquee(Final_Message, strlen(Final_Message) , dmd.width() - 5, 16, 0);
    }
      dmd.drawFilledBox(i-2, min_y, i-1, max_y, bg);  // clear artefacts after moving the text
      dmd.drawStringX(i, 0, Fixed, col[1]); //- fixed messAGE USED FOR CLOCK OR DATE USING DS3231
      i++;
      if (i >= (dmd.width()+2) ) i = -1 * fixed_w;
    
    dmd.swapBuffers(true);
 }

}
```

maxmurugan, 2024-04-13:
Now working fine

https://github.com/board707/DMD_STM32/assets/4962604/8db425a3-00af-49c0-8f4f-c800bcf19e9a

board707, 2024-04-13:
Thenk you
Can I use your videos as illustrations for a library manual?

maxmurugan, 2024-04-13:
ok sir
thanks

closed 2024-04-18 by board707