board707/DMD_STM32 #104: Add a new panel

ShiPC123, 2024-04-30:
## Panel description 
#### Dimensions (in pixels) : 80*40
#### Scan factor: 1/10
#### Chips on rear side: ,DP5125D RUL8158C

## Panel photos

[image]
[image]

## Sample videos

## Specifications

  - Library version: 1.1.4
  - Platform: STM32
  - MCU board: blackpil STM32F103RCT6
  - Connections: CLK  --  PA0   R0 --PA1  G0 --PA2  B0 --PA3   R1 --PA4  G1 --PA5  B1 --PA6  OE --PB0  LAT--PB1  A--PB15 B--PB14 C--PB13  D--PB12

ShiPC123, 2024-04-30:
I run the test code  Interface and MCU options are shown 
[image]
[image]
[image]

board707, 2024-05-02:
Hi
According with paper label on the panel, it used RUL5158C as multiplexor. This chip seems to be shift register type, so you need to use `DMD_RGB_SHIFTREG_ABC` class rather than `DMD_RGB`.

Add define below at the start of the sketch:
`#define RGB80x40_S10_ShiPC123 3,80,40,10,2`

Add setup the DMD class as
```
DMD_RGB_SHIFTREG_ABC <RGB80x40_S10_ShiPC123, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```
Run the `pattern_test` example with editions, mentioned above and show the video.

ShiPC123, 2024-05-05:
I ran the example, and the display screen was displayed in four places, the upper left corner was 1, the upper right corner was 2, the lower left corner was 3, the right small corner was 4, and the display sequence was 1-2-3-4
[image]

ShiPC123, 2024-05-05:
https://github.com/board707/DMD_STM32/assets/132763025/c4342f96-e5ed-4fce-bf4b-8fcb6e371bfc

ShiPC123, 2024-05-05:
If I use the following parameters #define RGB80x40_S10_ShiPC123 3,80,40,10,1
The display effect is as follows

https://github.com/board707/DMD_STM32/assets/132763025/c5dd10b5-355d-4b00-9a6f-f198e284debd

board707, 2024-05-05:
@ShiPC123 
Thank you for your tests. 
Unfortunately, the last file seems to be corrupted, please upload the video again.

ShiPC123, 2024-05-05:
Hello， code encountered an error
[image]

ShiPC123, 2024-05-05:
class DMD_RGB_SHIFTREG_ABC : public DMD_RGB<MUX_CNT, P_Width, P_Height, SCAN, SCAN_TYPE, COL_DEPTH>

Only 6 arguments can be passed in a function

board707, 2024-05-06:
O... sorry for this mistake.
Please replace the file `DMD_Panel_Templates.h` in the library to the file from archive, attached to the message.

[DMD_Panel_Templates.zip](https://github.com/board707/DMD_STM32/files/15217423/DMD_Panel_Templates.zip)

ShiPC123, 2024-05-06:
Hi, I tested #define RGB80x40_S10_ShiPC123, 3,80,40,10,101,4,PATTERN_UPS

Results like video

https://github.com/board707/DMD_STM32/assets/132763025/99e2096e-551a-494a-905c-78af11622976

board707, 2024-05-06:
Thank you
try this
`#define RGB80x40_S10_ShiPC123, 3,80,40,10,102,4,PATTERN_UPS`

ShiPC123, 2024-05-06:
Thank you very much ,that the display is now working ,Thank you. Thank you

board707, 2024-05-06:
Thanks for your feedback.
I will add the new pattern to the main library and let you know so you could use it without editing the library.

board707, 2024-05-06:
New pattern added.
Your can use it as any other library pattern, without additional definition:
`DMD_RGB_SHIFTREG_ABC <RGB80x40_S10_ShiPC123, COLOR_4BITS> dmd(mux_list....`

closed 2024-05-06 by board707