board707/DMD_STM32 #108: flickering problem in 2 panels

digi55, 2024-05-13:
Hi, 
I finally found a 128x64 panel compatible with our library.
Pattern testing, Bitmap testing etc. everything is fine.
But I need to use 2 panels in my project. (128x128)
In this case, flickering and some color problems occur on the panel.
Do you have any advice?

in this project, I am using stm32F103RCT6 (256K 72Mhz)
[image]
[image]

board707, 2024-05-13:
> But I need to use 2 panels in my project. (128x128)

It seems to me that your board not have enough memory for this.
As i see in the datasheet, stm32F103RCT6 has a 48K RAM, while the 128x128 scrren requires a 64K
(128*128 * 4 bytes) = 65536

digi55, 2024-05-13:
Thank you for your answer, I will try with stm32F103RET6

board707, 2024-05-13:
> I will try with stm32F103RET6

It also won't work, because the F103RET6 has a 64K, but you need 64k for your screen only. 
Choose the board with RAM 96K or more

digi55, 2024-05-13:
You are right, I should look for a processor with 96k ram.

board707, 2024-05-13:
Take in account a "Blackpill" board STM32F411CE (96k RAM) or Raspberry Pi Pico RP2040 - 264K

digi55, 2024-05-14:
how can use Blackpill" board STM32F411CE with Roger Clark Core.
Because don't have in list.

board707, 2024-05-14:
You probably using an old version of the core, the F411 board has added at November

digi55, 2024-05-15:
Hi again,
Today, I tested it with Blackpill STM32F411CE.
Result did not change.
Pattern test is ok.
The scrolling text sometimes has flicker and color irregularities.
also color distortion in the bitmap.

digi55, 2024-05-15:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32 

 Draw bitmap example.
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#include "bitmaps.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 2

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) command do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8

// uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = {PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
// DMD_RGB <RGB64x32plainS16,COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
//DMD_RGB <RGB80x40plainS20,COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB <RGB128x64plainS32,COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>(STM32 only) or <COLOR_4BITS> 

#define BRIGHTNESS_DEFAULT 50  // (brightness 0-255, recomended 30-100)

// bitmap images to show
const uint16_t* bitmaps[] = {tyson1, tyson2, Rocky1, Serkan3};
uint8_t ptr = 0;
/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void){
  //Serial.end();           // PA11 ve PA12 (USB) pinlerini, Standart GPIO olarak kullanabilmek için...
  enableDebugPorts();     // PA15 (JTAG) pini, Standart GPIO olarak kullanabilmek için...
	dmd.init(); 
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{
  if (ptr == (sizeof(bitmaps)/ sizeof(bitmaps[0]))) ptr =0;
  
  // initial brightness
  uint8_t b=BRIGHTNESS_DEFAULT;      
  dmd.setBrightness(b);
  
  // draw image starting from 0,0, width 64 height 32
  dmd.drawRGBBitmap(dmd.width() - 128, 0, bitmaps[ptr], 128, 128);
  
  // show the image during 15sec
  delay(3000);
  
  // fading to dark
  while (b) {
    dmd.setBrightness(b);
    delay(100);
    b--;
  }

  // blank screen 2sec
  dmd.clearScreen(true);
  delay(500);

  // increment image pointer
  ptr++;
}

digi55, 2024-05-15:
thank you for your attention!

board707, 2024-05-16:
Hi
I tested your code on the log analyzer. Unfortunately, I do not have two such matrices for direct testing, but the signals look quite good - all the delays are correct, there is no creeping of one data packet onto another. Due to the screen size, the refresh rate is low, around 80 Hz, but this should not lead to such distortion.
I would ask you to double-check all connections, especially the reliability of the GND lines.

You wrote that a code with one matrix does not have flickering. Please do two tests - connect one physical matrix and record a video for the code that uses one matrix in setup and for the code where there are two matrices in the settings, but only one is connected.

digi55, 2024-05-16:
Hi
I found the cause of the problem.
When the panel brightness value is below 100, noise and color defects occur in the image. I was able to get a perfect image with a value of 125.
This problem only exists with the p2.5 Panel (128x64). But, It works perfectly on P5 and P4 panels with every brightness value.

board707, 2024-05-16:
> This problem only exists with the p2.5 Panel (128x64). But, It works perfectly on P5 and P4 panels with every brightness value.

As I understand, you have a two p2.5 128x64 panels. Does the issue occurs on both such panels or only on a single one?

>  It works perfectly on P5 and P4 panels 

What is the pixel size of the P5 and P4 panels?

digi55, 2024-05-16:
P2.5 (128x64), I use 2 panels. A total of 128x128 panels are formed. The problem is much less obvious when using just a single panel. When I use 2 panels, the image is completely distorted.

Panels, which I tested
P5 Panel (64x32), P4 Panel (80x40)
I tested these Panels also 2 pieces of each. 64x64 and 80x80 in total

digi55, 2024-05-16:
Additionally ,
the <EEPROM.h> library, works without problem on stm32F103. 
But, on stm32F411 series gives an <EEPROM.h> error during compilation. 
Do you have any advice about this problem?
Thank you so much for your helps.

board707, 2024-05-16:
> The problem is much less obvious when using just a single panel. When I use 2 panels, the image is completely distorted.

Sorry, but I asked you to test the code for two panels with only one panel connected. This test is very important to understanding the issue. Could you please making it? If you need a further explanation how to perform the test - please ask.

board707, 2024-05-16:
> Additionally ,
the <EEPROM.h> library, works without problem on stm32F103.
But, on stm32F411 series gives an <EEPROM.h> error during compilation.

As far I know, EEPROM doesn't supports on STM32F4 in the Clark's STM32 package.

digi55, 2024-05-16:
> Sorry, but I asked you to test the code for two panels with only one panel connected. This test is very important to understanding the issue. Could you please making it? If you need a further explanation how to perform the test - please ask.

ok, i will test this and let you know.

digi55, 2024-05-16:
> As far I know, EEPROM doesn't supports on STM32F4 in the Clark's STM32 package.

Then the stm32F411 series is no longer an option for me. I need to find another processor with EEPROM support.

board707, 2024-05-16:
> I need to find another processor with EEPROM support.

Try using a  stm32F103RCT6 or RET6 again, I just realised that miscalculated the required memory. The two 128x64 panels need a 32K RAM to display.
I apologize for misleading you

digi55, 2024-05-17:
Hi again,
I tested it with only 1 P2.5 panel connected and the code set to 2 panels, the result did not change. Again noisy and distorted colors with low brightness value. If the brightness value is 100 or above, everything is normal.

During testing, I noticed that when I touched the data cable, the distortions decreased. When I checked with an oscilloscope, I detected very loud noise. I was using about 30cm data cable. I shortened this to 15 cm. All problems are completely over.

Even 50cm cable does not cause any problems in P4(80x40) and P5(64x32) panels. However, in the P2.5 (128x64) panel, data cable length is very important.

With the 15cm data cable, I can now get perfect images at all brightness levels. Thank you very much for your help. Your library works great.

I will test again with stm32F103RC.

digi55, 2024-05-17:
[image]

board707, 2024-05-17:
>  I can now get perfect images at all brightness levels. Thank you very much for your help. Your library works great.

Thank you for your feedback. Please star the library if you found it useful.

digi55, 2024-05-18:
Hi, 
I also tested it with stm32F103RCT6, there are no problems with the colors and noise in the image. However, There is flickering in the image. I think 72 Mhz speed is not enough.