board707/DMD_STM32 #109: STM32F407VGT6 + 32*16 (hub12) mono LED panel

MMB923, 2024-05-13:
## Panel description 
Hello Guys
I've used a library to run P10 LED Panel using a ST discovery board and it was successful.
As you guys see, the text is not as clear as it should be so I think there is a problem with refreshing part, As you see in video and picture attached. Can you guys help me for it?
P.S: All the libraries that I found are based on #include "Arduino.h" which is not fit with my hardware; so, the library should be based on "hal.h"
Thanks.
#### Dimensions (in pixels) : 32x16
#### Scan factor: 1/32
#### Chips on rear side:  TC7258, DP5125, HCT245

## Panel photos

https://github.com/board707/DMD_STM32/assets/38404466/482854a5-d56e-4ebb-a1ef-cbd0b90c5e65

[image]

/board707/DMD_STM32/assets/3

## Sample videos

## Specifications

  - Library version: 1.01
  - Platform: STM32
  - MCU board: Discovery F4
  - Connections: used pins

MMB923, 2024-05-13:
@board707

board707, 2024-05-13:
Is this panel a monochrome or RGB? 

> All the libraries that I found are based on #include "Arduino.h" which is not fit with my hardware;

As far as I know, the Discovery F4 can be used with Arduino. If you are ready to use Arduino code - we could try to run with your panel.
If this is not suited for you - I am sorry, I will not be able to help you

MMB923, 2024-05-13:
@board707 
Yes, it is a monochrome and about the scan rate, I made a mistake.
if it is possible to use Arduino code for discovery, please go ahead. I want to use it as well
Thanks

board707, 2024-05-13:
> if it is possible to use Arduino code for discovery, please go ahead. I want to use it as well

Yes, I hope it will work.

1.Please read the [Installation](https://github.com/board707/DMD_STM32/tree/dev-V2#installation) section in the library Readme. You first need to install the Arduino_STM32 package by Roger Clark, Adafruit_GFX library and then DMD_STM32 library.

2. Run an Arduino IDE, Tools menu and select "STM32 Discovery F407" board. Open a DMD_STM32 example `examples/STM32F4_RP2040/dmd_monochrome` and try to compile. 

If it compiled without errors, it means that you installation was successfull.

MMB923, 2024-05-14:
Thanks for your help, but it is too complicated and I need some help. I don't know how to install STM32 and I prefer to use cube IDE instead of Arduino.

board707, 2024-05-14:
Sorry, I didn't use a CubeIDE with my library and hardly can help you with it.

As about Installation of Arduino_STM32 - you can find a detailed guide there: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation
If you have any questions, please feel free to ask.

MMB923, 2024-05-14:
This one was very useful, but still I have no idea how to program it.

board707, 2024-05-14:
Did you install the core? Try to compile any example.
If succeed, than you need to install Adafruit_GFX library and DMD_STM32 library.

> I have no idea how to program it.

Hmmm...
Do you mean you don't know how to program Arduino or how to program at all?  In any way, once we'll install the library and will able to work - you will need to program the code for the panel itself. The library is not a user level product, it is for developing purpose.
What is your programming experience?

MMB923, 2024-05-14:
> Do you mean you don't know how to program Arduino or how to program at all?

I mean I'm using an Arduino file in order to program a ST Discovery board; That's that I don't know.

and about the Libraries, I've installed DMD_STM32 and Adafruit_GFX libraries

MMB923, 2024-05-14:
> > I've installed DMD_STM32
> 
> Run an Arduino IDE, Tools menu and select "STM32 Discovery F407" board. Open a DMD_STM32 example examples/STM32F4_RP2040/dmd_monochrome and try to compile.

[image]

board707, 2024-05-14:
It's seems that you missed a second point of Arduino_STM32 installation:

> Run the IDE, and on the Tools menu, select the Boards manager, and install the Arduino SAM boards (Cortex-M3) from the list of available boards. You must do this step, it installs the arm-none-eabi-g++ toolchain!

MMB923, 2024-05-14:
> missed a second point

I see. That's true. I'm installing it now. I'll inform you the update

MMB923, 2024-05-14:
> It's seems that you missed a second point of [Arduino_STM32 installation](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation):

now it is done. So, should I have an Arduino to use it as a programmer for my PCB or is it available to get any output file in .bin or .elf shape?

board707, 2024-05-14:
You could use a St-link to download the code. 
You also can use a firmware as .elf or .bin file. In order to find this files after compiling locate the line in Arduino compiler output that referred to `arm-none-eabi-objcopy`:
```
/home/user/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-objcopy -O binary /tmp/arduino_build_461166/dmd_monochrome.ino.elf /tmp/arduino_build_461166/dmd_monochrome.ino.bin
```
it point you to the temporary build folder with output files

MMB923, 2024-05-14:
> You could use a St-link to download the code. You also can use a firmware as .elf or .bin file. In order to find this files after compiling locate the line in Arduino compiler output that referred [...]

Thank you.
Just another question. As I told you before, I'm using ArduinoIDE without any Arduino hardware and I wanna program a ST processor; So, All the pin assignments should be according to to A ST processor and it is not available!!
See here:

[image]

[image]

board707, 2024-05-14:
According to the picture, you get an example for RP2040 board and not for STM32.

The pins for STM32:
```
#define DMD_PIN_A PA1
#define DMD_PIN_B PA4
#define DMD_PIN_nOE PB1
#define DMD_PIN_SCLK PA15
```
Are Discovery pins labelled other that in form "PA3" or "PC11" ?

MMB923, 2024-05-14:
> According to the picture, you get an example for RP2040 board and not for STM32.
> 
> The pins for STM32 looks like:
> 
> ```
> #define DMD_PIN_A PA1
> #define DMD_PIN_B PA4
> #define DMD_PIN_nOE PB1
> #define DMD_PIN_SCLK PA15
> ```
> 
> Are Discovery pins labelled other that in form "PA3" or "PC11" ?

[image]

[image]

All the examples that I have is visible in these pictures.
Actually I use STM32F405RGT6 on a PCB that I made. any change is needed for it? All F4 series are the same in programming

board707, 2024-05-15:
> All the examples that I have is visible in these pictures.

You have to use "DMD_STM32" library, and example `STM32F4_RP2040/dmd_monochrome` The pin names are in the conditional compilation directivas for `__STM32F4__`. If you have a problems to find the pin numbers - I will write it for you.

> Actually I use STM32F405RGT6 

Let's try to begin with "STM32 Discovery F407" board.

MMB923, 2024-05-15:
> > All the examples that I have is visible in these pictures.
> 
> You have to use "DMD_STM32" library, and example `STM32F4_RP2040/dmd_monochrome` The pin names are in the example inside conditional compilation directivas for `__STM32F4__`. If you have a problems to find the pin numbers - I will write it for you.
> 
> > Actually I use STM32F405RGT6
> 
> Let's try to begin with "STM32 Discovery F407" board.

It's OK. But I don't have discovery now. Just My PCB, Bluepill and Arduino are available for now.
Maybe I can Use F1 Bluepill now, then change it to F4?

board707, 2024-05-15:
Of course try with bluepill

MMB923, 2024-05-18:
> Of course try with bluepill

Not working bro :(

board707, 2024-05-18:
Please show your code

closed 2024-08-23 by board707