board707/DMD_STM32 #113: I need help with RGB panel and STM32F103

spale2000, 2024-06-05:
P5(2727)-64x32-8s-v1.0

 Dimensions (in pixels) : 64x32
 Chips on rear side:  fm6124,74hc245c

[image]
 There are 2 pins G1 on the panel?

https://github.com/board707/DMD_STM32/assets/171742715/85240805-5308-4e21-bc8b-ff6937be400b

  - Library version: 1.06
  - Platform: STM32
  - MCU board: STM32F103 8t6
  - Connections: libary default
  Panel       Stm
  R1             A0
 G1             A1
 B1.            A2
 R2.            A3
 G1.            A4
 B2.            A5
A.               B6
B.                B5
C.                B4
Clk.            A15
Lat.             B7
Oe.              B0

spale2000, 2024-06-06:
Hello, i did everything as you said.

https://github.com/board707/DMD_STM32/assets/171742715/946e8402-8b2c-4d4e-9783-03b0542165e3

[image]

board707, 2024-06-06:
Thank you for the video.
But the result of the tests is not very encouraging.

In this test, a green dot should move across the matrix and the matrix should be gradually filled with one color.
But on your video, you don’t have any fill, and what’s more, the whole picture is shaking and blinking.

Check how securely all your connections are made. The pins on the STM32 board must be soldered. The connecting wires should not dangle. If this doesn't help, then there is probably something wrong with your panel or board.

spale2000, 2024-06-06:
the colors work on the panel, but always the same image.
[image]

spale2000, 2024-06-06:
[image]

spale2000, 2024-06-06:
[image]

spale2000, 2024-06-06:
[image]

board707, 2024-06-06:
Please show your full code. 
How did you upload the code to the STM32 - by serial bootloader or by St-link?

spale2000, 2024-06-06:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"
#define RGB64x32_S8_pt2 3,64,32,8,2
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8

// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PA7
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS> 
// <COLOR_1BITS> mode is not recomended for bitmap

//DMD_RGB <RGB64x32_S8_pt2, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB <RGB64x32_S8_pt2, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    // uncomment the line below if you using St-Link with STM32F103xx boards
    //enableDebugPorts();
    
    dmd.init();
    fg = dmd.Color888(255, 0,0);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j <

spale2000, 2024-06-06:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"
#define RGB64x32_S8_pt2 3,64,32,8,2
//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8

// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PA7
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS> 
// <COLOR_1BITS> mode is not recomended for bitmap

//DMD_RGB <RGB64x32_S8_pt2, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB <RGB64x32_S8_pt2, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    // uncomment the line below if you using St-Link with STM32F103xx boards
    //enableDebugPorts();
    
    dmd.init();
    fg = dmd.Color888(255, 0,0);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(3);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}

spale2000, 2024-06-06:
i uploaded the code with st-link

board707, 2024-06-06:
If you using a St-Link, you have to uncomment the line 
`enableDebugPorts();` in the `setup()`.

Also set the delay after drawPixel to slightly longer, for example `delay(20)`

board707, 2024-06-11:
Hi
Is the problem has been solved?

spale2000, 2024-06-12:
Thank you very much, the problem is solved. Sorry, i didn't see that i should have uncomment the line enableDebugPorts();

Uploading video-d11b47c6127700eada516e0f86a632b5-V.mp4…

closed 2024-06-14 by board707