board707/DMD_STM32 #114: Проблема с панелью 128*64 пикселя

SilentWalker9, 2024-06-11:
## Panel description 
#### Dimensions (in pixels) : 128*64
#### Scan factor: 1/32
#### Chips on rear side:  ICN2037BP, 74HC245KAJ, 74HC04D, RUC7258D

## Panel photos
[image]

## Specifications

  - Library version: 1.1.4
  - Platform: Raspberry 
  - MCU board: Pi Pico
Проблема в том, что нулевая координата экрана начинается с середины.

board707, 2024-06-11:
Добрый день
Пожалуйста выложите код, которым тестируете

SilentWalker9, 2024-06-11:
> Добрый день
> Пожалуйста выложите код, которым тестируете
```
/*--------------------------------------------------------------------------------------
  Demo for RGB panels

  DMD_STM32a example code for STM32 and RP2040 boards

  The example shows the use of a multi_color scrolling and fixed text.

  In multicolor mode each letter of the text prints in different color. 
  If string contains more chars than there are colors in the list - the list restarts from
  beginning. 
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"
#include "pic1.h"

// Fonts includes
#include "st_fonts/UkrRusArial14.h"
#include "st_fonts/UkrRusSystemFont5x7.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER true
// ==== DMD_RGB pins ====
// mux pins - A, B, C... mux pins must be consecutive in ascending order
#define DMD_PIN_A 7
#define DMD_PIN_B 8
#define DMD_PIN_C 9
#define DMD_PIN_D 10
#define DMD_PIN_E 11
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE 4
#define DMD_PIN_SCLK 5

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.2
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be consecutive in ascending order
uint8_t custom_rgbpins[] = { 6, 12, 13, 14, 15, 16, 17 };  // CLK, R0, G0, B0, R1, G1, B1
// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB<RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>(STM32 only), <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_F(Arial_14);

void setup(void) {
  // initialize DMD objects
  dmd.init();
  // Disable scrolling method by moving entire screen
  dmd.disableFastTextShift(true);
  for (int x = 0; x < 100; x++) {
    dmd.fillScreen(dmd.Color888(255, 255, 255));
[... 100 lines cut ...]
          dmd.swapBuffers(true);
          break;
      }
    }
  }
  
}

void DrawRect(int x1, int y1, int x2, int y2, uint16_t color, uint8_t filled) {
  if(filled != 0)
  {
    /*uint32_t i;

      for ( i = y1; i <= y2; i ++ )
        dmd.drawFastHLine ( x1, i, x2, i, color );*/
  }
  else {
    dmd.drawFastHLine(x1, y1, x2-x1, color);
    dmd.drawFastVLine(x2, y1, y2-y1, color);
    dmd.drawFastHLine(x1, y2, x2-x1, color);
    dmd.drawFastVLine(x1, y1, y2-y1, color);
  }
}

```

board707, 2024-06-11:
На первый взгляд никаких проблем не вижу. У Вас одна такая панель или несколько? Если много - пробовали другие панели, на всех такой эффект?

SilentWalker9, 2024-06-11:
> На первый взгляд никаких проблем не вижу. У Вас одна такая панель или несколько? Если много - пробовали другие панели, на всех такой эффект?

К сожалению только одна. Никаких графических дефектов за исключением этого.

SilentWalker9, 2024-06-11:
> Ну раз одна, то проще будет поменять местами пины нижней и верхней половины. Только надо делать это не в коде, а непосредственно на панели:
> 
> Пины 12, 13, 14 - подключите к R1, G1, B1
> а 15, 16, 17  - к R0, G0, B0
> 
> Настройки в коде НЕ МЕНЯЙТЕ.
> 
> 
> Отпишитесь потом, помогло или нет.

Другое дело. [image]

Возможно мой косяк, не правильно понял как подключать, возник небольшой косяк на краю, но нужно отмыть сначала от флюса.

board707, 2024-06-11:
На обратной стороне матрицы обычно есть стрелочки, показывающие вверх. Интересно - они совпадают с ориентацией вашего изображения?

А так вроде похоже проблема решена?

SilentWalker9, 2024-06-11:
> На обратной стороне матрицы обычно есть стрелочки, показывающие вверх. Интересно - они совпадают с ориентацией вашего изображения?
> 
> А так вроде похоже проблема решена?

Да, стрелочки совпадают [image]

[image]

https://mega.nz/file/DGwASRaI#6hQD6Df1A8srJMmFC99rQvJPfFtRC5J6U7AfHUxJotE
По ссылке видео. Красные пиксели - большое напряжение, а вот что за артефакты.

board707, 2024-06-11:
> Красные пиксели - большое напряжение, а вот что за артефакты.

Как вы определили про напряжение?

Под артефактами вы имеете в виду иногда мигающую строчку по центру панели?

SilentWalker9, 2024-06-11:
> > Красные пиксели - большое напряжение, а вот что за артефакты.
> 
> Как вы определили про напряжение?
> 
> Под артефактами вы имеете в виду иногда мигающую строчку по центру панели?
> 

Не только, если присмотреться, то и под буквами промелькивает. А по напряжению, потому что от лабораторника запитан, провода на 10AWG поменял.
https://mega.nz/file/2bp33BoY#pqK7TMMz9QTjeioJGA3Xn9qqDL67E0slqtaMJYD7G9M

board707, 2024-06-11:
Попробуйте поиграть интервалом прокрутки текста - 8мс для такой матрицы мало. Изменится что-то или нет?

И также снизьте яркость от 255 до 200 -  вы разницу скорее всего не заметите, а с артефактами может и помочь.

SilentWalker9, 2024-06-11:
> Попробуйте поиграть интервалом прокрутки текста - 8мс для такой матрицы мало. Изменится что-то или нет?
> 
> И также снизьте яркость от 255 до 200 -  вы разницу скорее всего не заметите, а с артефактами может и помочь.
> 
> 

Да так уже и сделал, но странно, ранее она у меня тоже была на максимальной яркости, а артефактов не было. А на счёт скорости, да, великовата, плавает. Но это не основная задача этого экрана, поэтому сойдёт.

SilentWalker9, 2024-06-11:
> > ранее она у меня тоже была на максимальной яркости, а артефактов не было.
> 
> Эти матрицы работают на высоких частотах (до 25 МГц) - поэтому они крайне чувствительны к качеству соединений и длине проводов. Если артефакты то появляются. то изчеазают - это практически 100% плохие контакты.

Тогда не было бы зависимости от яркости. Только от содержимого. Разводка платы экрана, возможно, немного кривая. Кстати, как я понял, запись данных в экран идёт вне зависимости от обращения к библиотеке?

board707, 2024-06-11:
> Тогда не было бы зависимости от яркости. Только от содержимого. 

Это разные факторы. Контакты - аппаратная проблема, зависимость от яркости - программная. Я не могу сказать издалека, в чем у вас дело - одно, другое, а может и то и то разом, поэтому даю рецепты наугад.

> как я понял, запись данных в экран идёт вне зависимости от обращения к библиотеке?

Принцип работы ХАБ75 панелей требует обновления данных тысячи раз в секунду

SilentWalker9, 2024-06-11:
> Это разные факторы. Контакты - аппаратная проблема, зависимость от яркости - программная. Я не могу сказать издалека, в чем у вас дело - одно, другое, а может и то и то разом, поэтому даю рецепты [...]

Ну я понял, что он организован на сдвиговых регистрах. Теперь такой вопрос, тактовая экрана зависит от частоты проца?

SilentWalker9, 2024-06-11:
Кажется я нашёл проблему.[image]

Это моя обычная настройка фиолетовых плат. И на такой настройке артефакты имеются.
 [image]

Это стандартная настройка частоты, на ней нет артефактов. Не зависит от частоты процессора, но зависит от выставленного делителя на QSPI, что очень странно.

SilentWalker9, 2024-06-11:
Тактовая частота, что интересно, не меняется при смене делителя скорости флеш.

board707, 2024-06-11:
> зависит от выставленного делителя на QSPI, что очень странно.

Насколько я помню, на QSPI висит внутренний флеш. Возможно что память дает артефакты при оверклоке?

> такой вопрос, тактовая экрана зависит от частоты проца?

Прямой зависимости нет.

SilentWalker9, 2024-06-11:
> Насчет "зависиморсти нет" - был неправ, зависит.
> Делитель частоты PIO машины задан явно - и выбран из расчета дефолтной частоты проца. При оверклокинге Пико в два раза клок на матрице тоже будет в два раз быстрее, при таких значениях некоторые панели уже начинают сбоить. 

Тут проблема с доступом к флешу возникает, по всей видимости.
[image]

Тактовая QSPI в стоке.

[image]

В разгоне.

board707, 2024-06-11:
Для проверки попробуйте поменять делитель `pio_clkdiv ` со значения 3 на 5 или 6 и запустить с оверклоком. Если дело в этом - артефакты должны уйти.

`pio_clkdiv ` задается в файле `DMD_STM32a.h` line 472

SilentWalker9, 2024-06-11:
> Для проверки попробуйте поменять делитель `pio_clkdiv ` со значения 3 на 5 или 6 и запустить с оверклоком. Если дело в этом - артефакты должны уйти.
> 
> `pio_clkdiv ` задается в файле `DMD_STM32a.h` line 472

Нет, не влияет. Я это и пытался донести, что проблема не в тактировании экрана, а в районе кода. Попробую отключить двойной буфер.

board707, 2024-06-11:
> это не просто артефакты, а кратковременное смещение пикселей вниз, на один. 

на самом деле это один из самых типичных артефактов на подобных библиотеках (моей в том числе). Он связан с расинхроном вывода данных и переключения строк.
Если грубо, матрица работает по типу динамической индикации. В каждый конкретный момент на панели светится только одна горизонтальная строка. Матрица имеет на борту память только на одну строчку. Чтобы засветить следующую строчку, нужно сначала загрузить данные этой строки, а потом переключить вывод. Если переключить строчку чуть раньше или чуть позже, данные засветят соседнюю строчку.
В общем, это проблема скорее всего где-то в моем коде, но найти проблему без тестов на реальной матрице непросто.

SilentWalker9, 2024-06-11:
> на самом деле это один из самых типичных артефактов на подобных библиотеках (моей в том числе). Он связан с расинхроном вывода данных и переключения строк.
> Если грубо, матрица работает по типу [...]

Перетекание оптимизации на fast почти избавило от артефактов. Я готов предоставить помощь удаленно, единственное в свободное время от работы.

board707, 2024-06-11:
Спасибо. У меня сейчас нет ни одной стандартной матрицы такого размера, только всякая экзотика :) Так что мне может понадобится Ваша помощь для проверки на реальной панели, влияют ли мои исправления на артефакты или нет. 
И хорошо что Вы подняли эту тему, я всегда рад баг-репортам :)

SilentWalker9, 2024-06-11:
> Спасибо. У меня сейчас нет ни одной стандартной матрицы такого размера, только всякая экзотика :) Так что мне может понадобится Ваша помощь для проверки на реальной панели, влияют ли мои исправления на артефакты или нет. 
> И хорошо что Вы подняли эту тему, я всегда рад баг-репортам :)

Не за что, тестером быть забавно, можно находить то, что не должно выплывать)

board707, 2024-06-27:
Привет, возвращаясь к проблеме - если честно, каких-то явных причин для артефактов я не нашел. Кое-какие шероховатости поправил в коде, хотя не уверен, что они как-то повлияют. 
Можете в последней версии библиотеки заменить файлы из приаттаченного архива и снова запустить тест?
[dmd_rgb_dma_clk_test.zip](https://github.com/user-attachments/files/16012322/dmd_rgb_dma_clk_test.zip)

SilentWalker9, 2024-06-28:
https://mega.nz/file/nH4yQITI#fNdRTfgLmg97ElaNyPObOrLwlHwppnwZTxSoAotQaOc Это после замены файлов вашими. Артефакты ушли, но я не уверен, ибо мог и я до этого что-нибудь накрутить.
https://mega.nz/file/jfxDwBTZ#2AGppFo-NVzSdTooWo6t2kKB5ab5P0xtv9eDuyuf5TI А это при попытке включения SPI, но отсутствия общения по нему, ибо модуль NRF24L01 принимает данные, но они по какой-то причине искажённые до неузнаваемости.

board707, 2024-06-29:
> мог и я до этого что-нибудь накрутить

Можете мне прислать код библиотеки, который у Вас получился после Ваших и моих изменений?

> А это при попытке включения SPI, 

Честно говоря, не вижу разницы с предыдущим видео...

Что касается проблем с SPI - можете мне выложить полный код скетча, на котором это проявляется?

SilentWalker9, 2024-07-04:
> Можете мне прислать код библиотеки, который у Вас получился после Ваших и моих изменений?
>
>
> Честно говоря, не вижу разницы с предыдущим видео...
>
> Что касается проблем с SPI - можете мне выложить [...]

Я только что выяснил, что проблема не со стороны вашей либы.

closed 2024-08-23 by board707