board707/DMD_STM32 #116: Support for chip ICN2037

jan019, 2024-06-29:
I have bought the following panel https://vi.aliexpress.com/item/32735128260.html?spm=a2g0o.order_list.order_list_main.21.42f41802FZTVuE&gatewayAdapt=glo2vnm and the used chip on the one I received is ICN2037.

I tried using all available DMD classes for different chips, but non of them unfortunately work. Is there a plan or to provide support for the chip ICN2037?

Thanks

board707, 2024-06-29:
What is your problem with that panel? Do you see any leds lit on the panel while your tests?
Could you please provide a clear photo of panel's rear side and types of all chips (not the only main driver) on the matrix?

jan019, 2024-06-30:
Hi, I am attaching photos of all the differnt chips that are found on the panel. I am attaching a video of what can be seen when I run the following code:
```
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b,  13);

#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

#define ENABLE_DUAL_BUFFER false

#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = {DMD_PIN_A, DMD_PIN_B, DMD_PIN_C, DMD_PIN_D, DMD_PIN_E};

#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

uint8_t custom_rgbpins[] = {PA15, PA0, PA1, PA2, PA3, PA4, PA5}; // CLK, R0, G0, B0, R1, G1, B1
DMD_RGB <RGB64x32_S8_OKSingra, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0; // background - black
uint16_t fg = 0; // foreground

void setup()
{
    dmd.init();

  dmd.setBrightness(100);

  uint16_t bg = 0;                      // background - black
  uint16_t fg = dmd.Color888(255, 0, 0); // foreground - red

  char str[] = "H";

  dmd.selectFont(&GlametrixL);
  dmd.drawString(0, 0, str, strlen(str),fg);
}

void loop()
{
}
```

In the video, I am pressing the reset button on the bluepill board, and each time I press the reset button the LEDs shift by one place. 

Video: https://youtu.be/JqYKb7uCnM8

[image]
[image]

[image]

Thanks for you response.

board707, 2024-06-30:
Thank you for the chips photos, but I also need a general view of rear side of panel. What is the panel size in pixels?

What is your bluepill board - STM32F103C8T6 ? How do you upload a code to it - via Serial bootloader or via St-Link?

jan019, 2024-06-30:
This is how the panel looks from the back. It's size is 32x16, 1/4 scan. 
[image]
[image]

Yes, my bluebell board is STM32F103C8T6 and I upload the code using stlink.

Thanks!

board707, 2024-06-30:
If your panel is 32x16, why did you used a 64x32 pattern in your test code?
`DMD_RGB <RGB64x32_S8_OKSingra, COLOR_4BITS_Packed>`

Please upload the code below and show the video how it works:
```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
 #include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux

// Color depth - <COLOR_4BITS_Packed> or <COLOR_4BITS> 
// <COLOR_1BITS> mode is not recomended for bitmap

DMD_RGB <RGB32x16plainS4_pt2, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
    // uncomment the line below if you using St-Link with STM32F103xx boards
    enableDebugPorts();
    
    dmd.init();
    fg = dmd.Color888(0, 0, 255);
    dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}
```

jan019, 2024-06-30:
Sorry, I posted the last test code where I changed even the board size to check how it behaves. I however have tested the original example code that you have posted and it behaved it a very similar way as with this code - something was shown, and with each reset press, it shifted.  Tomorrow I will post a video showing the exact behaviour. Thanks.

jan019, 2024-07-01:
Hi, here is the video of the LED panel behaviour with the code you provided:
https://youtube.com/shorts/mUpwkmHzwtg?feature=share

board707, 2024-07-01:
Thank you, but your video shows nothing similar to that I expected. I haven't much to say about this. The ICDN2037 is a known driver and have tested with the library before.

According to the panel behavior, I would suggest a problems with contacts or power. Take the note, that you need at least  2.5A external power supply to run this code. And the maximum power consuming might be a 8A at white full brightness.

jan019, 2024-07-01:
The pwoer connections and power source are not definitly not the problem. I will rewire the control pins one more time, to be sure that the problem is not there. 
It's good news that the ICDN2037 is tested with the library. I'll let you know if the problem was with the wiring or not. Thanks

jan019, 2024-07-01:
I checked the wiring and it is correct, 100%. Using an osciloscope I am looking at OE and CLK lines. On the CLK line there are no pulses, on the DMD_PIN_SCLK line there are very short (20 ns) pulses every 150 us and on the OE pin there are also no pulses when the code that you sent is run. 

This is weird, is it not?
Thanks

board707, 2024-07-01:
> This is weird, is it not?

it definitely doesn't have to be that way.
What version of the library are you using? Please, show me a photo of your bluepill board, what is written on the chip?
Make sure you don't forget to uncomment the line at the beginning of the setup:
```
enableDebugPorts();
```

jan019, 2024-07-01:
I changed the pin OE from PB0 to PB1 and the LAT (SCLK) from PA15 to PA10 and now both signals look good. 
The pattern with 'RGB32x16plainS4' now looks like this https://youtu.be/kWqGHTmo8F8?feature=shared

It's definitly looking good! Is the patern as expected?

I had the code for emableDebugPorts() commented all the time, so this was not causing issues.

jan019, 2024-07-01:
Okay that was clearly not correct. I changed the pattern to `RGB32x16_S4` and now it fills the board from upper left corner to bottom right corner. It's now working great! Thanks for the help! Also, it might be good to add a note to the readme, that if the panel doesn't work, the user should try differnet pins.
[image]

board707, 2024-07-01:
First of all - my congratulations :)

However, I should point out that you have mixed up some things.
The line `emableDebugPorts() ` **should be uncommented (active)** if you use ST-Link. Without this line in the code, some pins on the controller do not work, in particular pin PA15 - which is what you encountered. If you had included this line in your code from the start, you wouldn't have had to change the pins.
Also, note that on pin PA15 there was not a LAT signal, but a CLK signal.

closed 2024-07-02 by board707