board707/DMD_STM32 #118: No template for P5 RGB HUB75 64x32 1/8 Scan Module

galagen8, 2024-08-06:
**Panel description:**
LED P5 OUTDOOR RGB SMD2121 HUB75 64x32 Module

**Dimensions (in pixels):**
64x32

**Scan factor:**
1/8

**Chips on rear side:**
ICN2012, ICN2037BP, MW245B
[image]
[image]
[image]

**Panel photos:**
[image]
[image]

**Specifications:**
Library version - 1.1.4
Platform - STM32
MCU board - Bluepill_STM32F103C8T6

**Connections:**
PA0 === R0	
PA2 === B0	
PA3 === R1	
PA5 === B1	
PB6 === A	
PB4 === C	
PA15 = CLK	
PB0 == OE	

G0 === PA1
GND = GND
G1 === PA4
E ==== PB8
B ==== PB5
D ==== PB3
LAT === PB7
GND = GND

[STM32_HUB75_CONN_1.pdf](https://github.com/user-attachments/files/16505391/STM32_HUB75_CONN_1.pdf)

I've connected according to description, tried demo examples but no any pixels are lighted up. Maybe it's because scan rate is unusual (1/8) for this resolution (64x32)? Also I didn't find any suited template for that panel.

board707, 2024-08-06:
Please show your test code

galagen8, 2024-08-07:
I downloaded this example code (dmd_rgb.ino). Honestly I tried all library examples 

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32plainS16, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font UkrRusArial_F(UkrRusArial_14);
// GFX font with sepatate parts for Latin and Cyrillic chars
DMD_GFX_Font GlametrixL((uint8_t*)&GlametrixLight12pt7b, (uint8_t*)&GlametrixLight12pt8b_rus, 0x80, 13);

/*--------------------------------------------------------------------------------------
  UTF8 char recoding

--------------------------------------------------------------------------------------*/
int utf8_rus(char* dest, const unsigned char* src) {

[... 145 lines cut ...]

                    }
                    else {
                        if (step[i] < 0) dmd.drawMarqueeX(m, 0, dmd.height());
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

                // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

           

        }
    }
}

board707, 2024-08-07:
> [Отслеживающее изображение][https://github.com/notifications/beacon/AL64JWWFPTNNTFOVLIKP2X3ZQGGUJA5CNFSM6AAAAABMBQTJLCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUHOQUXM.gif]

Thank you.
If you use a st-link for downloading the code, try to uncomment this line in the setup:

enableDebugPorts()

galagen8, 2024-08-07:
Thank you!
I really use ST-Link (Chinese ver.) and I've just uncommented this line in void setup(void):  enableDebugPorts() and it works!!!
I didn't notice this line before you've pointed me!

https://github.com/user-attachments/assets/fd42b750-63fc-47f9-acbc-d131a257c389

Here is how it works, text is doubled or even tripled, is it because of incorrect Scan Rate?

board707, 2024-08-07:
Please add the line below to the beginning of your sketch

#define RGB64x32plainS8  4,64,32,8,1               // 64x32 1/8

And try to use this template in the DMD class definition

DMD_RGB <RGB64x32plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

galagen8, 2024-08-07:
At first I added those 2 lines like you wrote me: 

```
#define RGB64x32plainS8 4,64,32,8,1 // 64x32 1/8
DMD_RGB <RGB64x32plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```
But fonts were doubled again. So I started experimenting with some numbers of #define line. And after I changed the last number in this #define line from 1 to 3 it just started to work normally:

`#define RGB64x32plainS8 4,64,32,8,3 // 64x32 1/8`

Voila! Here is the video:

https://github.com/user-attachments/assets/05f29341-83f2-4593-a7e3-9aba2c900d38

And here is the full code:

```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// add new template panel
#define RGB64x32plainS8 4,64,32,8,3 // 64x32 1/8 ____ 4,64,32,8,1 - old numbers

 // Fonts includes
#include "st_fonts/UkrRusArial14.h"
#pragma GCC diagnostic ignored "-Wnarrowing"
#pragma GCC diagnostic ignored "-Woverflow"
#include "gfx_fonts/GlametrixLight12pt7b.h"
#include "gfx_fonts/GlametrixBold12pt7b.h"
#pragma GCC diagnostic warning "-Wnarrowing"
#pragma GCC diagnostic warning "-Woverflow" 

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
[... 167 lines cut ...]
                        else dmd.drawMarqueeX(m, 0, 0);
                    }
                }
                // output mem buffer to matrix
                dmd.swapBuffers(true);
                break;

                // vertical scrolling    
            case 3:

                dmd.stepMarquee(-1, 0, 1);
                dmd.swapBuffers(true);
                break;

            }

           

        }
    }
}
```

I didn't tried to chain those panels together  but I will.
Big thanks to you!

galagen8, 2024-08-15:
I've tried to generate some random numbers (numbers are changed every 1 sec) on that P5 screen and I noticed some extra dots. After restarting the MCU first set of numbers always looks clear, but next random set will have some extra dots. Those ghost pixels mostly appears at 17-18 horizontal row of the panel. I've tried many fonts, took the same P5 panel, but seems the problem is not about that. 
[image]
Do you have any ideas how to find the issue?
[image]
[image]
[image]

[image]

galagen8, 2024-08-15:
```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// add new template panel
#define RGB64x32_GAL_S8 4,64,32,8,3 // 64x32 1/8 ____ 4,64,32,8,1

// Fonts includes
#include "st_fonts/Arial_black_16.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32_GAL_S8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// other options are:
// <RGB32x16plainS8> -  32x16 matrix with 8scans
// <RGB80x40plainS20> - 80x40 matrix with 20scans
// <RGB64x64plainS32> - 64x64 matrix with 32scans
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial16_F(Arial_Black_16);

// arduino timer variables
unsigned long previousMillis = 0;  // will store last time when arduino timer was updated
const long interval = 1000;  // arduino timer interval i ms

char colon = 58;
char slash = 47;

int digit1; // for hours
int digit2; // for hours
int digit3; // for minutes
int digit4; // for minutes

//  coordinate vars
int xspace = 5;
int yspace = -1;
int dot = 46;

/*
// variables
int h1;
int h2;
int m1;
int m2;
int dot = 46;

int xspace;
int yspace;

h1 = 49; //  1
h2 = 54; //  6
dot = 46; //  dot
m1 = 51; //  3
m2 = 53; //  5

//  coordinate vars
xspace = 5;
yspace = -1;
/*

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{
  // uncomment line below if you using St-Link with STM32F103xx boards
  enableDebugPorts();
  
  // initialize DMD objects
	dmd.init(); 
  
  // set the brightness
  dmd.setBrightness(1); 

  // create the colors
  uint16_t bg = 0;                      // background - black
  uint16_t fg = dmd.Color888(255, 0, 255); // foreground - purple 
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{
  //  arduino timer starts here:
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) 
  {
    // save the last time did smth
    previousMillis = currentMillis;

    //  generate random digits from 1 to 9
    digit1 = (rand() % 10);
    digit2 = (rand() % 10);
    digit3 = (rand() % 10);
    digit4 = (rand() % 10);

    // do smth
    char chdigit1 = digit1 + '0'; //  for hour char
    char chdigit2 = digit2 + '0'; //  for hour char
    char chdigit3 = digit3 + '0'; //  for min char
    char chdigit4 = digit4 + '0'; //  for min char
    

    dmd.selectFont(&Arial16_F);
    //  hours 1st digit
    dmd.drawChar(3 + xspace, 9 + yspace, chdigit1, 0xF81F, 0);  //  color magenta
    //  hours 2nd digit
    dmd.drawChar(14 + xspace, 9  + yspace, chdigit2, 0xF81F, 0);
    //  dots
    dmd.drawChar(25 + xspace, 9  + yspace, dot, 0xF81F, 0);
    dmd.drawChar(25 + xspace, 1  + yspace, dot, 0xF81F, 0);
    //  minutes 1nd digit
    dmd.drawChar(31 + xspace, 9  + yspace, chdigit3, 0xF81F, 0);
    //  minutes 1nd digit
    dmd.drawChar(41 + xspace, 9  + yspace, chdigit4, 0xF81F, 0);
    
  } // arduino timer ends here
  
}
```

board707, 2024-08-15:
Hi
The most fonts are transparent, it means that if the new digit doesn't fully cover the previous, you will see artefacts.
You have to clean the place before to draw the new digit.

galagen8, 2024-08-26:
And the code:
```

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// add new template panel
#define RGB64x32plainS8 4,64,32,8,3 // 64x32 1/8 ____ 4,64,32,8,1

// Fonts includes
#include "st_fonts/Arial_black_16.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 3
#define DISPLAYS_DOWN 3

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

// --- Define fonts ----
// DMD.h old style font
DMD_Standard_Font Arial_Black_16_F(Arial_Black_16);

// coordinate and color variables
int xspace = 20;
int yspace = 0;
long color = 0xF81F; 

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{
  // uncomment line below if you using St-Link with STM32F103xx boards
  enableDebugPorts();
  
  // initialize DMD objects
	dmd.init();

  // set connection scheme
  dmd.setConnectScheme(CONNECT_ZIGZAG);
  
  // set the brightness
  dmd.setBrightness(250);

  // clear screen
  dmd.clearScreen(true);

  // create the colors
  uint16_t bg = 0;                      // background - black
  uint16_t fg = dmd.Color888(255, 0, 255); // foreground - purple
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{      
  // select the font
  dmd.selectFont(&Arial_Black_16_F);
    
  // digit 1 row
  dmd.drawChar(0, yspace + 9, 48, color, 0);  //  0
  dmd.drawChar(xspace, yspace + 9, 49, color, 0);  //  1
  dmd.drawChar(xspace * 2, yspace + 9, 50, color, 0);  //  2
  dmd.drawChar(xspace * 3, yspace + 9, 51, color, 0);  //  3
  dmd.drawChar(xspace * 4, yspace + 9, 52, color, 0);  //  4
  dmd.drawChar(xspace * 5, yspace + 9, 53, color, 0);  //  5
  dmd.drawChar(xspace * 6, yspace + 9, 54, color, 0);  //  6
  //dmd.drawChar(xspace * 7, yspace + 9, 55, color, 0);  //  7
  dmd.drawChar(xspace * 7, yspace + 9, 56, color, 0);  //  8
  dmd.drawChar(xspace * 8, yspace + 9, 57, color, 0);  //  9

  // digit 2 row
  dmd.drawChar(xspace * 9, yspace + 9, 48, color, 0);  //  0
  dmd.drawChar(xspace * 10, yspace + 9, 49, color, 0);  //  1
  dmd.drawChar(xspace * 11, yspace + 9, 50, color, 0);  //  2
  dmd.drawChar(xspace * 12, yspace + 9, 51, color, 0);  //  3
  dmd.drawChar(xspace * 13, yspace + 9, 52, color, 0);  //  4
  dmd.drawChar(xspace * 14, yspace + 9, 53, color, 0);  //  5
  dmd.drawChar(xspace * 15, yspace + 9, 54, color, 0);  //  6
  //dmd.drawChar(xspace * 7, yspace + 9, 55, color, 0);  //  7
  dmd.drawChar(xspace * 16, yspace + 9, 56, color, 0);  //  8
  dmd.drawChar(xspace * 17, yspace + 9, 57, color, 0);  //  9

}
```

board707, 2024-08-26:
Hi
You screen build with 3x3 64x32 panels, so it pixel dimensions is 192x96.
But in your code you tried to put the char to the point 340,9:
```
int xspace = 20;
int yspace = 0;
dmd.drawChar(xspace * 17, yspace + 9, 57, color, 0);
```

galagen8, 2024-08-28:
Yes, thank you to show me my mistake, now I see.
But anyway if I try to add more than 1 row - screen is empty.
Connection scheme:
[image]
Here I just would like to draw just a small filled rectangle, but screen is empty again.
```
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// add new template panel
#define RGB64x32plainS8 4,64,32,8,3 // 64x32 1/8 ____ 4,64,32,8,1

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 3
#define DISPLAYS_DOWN 3

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8

// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32plainS8, COLOR_4BITS_Packed> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

// color variable
long color = 0xFFFF;

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{
  // uncomment line below if you using St-Link with STM32F103xx boards
  enableDebugPorts();
  
  // initialize DMD objects
  dmd.init();

  // set connection scheme
  dmd.setConnectScheme(CONNECT_NORMAL);
  
  // set the brightness
  dmd.setBrightness(250);

  // clear screen
  dmd.clearScreen(true);
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{
  //dmd.drawPixel(0, 0, color);
  dmd.drawFilledBox(0, 0, 4, 4, color);
}
```
What coordinate system is there? Like on this pic?
[image]

board707, 2024-08-28:
Hi
Do you really have a 9 panels or use this settings for test purpose only?

Try to output the rectangle to various different panels - 1, 3, 5, , 7 or 8...

galagen8, 2024-08-28:
Yes, I truly have 9 panels, and I would like to build the screen 3x3 same as the diagram in my post above.
It's not only for test purpose, I really want to make a information screen or the big clock screen.

Do you mean I have to test each panel separate with MCU? To be sure all of them are ok?

board707, 2024-08-28:
> Do you mean I have to test each panel separate with MCU?

No, I suggest to output the rectangle to different coordinates - so that it falls on different panels.

galagen8, 2024-08-28:
I use Bluepill_STM32F103C8T6.
I tried to use different coordinates according to every panel. I set here:
```
dmd.drawPixel(160, 80, color); // supposed to be panel 1 (bottom row)
dmd.drawPixel(95, 80, color);   // supposed to be panel 2 (bottom row)
dmd.drawPixel(30, 80, color);  // supposed to be panel 3 (bottom row)
dmd.drawPixel(160, 45, color); // supposed to be panel 4 (middle row)
dmd.drawPixel(95, 45, color);   // supposed to be panel 5 (middle row)
dmd.drawPixel(30, 45, color);  // supposed to be panel 6 (middle row)
dmd.drawPixel(160, 15, color); // supposed to be panel 7 (top row)
dmd.drawPixel(95, 15, color);   // supposed to be panel 8 (top row)
dmd.drawPixel(30, 15, color);  // supposed to be panel 9 (top row)
```
And nothing on screen. No any flickering ever.

Actually when I change the string `#define DISPLAYS_DOWN 1` to digit 2 or 3 all panels start to show me nothing at all.

Same thing happens if I add more than 5 panels to X-axis with only 1 row on Y-axis. For example:
```
#define DISPLAYS_ACROSS 6
#define DISPLAYS_DOWN 1
```
- with this code the screen shows me nothing at all too, it is just black. But in theory it should show... I can't figure it out. 

Maybe there is something in the library? Is this library able to work with more than 5 panels?

board707, 2024-08-28:
> I use Bluepill_STM32F103C8T6.

The board has a 20Kbytes RAM, while a 9 panels of 64x32 with 4bit color depth needs a 64x32x9x(3/2 bytes per pixel) = about a 27 Kbytes buffer - it is no surprise nothing works.

> Same thing happens if I add more than 5 panels

This is again expected. Five panels require 15K of buffer, 5K is left for the library code itself and it works. 6 panels already take up 18K of memory, the remaining 2K is not enough for the code.

For your screen you need to use boards with large memory capacity supported by the library - for example blackpill STM32F401/F411 or Raspberry Pico RP2040.

board707, 2024-08-28:
Alternatively, if you are creating an information board where you do not need many colors, you can use the library in single-bit color mode. In this case, only basic colors will be available to you - but the memory requirements will be reduced three times and you will be able to use a bluepill board.

```
DMD_RGB <RGB64x32plainS8, COLOR_1BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

board707, 2024-08-29:
> What functions I can use with 1bit color mode? Thank you

The library functions the same, it not depend on color mode. But the supported connection pins are different for COLOR_4BIT_Packed and other modes. I suspect that's the cause why you have problems displaying numbers.

As far as I understand, you've already made the PCB and can't change the pins? - I'll try to rebuild the library so you can use your connection in 1bit mode, but it will take me a few days.

In general, it is a good habit to test your design fully before the making a PCB....

galagen8, 2024-08-29:
Ohh.. I thought 1-bit mode uses the same connection scheme...

Thank you, but I think no need to rebuild the library if it takes several days, it's not worth it.
I suppose better way is to redesign the PCB — I've made 2 minor errors in PCB design. These errors do not affect the board, but in any case it is better to correct these errors.

About test your design fully before the making a PCB - it's just the main RULE but as you see I didn't follow it this time :D

And I already ordered Black Pill STM32F411CEU6. Is it one of the best choices for those Px Panels?

Just in case, can you tell me what pins are used for this 1-bit mode? If it saves microcontroller memory, one day I may connect more panels together

board707, 2024-08-29:
> I thought 1-bit mode uses the same connection scheme...

The problem is not in 1bit mode, but in 4bit_packed mode, which you used before and for which your PCB is designed. This is a specific mode for bluepill boards and it has different pinout options than for 1bit and standard 4bit.

As for reworking the library, I did not mean that I would need to work for several days :) I wanted to run some tests, but I am currently short on time and I am not sure that I will do it today or tomorrow.

board707, 2024-08-29:
Hi
 I slightly changed the library to able using your actual connection with 1bit color mode.
Please replace two files from the attached archive and test your code again.
[1bit_color.zip](https://github.com/user-attachments/files/16806279/1bit_color.zip)

galagen8, 2024-08-30:
Now it works! Thank you! You helped me a lot! What can I do for you?

[image]

And the question: when I tried to test whole screen just in case using the function `dmd.fillScreen()` - filling with Red Color was ok. Other colors like Cyan, White etc. looks different, especially upper rows:

[image]
`dmd.fillScreen(0x07FF)    // CYAN`

[image]
`dmd.fillScreen(0x07FF)    // WHITE`

Is it because STM32 BluePill run out of memory (all pixels are used on the screen)?

galagen8, 2024-08-30:
[image]

```
// tested 3x3 (XY) screen color rectangles OK 

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

// add new template panel
#define RGB64x32plainS8 4,64,32,8,3 // 64x32 1/8

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 3
#define DISPLAYS_DOWN 3

// Enable of output buffering
// if true, changes only outputs to matrix after
// swapBuffers(true) command
// If dual buffer not enabled, all output draw at matrix directly
// and swapBuffers(true) cimmand do nothing
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8

// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA15, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>
// We use 64x32 matrix with 16 scans and 4bit color:
DMD_RGB <RGB64x32plainS8, COLOR_1BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
// Color depth - <COLOR_4BITS_Packed>, <COLOR_4BITS> or <COLOR_1BITS>

// color pallete
long color_white = 0xFFFF;  // white color
long color_yellow = 0xFFE0; // yellow color
long color_cyan = 0x07FF; // cyan color
long color_magenta = 0xF81F; // magenta color

/*--------------------------------------------------------------------------------------
  setup
  Called by the Arduino architecture before the main loop begins
--------------------------------------------------------------------------------------*/

void setup(void)
{
  // uncomment line below if you using St-Link with STM32F103xx boards
  enableDebugPorts();
  
  // initialize DMD objects
  dmd.init();

  // set connection scheme
  dmd.setConnectScheme(CONNECT_NORMAL);
  
  // set the brightness
  dmd.setBrightness(100);

  // clear screen
  dmd.clearScreen(true);
}

/*--------------------------------------------------------------------------------------
  loop
  Arduino architecture main loop
--------------------------------------------------------------------------------------*/

void loop(void)
{      
  dmd.fillRect(48, 24, 32, 16, color_white);  	//	betweeen 9-8 x 6-5 panels
  dmd.fillRect(112, 24, 32, 16, color_yellow);	//	betweeen 8-7 x 5-4 panels
  dmd.fillRect(48, 56, 32, 16, color_magenta);  //	betweeen 6-5 x 5-4 panels
  dmd.fillRect(112, 56, 32, 16, color_cyan);  	//	betweeen 5-4 x 2-1 panels
}
```

board707, 2024-08-30:
> Now it works! Thank you! You helped me a lot! What can I do for you?

Great. Thanks for your feedback :)
Star the library if you satisfied with it.

> Is it because STM32 BluePill run out of memory?

No, if the library does not have enough memory - nothing will work at all, as you have already seen before.
This looks like a bug or hardware issue. Perhaps your screen simply does not have enough power when all pixels are illuminated. Try reducing the brightness in the code

board707, 2024-09-09:
Hi 
Did you resolve the issue? I would appreciate your feedback.

galagen8, 2024-09-10:
Sorry, I didn't write you back because I was digging into this screen.
Yes, the Library works, everything is good but sometimes I see some green pixels on the screen in top above the big numbers. I've already tried many power suppliers (Mean Well 5V 60A; Mean Well 5V 30A; Mean Well 5V 8A), changed PCBs, changed data cables and I still can't catch what the issue is...

... Maybe you had the same problems and you can give me an advice according to your experience? Thank you!

This is how these green pixels looks like:

[image]

[image]

[image]

galagen8, 2024-10-08:
Hello again! I think I found the issue why those erratic pixels appears!

I've set the voltage for LED Power Supply to 4.4 Volts (Power Supply for MCU is separate) using this adjusting resistor:

[image]

It quite interesting, because all those LED panels supposed to use 5V. If I make the voltage more than 4.4V — more erratic pixels appears, the screen become much more noisy. This method could help someone who faced with the same problem.

Maybe it happened because I didn't use Level Shifter (3.3V to 5V)? If someone knows please let me know.

And I suppose the topic could be closed. Big thanks for your support!