board707/DMD_STM32 #121: Add New Panel Please

digi55, 2024-08-07:
Hi @board 707
I have new ,P4 80x40  Outdoor Panel
Can you add this panel for our library.
ICN 2037 chipset 

[image]

https://github.com/user-attachments/assets/bd817d5a-07de-4960-b70a-03a2703be5c7

/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
#define DMD_PIN_D PB3
#define DMD_PIN_E PB8
// put all mux pins at list
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux
// <RGB80x40plainS20> - 80x40 matrix with 20scans

//DMD_RGB <RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
DMD_RGB <RGB80x40plainS20, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
//DMD_RGB <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
  enableDebugPorts();
  dmd.init();
  fg = dmd.Color888(0, 0, 255);
  dmd.setBrightness(50);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(10);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}

digi55, 2024-08-07:
QR code: WG500926 SN:10815 GXY4SL 1921 2037 2018 10S

board707, 2024-08-07:
Hi
I am not in home now, I will return on next week.
Dmitry

board707, 2024-08-12:
Hi, I just returned, we could continue.

Please, what other chips are there on the panel besides the driver?

digi55, 2024-08-13:
Hi @board707 , I'm on vacation right now. I'll be back next week.
Since the panel is outdoor, the integrated circuits are covered with
varnish, it is very difficult to read. I will try to find a solution. Thank
you very much for your help.

12 Ağu 2024 Pzt 06:36 tarihinde board707 ***@***.***> şunu
yazdı:

board707, 2024-08-13:
Ok, will continue later

digi55, 2024-08-18:
Hi , @board707 I just returned.
I found some information about this panel.
LED chips : Sm16169
I hope it helps.
I will try to read it with a microscope too.

board707, 2024-08-18:
> LED chips : Sm16169

Are you sure?
Unfortunately, this driver is known as a new SM-PWM type, and I don't have a datasheet for it. It is not supported by the library yet.

digi55, 2024-08-19:
Hi, @board707
it's very hard to read because it's varnished.
as far as I can read
CHIPONE
ICN2037, ICND2010A chips are available.
I ordered a microscope. I hope to be able to give more precise information when it arrives.

board707, 2024-08-19:
Thanks
This is encouraging since ICDN2037 is a supported driver.

Could you change the definition of the DMD class to options below:
Variant A
```
#define RGB80x40s10_pt1 4,80,40,10,1
DMD_RGB <RGB80x40s10_pt1, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```
Variant B
```
#define RGB80x40s10SR_pt1 3,80,40,10,1
DMD_RGB_SHIFTREG_ABC <RGB80x40s10SR_pt1, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```
and run the test for each?

Also please increase delay between drawing pixels to 30ms

digi55, 2024-08-19:
Variant A
,https://github.com/user-attachments/assets/112d39e4-0521-43f6-9ed6-f86218027611

digi55, 2024-08-19:
Variant B
https://github.com/user-attachments/assets/10d3b990-6e20-495d-ab35-c1fa2001f4bf

board707, 2024-08-19:
Please confirm - variant B fills the entire panel evenly if you continue?

digi55, 2024-08-19:
I will test it tomorrow and let you know.

digi55, 2024-08-20:
> Please confirm - variant B fills the entire panel evenly if you continue?

Yes

board707, 2024-08-20:
Thank you, I'll work with it

digi55, 2024-08-20:
Thank you too

board707, 2024-08-20:
Please add this code after `#include "DMD_RGB.h"` line:
```
template <int MUX_CNT, int P_Width, int P_Height, int SCAN, int Pixbase, int Pattern, int COL_DEPTH>
class DMD_RGB<MUX_CNT, P_Width, P_Height, SCAN, 122, Pixbase, Pattern, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(MUX_CNT, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, SCAN, P_Width, P_Height)
  {
        this->fast_Hbyte = false;
        this->use_shift = false;
        }
    // Fast text shift is disabled for complex patterns, so we don't need the method
        void disableFastTextShift(bool shift) override {}

protected:
  
  uint16_t get_base_addr(int16_t& x, int16_t& y) override {
    this->transform_XY(x, y);
    uint8_t pol_y = y % this->pol_displ;
    x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
    uint16_t base_addr = (pol_y % this->nRows) * this->x_len +
            (x / Pixbase) * this->multiplex * Pixbase + x % Pixbase;
    if (((x % P_Width)/Pixbase)%2) { if (pol_y >=  this->nRows)  base_addr += Pixbase; }          // LPS
    else { if (pol_y < this->nRows)  base_addr += Pixbase; }    // UPS
     
    return base_addr;
  }
  
};
```
and use it in definition of the class:
```
#define RGB80x40s10SR_digi55 3,80,40,10,122,16,1
DMD_RGB_SHIFTREG_ABC <RGB80x40s10SR_digi55,  COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
```

digi55, 2024-08-20:
/*--------------------------------------------------------------------------------------
 Demo for RGB panels

 DMD_STM32a example code for STM32F103xxx board

 Pattern test for 32x16 1/2 1/4 RGB panels
 ------------------------------------------------------------------------------------- */
#include "DMD_RGB.h"

template <int MUX_CNT, int P_Width, int P_Height, int SCAN, int Pixbase, int Pattern, int COL_DEPTH>
class DMD_RGB<MUX_CNT, P_Width, P_Height, SCAN, 122, Pixbase, Pattern, COL_DEPTH> : public DMD_RGB_BASE2<COL_DEPTH>
{
public:
  DMD_RGB(uint8_t* mux_list, byte _pin_nOE, byte _pin_SCLK, uint8_t* pinlist,
    byte panelsWide, byte panelsHigh, bool d_buf = false) :
    DMD_RGB_BASE2<COL_DEPTH>(MUX_CNT, mux_list, _pin_nOE, _pin_SCLK, pinlist,
      panelsWide, panelsHigh, d_buf, COL_DEPTH, SCAN, P_Width, P_Height)
  {
        this->fast_Hbyte = false;
        this->use_shift = false;
        }
    // Fast text shift is disabled for complex patterns, so we don't need the method
        void disableFastTextShift(bool shift) override {}

protected:
  
  uint16_t get_base_addr(int16_t& x, int16_t& y) override {
    this->transform_XY(x, y);
    uint8_t pol_y = y % this->pol_displ;
    x += (y / this->DMD_PIXELS_DOWN) * this->WIDTH;
    uint16_t base_addr = (pol_y % this->nRows) * this->x_len +
            (x / Pixbase) * this->multiplex * Pixbase + x % Pixbase;
    if (((x % P_Width)/Pixbase)%2) { if (pol_y >=  this->nRows)  base_addr += Pixbase; }          // LPS
    else { if (pol_y < this->nRows)  base_addr += Pixbase; }    // UPS
     
    return base_addr;
  }
  
};

//Number of panels in x and y axis
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

// Output buffering - false
#define ENABLE_DUAL_BUFFER false

// ==== DMD_RGB pins ====
// mux pins - A, B, C... all mux pins must be selected from same port!
#define DMD_PIN_A PB6
#define DMD_PIN_B PB5
#define DMD_PIN_C PB4
//#define DMD_PIN_D PB3
//#define DMD_PIN_E PB8
// put all mux pins at list
//uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C , DMD_PIN_D , DMD_PIN_E };
uint8_t mux_list[] = { DMD_PIN_A , DMD_PIN_B , DMD_PIN_C};

// pin OE must be one of PB0 PB1 PA6 PA7
#define DMD_PIN_nOE PB0
#define DMD_PIN_SCLK PB7

// Pins for R0, G0, B0, R1, G1, B1 channels and for clock.
// By default the library uses RGB color order.
// If you need to change this - reorder the R0, G0, B0, R1, G1, B1 pins.
// All this pins also must be selected from same port!
uint8_t custom_rgbpins[] = { PA6, PA0,PA1,PA2,PA3,PA4,PA5 }; // CLK, R0, G0, B0, R1, G1, B1

// Fire up the DMD object as dmd<MATRIX_TYPE, COLOR_DEPTH>

// For pattern testing use one of matrix below:
// RGB32x16plainS4         - 32x16 1/4 Binary mux
// RGB32x16plainS4_DIRECT  - 32x16 1/4 DIRECT mux
// RGB32x16plainS2_DIRECT  - 32x16 1/2 Binary mux
// <RGB80x40plainS20> - 80x40 matrix with 20scans

//DMD_RGB <RGB64x32plainS16, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
//DMD_RGB <RGB80x40plainS20, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);
//DMD_RGB <RGB128x64plainS32, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

//#define RGB80x40s10_pt1 4,80,40,10,1
//DMD_RGB <RGB80x40s10_pt1, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

//#define RGB80x40s10SR_pt1 3,80,40,10,1
//DMD_RGB_SHIFTREG_ABC <RGB80x40s10SR_pt1, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

#define RGB80x40s10SR_digi55 3,80,40,10,122,16,1
DMD_RGB_SHIFTREG_ABC <RGB80x40s10SR_digi55, COLOR_4BITS> dmd(mux_list, DMD_PIN_nOE, DMD_PIN_SCLK, custom_rgbpins, DISPLAYS_ACROSS, DISPLAYS_DOWN, ENABLE_DUAL_BUFFER);

uint16_t bg = 0;  // background - black
uint16_t fg = 0;  // foreground

void setup(void)
{
  enableDebugPorts();
  dmd.init();
  fg = dmd.Color888(0, 0, 255);
  dmd.setBrightness(100);
}

void loop(void) {
    // fill the matrix with points row by row
    for (int i = 0; i < dmd.height(); i++) {
        for (int j = 0; j < dmd.width(); j++) {
            dmd.drawPixel(j,i, fg);
            delay(30);
        }
    }
    // clear the screen
    dmd.fillScreen(bg);
}

digi55, 2024-08-20:
C:\Users\omero\Documents\Arduino\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:89:56: error: wrong number of template arguments (8, should be 6)
In file included from c:\Users\omero\Documents\Arduino\libraries\DMD_STM32-dev-V2/DMD_RGB.h:534:0,
                 from C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:8:
c:\Users\omero\Documents\Arduino\libraries\DMD_STM32-dev-V2/DMD_Panel_Templates.h:908:7: error: provided for 'template<int MUX_CNT, int P_Width, int P_Height, int SCAN, int SCAN_TYPE, int COL_DEPTH> class DMD_RGB_SHIFTREG_ABC'
 class DMD_RGB_SHIFTREG_ABC : public DMD_RGB<MUX_CNT, P_Width, P_Height, SCAN, SCAN_TYPE, COL_DEPTH>
       ^
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:89:61: error: invalid type in declaration before '(' token
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:89:165: error: expression list treated as compound expression in initializer [-fpermissive]
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino: In function 'void setup()':
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:98:7: error: request for member 'init' in 'dmd', which is of non-class type 'int'
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:99:12: error: request for member 'Color888' in 'dmd', which is of non-class type 'int'
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:100:7: error: request for member 'setBrightness' in 'dmd', which is of non-class type 'int'
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino: In function 'void loop()':
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:105:29: error: request for member 'height' in 'dmd', which is of non-class type 'int'
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:106:33: error: request for member 'width' in 'dmd', which is of non-class type 'int'
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:107:17: error: request for member 'drawPixel' in 'dmd', which is of non-class type 'int'
C:\Users\omero\Documents\Arduino\Ömer_stm32\RGB Panel\STM32_P5_dmd_rgb_pattern_test4\STM32_P5_dmd_rgb_pattern_test4.ino:112:9: error: request for member 'fillScreen' in 'dmd', which is of non-class type 'int'

exit status 1

Compilation error: wrong number of template arguments (8, should be 6)

digi55, 2024-08-20:
I did what you said, I get an error like this

board707, 2024-08-20:
> Compilation error: wrong number of template arguments (8, should be 6)

This error means that you use a version of the library prior to 1.1.3
Please install the latest version.

digi55, 2024-08-20:
Now everything works correctly.
Thank you so much @board707

board707, 2024-08-20:
You are welcome :)

digi55, 2024-08-20:
The scrolling text is not smooth as the P4 panel, which I used before. I will do some more testing tomorrow. Do you have any advice on why this is happening?

board707, 2024-08-20:
Is this the same as isuue #86  ?

board707, 2024-08-23:
Hi
I just added a new pattern `RGB80x40s10SR_digi55` for your matrix from this topic to the new library release. Now you can use it without adding a pattern code to your sketch.

digi55, 2024-08-23:
Thank you so much @board707

closed 2024-08-27 by board707